esp-rs/esp-idf-sys #173: How to modify SDK config when using native builder?

VorpalBlade, 2023-02-27:
I'm a bit confused by the stub documentation for modifying the SDK config in the README:

> (native builder only) Using cargo-idf to interactively modify ESP-IDF's sdkconfig file
> TBD: Upcoming

While I understand that this says that `cargo-idf` is not yet documented, I can't even find `cargo-idf` at all as a crate or github repo (I thought I could probably figure it out on my own if I had the tool).

Or is this supposed to mean that the tool in question isn't even written yet? It is not clear. If that is the case, what is the current approach for dealing with this?

Vollbrecht, 2023-02-27:
there is no tool for rust that lets you configure your project, like you used to config it with idf.py. you still can use all the configs but you have to manually configure them by simply writing them in the sdkconfig.defaults in the simplest case. this file gets picked up in the build process and makes sure things are configured like you wanted to. 
There was a plan to build an tool like that for the rust env but nobody done it. Everybody uses the sdkconfig files

ivmarkov, 2023-02-28:
Closing - question.

closed 2023-02-28 by ivmarkov

Vollbrecht, 2023-03-29:
To follow up for future people. If you want to have a full sdkconfig.default file with all the nobs in it, use idf.py to generate an new project for your target and use its sdkconfig file as an base. 
For example for an esp32c3 use:
- idf.py create-project test
- idf.py set-target esp32c3