esp-rs/esp-idf-sys #175: Error when building

elkel53930, 2023-02-28:
OS : Ubuntu 22.04
I installed the development environment according to The Rust on ESP Book.
Then I created a project for ESP32 by `cargo generate --git https://github.com/esp-rs/esp-idf-template cargo`.
I chose ESP-IDF version 4.4 and STD.
Then I ran `cargo build`.
But I get the following error.

```
(omitted)

   Compiling const_format v0.2.30
error: failed to run custom build command for `esp-idf-sys v0.31.12`

Caused by:
  process didn't exit successfully: `/home/k-iida/dev/esp-rust/target/debug/build/esp-idf-sys-3840c2390848d2c2/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.3.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/k-iida/dev/esp-rust/.embuild/espressif/esp-idf/release-v4.4",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: '/home/k-iida/dev/esp-rust/.embuild/espressif/esp-idf/release-v4.4' is not a git respository

  Caused by:
      0: command '"git" "rev-parse" "--show-toplevel"' failed to start
      1: No such file or directory (os error 2)
warning: build failed, waiting for other jobs to finish...
```

Do you have any idea what it is?

elkel53930, 2023-03-01:
Thank you for your advice MabezDev!

Actually I tried that many times yesterday.
But I just tried again now and it worked fine!
Did the remote repository have some trouble?
Anyway, thank you so much.

closed 2023-03-01 by elkel53930

m1212e, 2023-03-11:
So for anyone stumbling across this in the future: This can occurr if python is not installed on the system. Try running `cargo build -vv` (verbose mode) and see if this might be the case for you. Took me HOURS to find this.