esp-rs/esp-idf-sys #176: ESP-IDF master and v5.1-dev compile errors

jasta, 2023-03-01:
ESP-IDF v5.1-dev appears to have changed something that breaks esp-idf-sys' builds with errors like this:

```
.../components/riscv/include/riscv/rv_utils.h: Assembler messages:
.../components/riscv/include/riscv/rv_utils.h:79: Error: unrecognized opcode `csrw mtvec,a5', extension `zicsr' required
```

I did some debugging and the command-line is different if I go mucking around in `.embuild` manually.  Here's the output snippet for esp-idf-sys building things:

https://gist.github.com/07fd83db22a882c5e11af103d1bf4c85

And here's the working command from `idf.py -v build` in one of the samples from the same `.embuild` dir:

https://gist.github.com/d360c2e6f716af9416d7e29428348742

Biggest difference I see is that the idf.py build passes -march twice and also -mabi/-mcmodel but I'm not familiar enough with embuild and friends to know why that would be different and only broken for master/v5.1-dev.

jasta, 2023-03-01:
Updating esp-idf-hal to master got me past the soc_periph_ledc_clk_src_legacy_t issue, someone else already patched it.  I can confirm that I can successfully work around (for now) the -march issues by using:

```
CRATE_CC_NO_DEFAULTS=1 cargo build
```

This I think is pretty bad though as it drops -mabi and -mcmodel which seems like it could backfire in surprising ways when we start using cc in other crates.

jasta, 2023-03-02:
Oh interesting, cc-rs will actually work just fine if we change the target to riscv32imc_zicsr_zifencei-esp-espidf (worst target name ever).  That breaks several other assumptions we've made though, including in cc-rs and even in rustc.  This is awkward, but it really does seem like gcc is telling us that we should update everything.  Not cool RISC-V, not cool...

jasta, 2023-03-02:
Cc @ivmarkov, I see you're the one who committed riscv32-esp-espidf target support to rustc, I'm betting you know a lot more about this than I do :)

ivmarkov, 2023-03-02:
> Cc @ivmarkov, I see you're the one who committed riscv32-esp-espidf target support to rustc, I'm betting you know a lot more about this than I do :)

I don't. `riscv32imc-esp-espidf` is essentially a replica of Rust's pre-existing `riscv32imc-unknown-none-elf` target, with added extra info like `vendor = espressif` and `target_os = espidf`. The only purpose `riscv32imc-esp-espidf` exists is so that we can build Rust STD for the / on top of ESP IDF, for which we need to branch in the Rust STD code depending on whether the underlying "OS" is the ESP IDF or some other Posix OS.

With that said, I'm unsure how to proceed with that particular issue. Some analysis:
- First of all, Rust seems to be still unaware of the "new RiscV spec", in that `rustc --print target-list` does not list any RISCV targets that contain `zicsr`. So in a way, it is still complying to the old spec, which - to my understanding - has the `zicsr` & friends considered part of the 'I' extension.
- Old GCC versions (and thus ESP IDFs < 5.1) work the same way
- New GCC versions (and thus ESP IDFs >= 5.1) work the new way

For the foreseeable future, we need to support both combinations (Rust + old ESP IDF SDK and Rust + new ESP IDF SDK). 

Perhaps we can do some branching/conditional in `embuild` based on the ESP IDF version OR based on the GCC toolchain? As in if we detect that the GCC toolchain is newer, we should force `cc-rs` to use the `riscv32imc_zicsr_zifencei-esp-espidf` (which of course does not really exist, but who cares - for `cc-rs` this is just a string that it parses to derive what to pass to the C compiler).

ivmarkov, 2023-03-08:
> This is my understanding as well. Rust is unaware of this because we don't have targets for the new architecture (which ofc aren't really new but for our purposes I think that's how we should treat it). This is sort of like what would happen if all the sudden we got an x86 128 bit arch, you'd have to add this new arch in dozens of places before you can use it.

I think the comparison you do above might be slightly misleading. The way I understand it, is that Rust's **current** `riscv32imc-` targets are in fact `riscv32imc_zicsr_zifencei-` targets according to the *new* RISCV spec. Because what is named in the old RISCV spec `riscv32imc-` is in fact equal to `riscv32imc_zicsr_zifencei-` in the new RISCV spec.

What Rust doesn't have is the new "pure" `riscv32imc-` targets, which do *not* implement the `zicsr` and `zifencei` extensions, but since the esp32cXX chips in fact DO implement these extensions, we can just continue to use the existing Rust targets on those.

> 1. Just add the new arch alongside the old one (to rustc, cc, our examples, etc).  User has to work this out and if they want to use 5.1+, they'll need to change the --target they use in .cargo/config.toml.

Waiting for Rust to do - whatever - just to allow folks to use ESP IDF 5.1 seems like the wrong choice for me.

It is absolutely not so simple to add the new targets. 
* First of all, to add those to Rust, LLVM itself has to have these new targets (and this is a target which is the current one MINUS zicsr and minus  zifencei). 
* Further, we have the problem of backwards compatibility, because - again - the existing `riscv32imc-` target is in fact the target that you need, except wrongly named (and that's a wrong name only according to the NEW spec). 
* Last but not least - you actually need the *existing* targets (putting aside how these are named).

> 2. Pass compatibility option in cc-rs (--isa-spec=2.2 iirc) that makes gcc 12 behave like gcc 11.  Not recommended by RISC-V folks it seems but I don't know why exactly.

Why not? Given the situation, it seems this is the cleanest option, until everyone had upgraded?

> 3. Swap the order cc-rs adds -march vs CFLAGS such that the CMAKE_C_FLAGS we pick up from esp-idf ends up turning the gcc command line into -march=rv32imc ... -march=rv32imc_zicsr_ziferencei.  See my log output above which shows we actually are passing both the correct and incorrect -march, the order is such that the wrong arch is last.  This probably works by sheer dumb luck but I suspect is fragile with lots of corner cases.
> I feel like (3) isn't really an option and will blow up in our face quickly. (1) and (2) are pretty significant changes that commit us down a heavy maintenance path and likely need to include a few more stakeholders, but I'm not sure who to contact.

If we can't do (2) from above, we should be doing (3) IMO. So my order in terms of feasibility would be:
(2)
(3)
(1)

jasta, 2023-03-08:
> I think the comparison you do above might be slightly misleading. The way I understand it, is that Rust's **current** `riscv32imc-` targets are in fact `riscv32imc_zicsr_zifencei-` targets according [...]

Fair, but it's actually worse for a target meaning to change outside the Rust ecosystem since now passing --target riscv32imc-esp-espidf (as just one example) is ambiguous: do we get Zicsr and Zifencei opcodes or not?  It's unknowable outside of Rust (only by knowing gcc version and -misa-spec flags can we tell), which means tools will need to become smarter to try to work this out if we wanted to support both the old and new way of specifying these extensions.  See below tho since I agree with you maybe side stepping this mess is better.

> 
> What Rust doesn't have is the new "pure" `riscv32imc-` targets, which do *not* implement the `zicsr` and `zifencei` extensions, but since the esp32cXX chips in fact DO implement these extensions, we can just continue to use the existing Rust targets on those.

Unfortunately we shouldn't patch cc-rs or rustc with esp32 specific stuff since ofc the RISC-V ecosystem is way bigger than esp and rust has an obligation to support it (and ideally correctly) for everyone.  Solving this outside of cc-rs/rustc is fragile since our only real option is to try to hackily override the hard coded -march, so we become kind of stuck with this line of thinking.

> Waiting for Rust to do - whatever - just to allow folks to use ESP IDF 5.1 seems like the wrong choice for me.

Agreed, and actually I presented this option a bit unfairly.  We could actually do (2) or (3) first _then_ (1) later, which I think is my preference.

> It is absolutely not so simple to add the new targets.
> * First of all, to add those to Rust, LLVM itself has to have these new targets (and this is a target which is the current one MINUS zicsr and [...]

Yeah honestly rustc would need a flag here beyond --target that would let you flavor the target a little, which is way out of scope of what I'm willing to support.  I think we should let rustc and RISC-V folks hash this out directly.

> 
> > 2. Pass compatibility option in cc-rs (--isa-spec=2.2 iirc) that makes gcc 12 behave like gcc 11.  Not recommended by RISC-V folks it seems but I don't know why exactly.
> 
> Why not? Given the situation, it seems this is the cleanest option, until everyone had upgraded?

I couldn't say, the post I linked to in my earlier comments was from some RISC-V gcc maintainer that said it's strongly not recommended but without explanation.  I agree, this seems like the best solution for cc-rs today.  I can't 100% tell if this solves every problem tho as I'm now wondering whether rustc or anything else is going to end up trying to use gcc 12 and mess stuff up.  Still, i prefer to patch cc-rs and find out later.

> If we can't do (2) from above, we should be doing (3) IMO. So my order in terms of feasibility would be:
> (2)
> (3)
> (1)
L

Agreed, but we should file an issue about (1) at least so upstream can start getting ahead of this as well.  I'll go ahead and prepare a patch to cc-rs and verify (2) truly works then file an issue and we can close this one once that's all done.

> 
> 
>

ivmarkov, 2023-03-09:
> I couldn't say, the post I linked to in my earlier comments was from some RISC-V gcc maintainer that said it's strongly not recommended but without explanation. I agree, this seems like the best [...]

I think this message of yours is the crux of our misunderstanding. You seem to assume that Rustc is somehow dependent on GCC, which is not the case(*). Rustc is not part of GCC. Rustc does not use GCC codegen (that is, aside from the experimental libgcjit support). Rustc DOES use LLVM for codegen. So Rustc's "view" of what `riscv32imc` means is essentially the LLVM view. [Here's the **current** LLVM view of `riscv32imc`](https://discourse.llvm.org/t/support-for-zicsr-and-zifencei-extensions/68369). To summarize for you:
* LLVM treats `riscv32imc` as something which does have the the Z* extensions
* LLVM adheres to the earlier 2.0 version of the riscv spec
* All `riscv32imc` targets of LLVM (and thus Rustc) assume the Z* extensions
* (Obviously) Rustc will change its behavior w.r.t. Z* extensions only when LLVM itself does
 
So yeah, it IS safe to assume that all Rustc `riscv32imc-` targets are Z* targets.

> > >    I feel like (3) isn't really an option and will blow up in our face quickly. (1) and (2) are pretty significant changes that commit us down a heavy maintenance path and likely need to include a few more stakeholders, but I'm not sure who to contact.

> Agreed, but we should file an issue about (1) at least so upstream can start getting ahead of this as well. I'll go ahead and prepare a patch to cc-rs and verify (2) truly works then file an issue and we can close this one once that's all done.

Upstream what exactly? I hope in light of what I explained above you realize there is nothing to upstream (yet). In fact, it would be very risky to upstream anything in the current moment, because I don't think anybody in the Rust / LLVM communities is clear on how the new riscv spec would be supported:
a) In a backward compatible way, where `riscv32imc` would remain to mean "+Z* extensions"
b) In a backwards incompatible way, where `riscv32imc` would mean MINUS "Z*" and all current targets would need to be renamed then

(*) Rustc might use the GCC **linker** (as an external executable only). It calls it the "OS" linker. The "OS" linker is GCC on *nix and on Windows with the MingW Rustc flavor, and the MSVC linker for "standard" Windows Rustc. I don't recall right now as to whether you had to specify the arch target param when linking. I hope not.

ivmarkov, 2023-03-09:
One more thing: changes to `cc-rs` should be done carefully, and only in the code path that invokes GCC for compilation. If `cc-rs` invokes clang, the current behavior should stay unchanged, as clang is also llvm based.

jasta, 2023-03-10:
Thanks for the context, that's definitely the piece I was missing.  In that case then yeah for sure the best path forward is (2).  I'll work on a patch to cc-rs to confirm it works with -misa-spec=2.2 for the GCC toolchain path.

MabezDev, 2023-04-03:
@jasta Did you make any progress with the cc-rs patch?

shauder, 2023-04-11:
@jasta I am also wondering if you made any progress with a cc-rs patch? I am running into the same issue with "extension `zicsr' required". 

I can get around that, like you, using this:
```
CRATE_CC_NO_DEFAULTS=1 cargo build
```

But then I run into different errors, which look similar to what you fixed by moving to the master branch for 'esp-idf-hal'. 
```
   Compiling mipidsi v0.5.0
error[E0425]: cannot find value `ledc_clk_cfg_t_LEDC_AUTO_CLK` in this scope
     --> /Users/shane/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.40.1/src/ledc.rs:119:22
      |
119   |             clk_cfg: ledc_clk_cfg_t_LEDC_AUTO_CLK,
      |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a constant with a similar name exists: `ledc_clk_src_t_LEDC_APB_CLK`
      |
     ::: /Users/shane/Documents/projects/rust/rust-esp32-std-demo/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-bb7e60c1e8354035/out/bindings.rs:53673:1
      |
53673 | pub const ledc_clk_src_t_LEDC_APB_CLK: ledc_clk_src_t = 4;
      | ----------------------------------------------------- similarly named constant `ledc_clk_src_t_LEDC_APB_CLK` defined here

For more information about this error, try `rustc --explain E0425`.
error: could not compile `esp-idf-hal` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
```

Admittedly I am in a bit above my head at this point but plan to keep poking at it.

Nassiel, 2023-05-01:
Hi @jasta if you replace the constant value with the real one, so you set `clk_cfg: 4` (as a temporary solution), then a million more errors arise. `cc_no_defaults` may not be the final solution :( 

Report:
```
error[E0425]: cannot find value `SNTP_OPMODE_POLL` in this scope
  --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:40:36
   |
40 |             OperatingMode::Poll => SNTP_OPMODE_POLL as u8_t,
   |                                    ^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `SNTP_OPMODE_LISTENONLY` in this scope
  --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:41:42
   |
41 |             OperatingMode::ListenOnly => SNTP_OPMODE_LISTENONLY as u8_t,
   |                                          ^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0308]: mismatched types
   --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/espnow.rs:126:53
    |
126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
    |                                                ---- ^^^^^^^^^^^^^^^^^^^ expected fn pointer, found fn item
    |                                                |
    |                                                arguments to this enum variant are incorrect
    |
    = note: expected fn pointer `unsafe extern "C" fn(*const esp_now_recv_info, _, _)`
                  found fn item `extern "C" fn(*const u8, _, _) {EspNow::recv_callback}`
help: the type constructed contains `extern "C" fn(*const u8, *const u8, i32) {EspNow::recv_callback}` due to the type of the argument passed
   --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/espnow.rs:126:48
    |
126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
    |                                                ^^^^^-------------------^
    |                                                     |
    |                                                     this argument influences the type of `Some`
note: tuple variant defined here
   --> /home/<myhome>/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/option.rs:571:5
    |
571 |     Some(#[stable(feature = "rust1", since = "1.0.0")] T),
    |     ^^^^

error[E0425]: cannot find function `sntp_setoperatingmode` in this scope
     --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:144:18
      |
144   |         unsafe { sntp_setoperatingmode(u8_t::from(conf.operating_mode)) };
      |                  ^^^^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `esp_sntp_setoperatingmode`
      |
     ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:28997:5
      |
28997 |     pub fn esp_sntp_setoperatingmode(operating_mode: esp_sntp_operatingmode_t);
      |     -------------------------------------------------------------------------- similarly named function `esp_sntp_setoperatingmode` defined here

error[E0425]: cannot find function `sntp_setservername` in this scope
     --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:150:22
      |
150   |             unsafe { sntp_setservername(i as u8, c_server.as_ptr()) };
      |                      ^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `esp_sntp_setservername`
      |
     ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:29013:5
      |
29013 |     pub fn esp_sntp_setservername(idx: u8_t, server: *const ::core::ffi::c_char);
      |     ---------------------------------------------------------------------------- similarly named function `esp_sntp_setservername` defined here

error[E0425]: cannot find function `sntp_init` in this scope
     --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:156:13
      |
156   |             sntp_init();
      |             ^^^^^^^^^ help: a function with a similar name exists: `memp_init`
      |
     ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:28336:5
      |
28336 |     pub fn memp_init();
      |     ------------------ similarly named function `memp_init` defined here

error[E0425]: cannot find function `sntp_stop` in this scope
     --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:184:22
      |
184   |             unsafe { sntp_stop() };
      |                      ^^^^^^^^^ help: a function with a similar name exists: `httpd_stop`
      |
     ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:39161:5
      |
39161 |     pub fn httpd_stop(handle: httpd_handle_t) -> esp_err_t;
      |     ------------------------------------------------------ similarly named function `httpd_stop` defined here
```

ivmarkov, 2023-05-02:
> Hi @jasta if you replace the constant value with the real one, so you set `clk_cfg: 4` (as a temporary solution), then a million more errors arise. `cc_no_defaults` may not be the final solution :(. In this case, I'm using release/5.1, not the -dev branch but seems that the issue is the same. I'm also using the latest versions of hal-0.40.1, sys-0.32.1 and svc-0.45.
> 
> Report:
> 
> ```
> error[E0425]: cannot find value `SNTP_OPMODE_POLL` in this scope
>   --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:40:36
>    |
> 40 |             OperatingMode::Poll => SNTP_OPMODE_POLL as u8_t,
>    |                                    ^^^^^^^^^^^^^^^^ not found in this scope
> 
> error[E0425]: cannot find value `SNTP_OPMODE_LISTENONLY` in this scope
>   --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:41:42
>    |
> 41 |             OperatingMode::ListenOnly => SNTP_OPMODE_LISTENONLY as u8_t,
>    |                                          ^^^^^^^^^^^^^^^^^^^^^^ not found in this scope
> 
> error[E0308]: mismatched types
>    --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/espnow.rs:126:53
>     |
> 126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
>     |                                                ---- ^^^^^^^^^^^^^^^^^^^ expected fn pointer, found fn item
>     |                                                |
>     |                                                arguments to this enum variant are incorrect
>     |
>     = note: expected fn pointer `unsafe extern "C" fn(*const esp_now_recv_info, _, _)`
>                   found fn item `extern "C" fn(*const u8, _, _) {EspNow::recv_callback}`
> help: the type constructed contains `extern "C" fn(*const u8, *const u8, i32) {EspNow::recv_callback}` due to the type of the argument passed
>    --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/espnow.rs:126:48
>     |
> 126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
>     |                                                ^^^^^-------------------^
>     |                                                     |
>     |                                                     this argument influences the type of `Some`
> note: tuple variant defined here
>    --> /home/<myhome>/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/option.rs:571:5
>     |
> 571 |     Some(#[stable(feature = "rust1", since = "1.0.0")] T),
>     |     ^^^^
> 
> error[E0425]: cannot find function `sntp_setoperatingmode` in this scope
>      --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:144:18
>       |
> 144   |         unsafe { sntp_setoperatingmode(u8_t::from(conf.operating_mode)) };
>       |                  ^^^^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `esp_sntp_setoperatingmode`
>       |
>      ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:28997:5
>       |
> 28997 |     pub fn esp_sntp_setoperatingmode(operating_mode: esp_sntp_operatingmode_t);
>       |     -------------------------------------------------------------------------- similarly named function `esp_sntp_setoperatingmode` defined here
> 
> error[E0425]: cannot find function `sntp_setservername` in this scope
>      --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:150:22
>       |
> 150   |             unsafe { sntp_setservername(i as u8, c_server.as_ptr()) };
>       |                      ^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `esp_sntp_setservername`
>       |
>      ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:29013:5
>       |
> 29013 |     pub fn esp_sntp_setservername(idx: u8_t, server: *const ::core::ffi::c_char);
>       |     ---------------------------------------------------------------------------- similarly named function `esp_sntp_setservername` defined here
> 
> error[E0425]: cannot find function `sntp_init` in this scope
>      --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:156:13
>       |
> 156   |             sntp_init();
>       |             ^^^^^^^^^ help: a function with a similar name exists: `memp_init`
>       |
>      ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:28336:5
>       |
> 28336 |     pub fn memp_init();
>       |     ------------------ similarly named function `memp_init` defined here
> 
> error[E0425]: cannot find function `sntp_stop` in this scope
>      --> /home/<myhome>/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/sntp.rs:184:22
>       |
> 184   |             unsafe { sntp_stop() };
>       |                      ^^^^^^^^^ help: a function with a similar name exists: `httpd_stop`
>       |
>      ::: /home/<myhome>/Code/rust/esp32-idf-garden-kit/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-ff84662b16cd1bee/out/bindings.rs:39161:5
>       |
> 39161 |     pub fn httpd_stop(handle: httpd_handle_t) -> esp_err_t;
>       |     ------------------------------------------------------ similarly named function `httpd_stop` defined here
> ```

But you have to patch the `esp-idf-*` crates, or at least the `esp-idf-hal` one with their `master` versions.

Nassiel, 2023-05-06:
Hi, @ivmarkov. I continue working on this one, and I opened (I think is even related) issue #200 I used the master, modified it and removed the hal/svc to simplify.

ivmarkov, 2023-05-08:
This should be fixed as of https://github.com/esp-rs/esp-idf-sys/commit/0e20985a39a0b3bc5798e73e211ea0405dc156d6
There is a bit more to be desired w.r.t. detection of GCC 12+ (I do it by the ESP IDF version which likely won't work well when using GIT hashes), but the rest should be mostly OK actually.