esp-rs/esp-idf-sys #177: How do I enable the PSRAM component?

defunctzombie, 2023-03-06:
I see there is some bindings code already present for psram (https://github.com/esp-rs/esp-idf-sys/blob/5ca2e76feaeb73731ad3be0e1adf395d0a65361e/src/include/esp-idf/bindings.h#L24) but it is gated by the define check.

I've read the README guide for extra components: https://github.com/esp-rs/esp-idf-sys#extra-esp-idf-components but unfortunately I am still lost on what I should change in my config.toml or build.rs to enable this component.

There is also a [CONFIG_SPIRAM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram) from the kconfig docs from espressif.

Any points in the right direction on what config to twiddle would be super! Or a point to a better place to post this question.

ivmarkov, 2023-03-08:
@defunctzombie I don't think @bugadani has tested the `ESP_IDF_COMP_ESP_PSRAM_ENABLED` check when this code was contributed. as he is working on ESP IDF 4.4. Neither did I. `psram` does look like a separate component, but that might be wrong. If simply removing this `#ifdef` works, I'm willing to bite the bullet and remove it for now.

ivmarkov, 2023-03-08:
^^^ Just patch your `esp-idf-sys` code in `Cargo.toml` with `[patch.crates-io]`

bugadani, 2023-03-08:
Indeed, I have not tested this with IDF 5.0. But in theory, you need something like this in your `Cargo.toml`:

```toml
[package.metadata.esp-idf-sys]
esp_idf_components = ["esp_psram"]
```

ivmarkov, 2023-03-09:
> Indeed, I have not tested this with IDF 5.0. But in theory, you need something like this in your `Cargo.toml`:
> 
> ```toml
> [package.metadata.esp-idf-sys]
> esp_idf_components = ["esp_psram"]
> ```

The point is, if you don't specify `esp_idf_components`, esp-idf-sys enables all of them, which implies that there is no such thing as an `esp_psram` component in the first place, though the psram code seems to live in its dedicated component folder in esp idf 5.

bugadani, 2023-03-09:
Maybe the issue is that the last esp-idf-sys release is 3 months old and doesn't include my patch? The component exists and it is also picked up by the build system, I can see it listed in multiple places.

ivmarkov, 2023-03-09:
@defunctzombie Are you using a released version? I was assuming you are on master...

bugadani, 2023-03-09:
I have just tested this with esp-idf master (just because I'm low on sleep and the template didn't offet 5.0), and esp-idf-sys master and the psram component is picked up just fine.

[image]

defunctzombie, 2023-03-09:
@ivmarkov I started with one of the template examples and have the following in my Cargo.toml dependencies:

```
esp-idf-sys = { version = "0.32.1", features = ["binstart"] }
esp-idf-hal = "0.40.1"
```

I'm happy to try newer versions etc since this is an early stage project without much lock in to anything specific.

ivmarkov, 2023-03-09:
@defunctzombie You should have checked the source code tagged with `V0.32.1` then, not master. :)

Please patch your `esp-idf-sys` dependency with the `master` branch until we release a new version. I mean, as in ujsing Cargo's `[patch.crates-io]` section, which should be pretty easy. Closing. @bugadani thanks for following up on this!

closed 2023-03-09 by ivmarkov

defunctzombie, 2023-03-09:
> Please patch your `esp-idf-sys` dependency with the `master` branch until we release a new version. I mean, as in using Cargo's `[patch.crates-io]` section,

@ivmarkov Could you clarify this for me (I am new to Rust and am learning which ways to update dependencies are correct in which settings). Do you mean I should update my dependency to use the `master` branch via git directly in the dependencies list (This is something I've done), or some other technique?

Once I update the dependency to the master branch is there any further action I need to take from what is described in this discussion thread or will I be able to invoke the expected PSRAM methods?

ivmarkov, 2023-03-09:
I think the maintainers of Cargo [had explained it much more clearly](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html) than what I would be able to do here.

mohsenasm, 2024-04-07:
In version `v0.34.1`, we just need to add this line in the `sdkconfig.defaults` file to enable the PSRAM:
```
CONFIG_ESP32_SPIRAM_SUPPORT=y
```

fko-kuptec, 2024-04-30:
> In version `v0.34.1`, we just need to add this line in the `sdkconfig.defaults` file to enable the PSRAM:
> 
> ```
> CONFIG_ESP32_SPIRAM_SUPPORT=y
> ```

In my testing, this was not sufficient to get the PSRAM working. Thankfully, I found this [post](https://github.com/esp-rs/esp-idf-sys/issues/273#issuecomment-1879003866) in another issue with a `sdkconfig.defaults` snipped for the PSRAM, and with this it works. I think, the `CONFIG_SPIRAM=y` flag is the most important line, since I tried the others and it still wouldn't work.

```config
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y
```

lorenzo-pirchio, 2024-06-27:
Hi, I'm working on an esp32s3 m0n8r8 on a project in Rust  for a xmpp client.
Since the crate is heavy, using the internal RAM, I get "stack overflow error".
So I decided to use the PSRAM available as heap to increase the size of the stack.
I tried to add these lines:

CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y

But I can't see if it works.
Should it show the size at the boot up?

ivmarkov, 2024-06-27:
> Hi, I'm working on an esp32s3 m0n8r8 on a project in Rust for a xmpp client. My project is based on the esp-idf-template. Since the crate is heavy, using the internal RAM, I get "stack overflow error". So I decided to use the PSRAM available as heap to increase the size of the stack. I tried to add these lines in config.default:
> 
> CONFIG_SPIRAM=y CONFIG_SPIRAM_MODE_OCT=y CONFIG_SPIRAM_SPEED_80M=y
> 
> But I can't see if it works. Should it show the size at the boot up?

Read [this](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/external-ram.html). Placing your stacks in PSRAM is difficult.

lorenzo-pirchio, 2024-06-27:
I get this line, as if it's not activated:

I (334) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM

To active it, I have to add these lines:

CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y

Or I have to call other functions in the main?

Vollbrecht, 2024-06-27:
Besides setting the sdkconfigs you don't need anything to "activate" it.

For using it you can read through what ivmarkov linked to understand the different possibility are for example with respect to flags like [this](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-spiram-use)

lorenzo-pirchio, 2024-06-27:
I have read the link @ivmarkov provided me, but it's very difficult to understand what I should do since I'm a noob in embedded programming.
I thought about using either heap_caps_malloc() or malloc().
Should I call these functions in the main?
If you could help me, I'd be thankful.

ivmarkov, 2024-06-27:
> I have read the link @ivmarkov provided me, but it's very difficult to understand what I should do since I'm a noob in embedded programming. I thought about using either heap_caps_malloc() or malloc(). Should I call these functions in the main? If you could help me, I'd be thankful.

1. You might **not** even have to call `heap_caps_malloc` or `malloc` "manually" if you enable `CONFIG_SPIRAM_USE=CONFIG_SPIRAM_USE_MALLOC`.

What will happen then is that you can just call `Box::new` from Rust (which more or less resolves to calling `malloc`) and ESP IDF will **intelligently** decide if to allocate the `Box` for you from SPIRAM, or from regular internal RAM.

The problem is with "intelligently". You have no guarantee from where the memory will be given to you - from SPIRAM or from RAM. I think for large `Box`-es it will give them out from SPIRAM anyway, but for smaller, it could be from anywhere. To some extend you can control this with `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` which is also described there.

2. Stack allocations

If you have problems with stack memory, NONE of the above would help you (because ESP IDF cannot easily allocate your thread stacks from SPIRAM), so you **do** you find a way to allocate the necessary RAM directly on heap. Ideally avoiding the typical for rust "first-allocate-on-stack-then-move-to-heap" drama.

Sorry if the above is too complex, but this space is complex. I suggest you pay a bit with putting stuff in `Box::new` with maximum optimization settings - as a start. You can do that with or without SPIRAM. And try to understand what is causing your stack blowups after all (which object) by minimizing your code until it starts to work. Then put that object in a `Box`.

ivmarkov, 2024-06-27:
Also: `xmpp` by the way is one of those crates that was never designed to run on an MCU. Without (potentially heavily) patching it, you might not have a chance. You have to understand you are trying a bit to fit an elephant in a bottle here.

lorenzo-pirchio, 2024-06-27:
I'm just trying to connect it to the server but I get stack overflow so my idea was to use the external RAM. 
Firstly I tried to remove useless code, but there are just few lines of code remaining. 
But I think I'll start patching it before trying to execute it again.

lorenzo-pirchio, 2024-06-27:
> Also: `xmpp` by the way is one of those crates that was never designed to run on an MCU. Without (potentially heavily) patching it, you might not have a chance. You have to understand you are trying a bit to fit an elephant in a bottle here.

Do you think the problem is the CPU or the RAM?

ivmarkov, 2024-06-27:
RAM of course. And flash space to less extent.