esp-rs/esp-idf-sys #180: No `c_types` in the root when using bindings_module

wouterdebie, 2023-03-08:
I'm trying to add the `esp32-camera` component to my `esp-idf-template` based project. I get the code to compile, but when I use the `bindings_module` option in `extra_components`, I get the following error when compiling:

```
error[E0432]: unresolved import `crate::c_types`
     --> /Users/wouter.de.bie/prjs/wouter/esp-cam-test/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-6f1b4a5534d41a56/out/bindings.rs:63873:9
      |
63873 |     use crate::c_types; /* automatically generated by rust-bindgen 0.63.0 */
      |         ^^^^^^^^^^^^^^ no `c_types` in the root
```

I'm building on MacOS with v5.0, although the same error happens on v4.4.

wouterdebie, 2023-03-08:
That does the trick. Thanks!

ivmarkov, 2023-03-08:
Change pushed, thanks for noticing.

closed 2023-03-08 by ivmarkov