esp-rs/esp-idf-sys #201: How can we get bindgen output with wrap_static_fns() ?

gopakumarce, 2023-05-18:
So for example I want to include esp32/incude/hal/i2s_ll.h and all the static inlines inside it, but by default all the static inlines are not generated in the bindings. How can I include wrap_static_fns() so that I get the inline utility APIs inside i2s_ll.h ? From my project's build.rs can I pass in something to the embuild flags or anything of that sort to achieve this ?

ivmarkov, 2023-05-19:
`wrap_static_fns` is a brand new, experimental feature of bindgen and as such it is not supported out of the box in `esp-idf-sys`. Moreover, it is not just about "passing the flag to bindgen", as in that case bindgen would generate a C file too, that you need to compile with clang (or gcc would do as well?) and then you need to use cargo build cmds to pass down the compiled output to the linker.

Not a rocket science, but would require some changes to `esp-idf-sys` - in the native as well as in the pio code paths, (as long as it can even be supported in the latter).

Might also require a small change to `embuild` so that it's bindgen dependency is updated to a version which supports bindgen 0.65, if that's not the case already.

All in all it would require a PR. Are you willing to work on that?

gopakumarce, 2023-05-19:
@ivmarkov yes ill take a stab at it

gopakumarce, 2023-05-22:
PR here https://github.com/esp-rs/esp-idf-sys/pull/202

Vollbrecht, 2024-06-21:
We now use bindgen 0.69 both in embuild and this crate. I think they made additional changes how to handle inline functions across FFI layer,  but it seams that it would still need a lot of handholding on our part.