esp-rs/esp-idf-sys #203: error[E0308]: mismatched types when running cargo run (v0.33.0)

aecorredor, 2023-05-25:
I've followed the set up flow in the esp-rust book and I'm getting this error after `cargo run`:

```
  --> /.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.33.0/src/lib.rs:36:62
   |
36 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`
```

Vollbrecht, 2023-05-26:
did you use the cargo-generate template to generate your project? you need to use `rustflags = [ "--cfg",  "espidf_time64"]` when using esp-idf version 5+

aecorredor, 2023-05-26:
@Vollbrecht thanks for the response! I did use a template. I specifically used:

`cargo generate --git https://github.com/esp-rs/esp-idf-template cargo`

I just tried it with those flags (under the [build] section) and things kept going. Now I got a connection error (I'm using an M5Stack Core 2):

```
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Serial port not found
  help: Ensure that the device is connected and your host recognizes the serial adapter
```

I'm guessing I need a way to specify the port, right? Or should it automatically get it?

Also, my ESP-IDF installation is 5.0.2, should I change the `ESP_IDF_VERSION = "release/v4.4.4"` under `[env]` to that?

aecorredor, 2023-05-26:
I'm closing this since the original issue was resolved with the previous suggestion. The one feedback item I'd have would be to point this out in the esp-rust book. I'm also running into issues with the specific targets (risc vs xtensa), so it might be good to add more details about when to use each maybe (or maybe not? I'm just starting out in IoT, so maybe I'm just still pretty new).