esp-rs/esp-idf-sys #210: Restart after leaving deep sleep

Veanir, 2023-06-14:
Problem:
- My esp32s3 is restarting when i'm trying to do 'something' with deep sleep. I think that the problem is related to coming out of deep sleep. At first i stumbled upon it in my bigger project thinking it's related to bluedroid, but then I reproduced it in clean 'hello-world' [template](https://github.com/esp-rs/esp-idf-template)

Code:
```rs
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use log::*;

fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_sys::link_patches();
    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    info!("Hello, world!");

    std::thread::sleep(std::time::Duration::from_secs(1));

    unsafe {
        esp_idf_sys::esp_sleep_enable_timer_wakeup(1_000_000u64);
        esp_idf_sys::esp_deep_sleep_start();
    }

    // This line will never be reached
    info!("Goodbye, world!");
}
```

I tried it also with esp_deep_sleep(u64) with same result.

Is it really an issue or am I missing something?

Serial monitor:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - _bss_end
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - _iram_bss_end
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - _iram_bss_end
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - _iram_bss_end
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - _iram_bss_end
    at ??:??
I (48) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (49) boot: compile time Jun  7 2023 08:07:32
I (50) boot: Multicore bootloader
I (54) boot: chip revision: v0.1
I (58) boot.esp32s3: Boot SPI Speed : 40MHz
I (62) boot.esp32s3: SPI Mode       : DIO
I (67) boot.esp32s3: SPI Flash Size : 8MB
I (72) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (96) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (103) boot:  2 factory          factory app      00 00 00010000 007f0000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=3073ch (198460) map
I (173) esp_image: segment 1: paddr=00040764 vaddr=3fc91ba0 size=02410h (  9232) load
I (176) esp_image: segment 2: paddr=00042b7c vaddr=40374000 size=0d49ch ( 54428) load
I (194) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=48630h (296496) map
I (268) esp_image: segment 4: paddr=00098658 vaddr=4038149c size=00704h (  1796) load
I (269) esp_image: segment 5: paddr=00098d64 vaddr=600fe000 size=0002ch (    44)
I (280) boot: Loaded app from partition at offset 0x10000
I (280) boot: Disabling RNG early entropy source...
I (297) cpu_start: Pro cpu up.
I (297) cpu_start: Starting app cpu, entry point is 0x403760e0
0x403760e0 - call_start_cpu1
    at C:\WORKSPACE\CODE\test\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-7e8e8177fbcff550\out\build\C:/WORKSPACE/CODE/test/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/port\cpu_start.c:147
I (0) cpu_start: App cpu up.
I (311) cpu_start: Pro cpu start user code
I (311) cpu_start: cpu freq: 160000000
I (311) cpu_start: Application information:
I (314) cpu_start: Project name:     libespidf
I (319) cpu_start: App version:      1
I (323) cpu_start: Compile time:     Jun 14 2023 20:53:33
I (329) cpu_start: ELF file SHA256:  0000000000000000...
I (335) cpu_start: ESP-IDF:          e33a180-dirty
I (341) cpu_start: Min chip rev:     v0.0
I (346) cpu_start: Max chip rev:     v0.99
I (350) cpu_start: Chip rev:         v0.1
I (355) heap_init: Initializing. RAM available for dynamic allocation:
I (362) heap_init: At 3FC94A48 len 00054CC8 (339 KiB): D/IRAM
I (369) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (376) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (382) heap_init: At 600FE02C len 00001FD4 (7 KiB): RTCRAM
I (389) spi_flash: detected chip: generic
I (393) spi_flash: flash io: dio
I (398) sleep: Configure to isolate all GPIO pins in sleep state
I (404) sleep: Enable automatic switching of GPIO sleep configuration
I (411) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (2431) test: Hello, world!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - _bss_end
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - _iram_bss_end
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - _iram_bss_end
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - _iram_bss_end
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
```
Context:
- Board: ESP32S3 DevKitC
- Chip ESP32-S3-WROMM-1
- esp-idf version: v4.4 (probably, I selected it when generating template, but serial monitor in one place mentions ```I (48) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader```

Vollbrecht, 2023-06-14:
i think deep-sleep and connecting via USB/JTAG at the same time might not be possible, or at least interfere here. Could you try testing it only powering it and not be connected over USB/JTAG?

Veanir, 2023-06-14:
I was connected via uart that time, not sure why it says ```pro cpu reset by JTAG```.
Anyway, I tried the rmt_neopixel example from esp-idf-hal and put deep sleep in the middle, to have some visual feedback, and powered the board from power brick with what appears to be usb cable without data lines. But still no luck - led stays white, wich means that it doesn't pass esp_deep_sleep().

Code:
```rs
(...)
// 3 seconds white at 10% brightness
    neopixel(
        Rgb {
            r: 25,
            g: 25,
            b: 25,
        },
        &mut tx,
    )
    .unwrap();

    FreeRtos::delay_ms(3000);

    unsafe {
        esp_idf_sys::esp_deep_sleep(10);
    }

    // rainbow loop at 20% brightness
    let mut i: u32 = 0;
    loop {
        let rgb = hsv2rgb(i, 100, 20).unwrap();
        neopixel(rgb, &mut tx).unwrap();
        if i == 360 {
            i = 0;
        }
        i += 1;
        FreeRtos::delay_ms(10);
    }
}fn main
```

Vollbrecht, 2023-06-14:
neopixel stay's on even if the pin that drives it goes low. In your first log you see that the esp was in deepsleep but it get a wakeup reset `DSLEEP` . This and the fact that it reached a HelloWorld but nothing else shows that it sleeped. You should look into https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/sleep_modes.html and try disabling all wakeup reasons. than it should only wakeup with a hard reset.

Veanir, 2023-06-14:
Oh, it's my dummy miunderstanding. 
I thought that deep sleep is just a sleep but with very low power consumption. So code gets executed like:
```
Code A
deep sleep
Code B
```

But i haven't known that esp ALWAYS performs restart (assuming chatGPT told me truth) when returning from deep sleep.

so it turns out it's okay that code execution looks like:
```
loop {
   Code A
   deep sleep
  }
```

And now it's clear why I couldn't see any similiar issues - I'm just dumb haha
Either way, thank You, i don't know how much longer i would be debugging it when there are no bugs to find (hopefully) :d

And maybe someone finds my "discovery" helpful :) 

Summing up, what i thought was wrong behavior was actually correct - esp32 restarts after deep sleep.

closed 2023-06-14 by Veanir