esp-rs/esp-idf-sys #214: Linking fails with symbol `atexit` undefined for target `xtensa-esp32s2-espidf`

Shemnei, 2023-06-28:
Minimal reproducible example: https://github.com/Shemnei/esp32s2-nvs-linker-failure

Linking fails when calling the function `esp_idf_sys::nvs_flash_init_partition` on target `xtensa-esp32s2-espidf` with the following error:

```text
  = note: [ldproxy] Running ldproxy
          Error: Linker /home/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/lib/no-rtti/libstdc++.a(atomicity.o):(.literal._ZN9__gnu_cxx18__exchange_and_addEPVii+0x8): undefined reference to `atexit'
          /home/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/lib/no-rtti/libstdc++.a(atomicity.o): in function `_ZN9__gnu_cxx18__exchange_and_addEPVii':
          /builds/idf/crosstool-NG/.build/xtensa-esp32s2-elf/build/build-cc-gcc-final/xtensa-esp32s2-elf/no-rtti/libstdc++-v3/libsupc++/atomicity.cc:33: undefined reference to `atexit'
          collect2: error: ld returned 1 exit status

          Stack backtrace:
             0: <unknown>
             1: <unknown>
             2: <unknown>
             3: <unknown>
             4: <unknown>
             5: <unknown>
             6: __libc_start_main
             7: <unknown>

  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `aaaa` (bin "aaaa") due to previous error
```

Versions used:

- `esp-idf`: `v5.0.2`
- `embuild`: `0.31.2`
- `esp-idf-sys`: `0.33.1`
- `toolchain`: `1.70` (installed by `espup`)

The exact same setup compiles fine for the following targets:

- `xtensa-esp32-espidf`
- `xtensa-esp32s3-espidf`

If anyone can point to the right direction, i would be happy to help with fixing this.

Shemnei, 2023-06-29:
Hey, sorry for the late response.

I tried with a few other esp-idf versions:

- `v4.4.4`: Success
- `master`(https://github.com/espressif/esp-idf/commit/cf7e743a9b2e5fd2520be4ad047c8584188d54da): Fails with same error
- `v5.1-rc2`: Fails with same error

tlatzko, 2023-10-19:
Hi, for me it also failse with v4.4.6. Do you have any suggestions? What could be the reason for this?

Vollbrecht, 2023-10-19:
> Hi, for me it also fail with v4.4.6. Do you have any suggestions? What could be the reason for this?

A: Is your error the "exact" same error
- Do you can build with a fresh hello-world from the esp-idf-template?
- For what target are you building, how did you setup your env?
- Are you forgetting to call `esp_idf_sys::link_patches();` 
B: If not post your "exact" error with enough information in a separate issue

groffta, 2023-10-20:
Also having the same issue whenever I use anything from `esp_idf_sys`:
```
rustup --version
rustup 1.26.0 (5af9b9484 2023-04-05)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.73.0-nightly (9163a2087 2023-10-03) (1.73.0.0)`
```
```
espup --version
espup 0.6.1
```
```
ESP_IDF_VERSION = "v5.1.1"
```
Also tried with `v4.4.6`

Crate generated from template with `cargo generate esp-rs/esp-idf-template`

```
note: [ldproxy] Running ldproxy
          Error: Linker C:/Users/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe failed: exit code: 1
          STDERR OUTPUT:
          c:/users/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:/users/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/lib/no-rtti\libstdc++.a(atomicity.o):(.literal._ZN9__gnu_cxx18__exchange_and_addEPVii+0x8): undefined reference to `atexit'
          c:/users/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:/users/x/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/lib/no-rtti\libstdc++.a(atomicity.o): in function `__gnu_cxx::__exchange_and_add(int volatile*, int)':
          /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s2-elf/build/build-cc-gcc-final/xtensa-esp32s2-elf/no-rtti/libstdc++-v3/libsupc++/atomicity.cc:33: undefined reference to `atexit'
          collect2.exe: error: ld returned 1 exit status

          Stack backtrace:
             0: <unknown>
             1: <unknown>
             2: <unknown>
             3: <unknown>
             4: <unknown>
             5: <unknown>
             6: <unknown>
             7: <unknown>
             8: <unknown>
             9: <unknown>
            10: BaseThreadInitThunk
            11: RtlUserThreadStart

  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)
```

`main.rs`
```
// If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use log::*;
use esp_idf_hal::prelude::*;
use esp_idf_hal::gpio;
use esp_idf_hal::peripherals::Peripherals;
use esp_idf_svc::wifi::EspWifi;
use embedded_svc::wifi::{AuthMethod, ClientConfiguration, Configuration};

fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_sys::link_patches();
    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    ...
}

```

wzhd, 2023-11-02:
I get the same `undefined reference to atexit` with esp32c3 and esp-idf v5.1.1. The error is basically identical, though it doesn't show a meaningful function name in `libstdc++.a(atomicity.o)`.

> = note: [ldproxy] Running ldproxy
           Error: Linker /home/wzhd/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/wzhd/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/wzhd/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/lib/rv32imc_zicsr_zifencei/ilp32/no-rtti/libstdc++.a(atomicity.o): in function `.L0 ':
          /builds/idf/crosstool-NG/.build/riscv32-esp-elf/build/build-cc-gcc-final/riscv32-esp-elf/rv32imc_zicsr_zifencei/ilp32/no-rtti/libstdc++-v3/libsupc++/atomicity.cc:33: undefined reference to `atexit'
          collect2: error: ld returned 1 exit status

Stack backtrace:

             0: anyhow::error::<impl anyhow::Error>::msg
             1: ldproxy::main
             2: std::sys_common::backtrace::__rust_begin_short_backtrace
             3: std::rt::lang_start::{{closure}}
             4: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/core/src/ops/function.rs:287:13
             5: std::panicking::try::do_call
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/panicking.rs:485:40
             6: std::panicking::try
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/panicking.rs:449:19
             7: std::panic::catch_unwind
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/panic.rs:140:14
             8: std::rt::lang_start_internal::{{closure}}
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/rt.rs:148:48
             9: std::panicking::try::do_call
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/panicking.rs:485:40
            10: std::panicking::try
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/panicking.rs:449:19
            11: std::panic::catch_unwind
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/panic.rs:140:14
            12: std::rt::lang_start_internal
                       at /rustc/2eaeb1eee1b21772de8b935236d16ff8e03fdcf5/library/std/src/rt.rs:148:20
            13: main
            14: <unknown>
            15: __libc_start_main
            16: _start

>  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

An empty main() links fine, I located the part that would cause the error when included:

```
let wifi = esp_idf_svc::wifi::EspWifi::new(
    peripherals.modem,
    EspSystemEventLoop::take()?,
    Some(EspNvsPartition::<NvsDefault>::take()?)
)?;
```

More precisely, it's the nvs init, `unsafe { nvs_flash_init() };` is enough.

I put together the minimal code and ran it on Actions: https://github.com/wzhd/link-no-atexit/actions/runs/6726578699/job/18283082241

Vollbrecht, 2023-11-02:
please run the same .config.toml as our esp-idf-template e.g add `"-C", "default-linker-libraries"` to the rustflags inside the config.toml

wzhd, 2023-11-02:
I see, `esp-idf-template/cargo/.cargo/config.toml` still has `"-C", "default-linker-libraries"` in `rustflags`, but `rust-esp32-std-demo/.cargo/config.toml` has already removed the argument and no longer consider it needed
 
https://github.com/ivmarkov/rust-esp32-std-demo/commit/359bc826879ecf5984e510f70d72dec81deaed58#diff-9a4f3e4537ebb7474452d131b0d969d89a51286f4269aac5ef268e712be17268L26

It works almost perfectly without `"-C", "default-linker-libraries"`, currently I'm just completely avoiding nvs.

Vollbrecht, 2024-06-21:
with the upstream gcc13 based xtensa compiler there were numerous changes how c++ code is handled. I don't think this is currently a problem anymore with ESP_IDF_VERSION >= v5.2.2, and people should not need to use `default-linker-libraries`. So i am closing this for now.

 If anybody still runs into this issue with the newest versions feel free to reopen this issue or create a new one.

closed 2024-06-21 by Vollbrecht