esp-rs/esp-idf-sys #216: Include bindings for esp_lcd

anlumo, 2023-07-02:
The ESP32-S3 has an LCD peripheral included. I got it to work with esp-idf-sys, but the functions and types aren't generated for it.

I solved it by creating my own bindings.h with the following content:

```c
#if CONFIG_IDF_TARGET_ESP32S3
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/include/esp_lcd_types.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/include/esp_lcd_panel_rgb.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/include/esp_lcd_panel_io.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/include/esp_lcd_panel_ops.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/include/esp_lcd_panel_vendor.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/include/esp_lcd_panel_commands.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/interface/esp_lcd_panel_interface.h"
#include ".embuild/espressif/esp-idf/release-v5.1/components/esp_lcd/interface/esp_lcd_panel_io_interface.h"
#endif
```

and then included it in Cargo.toml:

```toml
[package.metadata.esp-idf-sys]
extra_components = [{ bindings_header = "bindings.h" }]
```

This should be included by default (with the correct relative path of course, I'm doing this based on the installation in the project).

ivmarkov, 2023-07-09:
Would you like to open a PR against `esp-idf-sys` that adds these bindings to its `bindings.h` file?

ivmarkov, 2023-11-20:
Fixed by https://github.com/esp-rs/esp-idf-sys/commit/9551878ec922d534f28a68b227e363e8134e4467

closed 2023-11-20 by ivmarkov