esp-rs/esp-idf-sys #219: i was planning to do a HelloBoard test program on esp32-c3,but i got this issue,how can i solve this problem.

majorzpley, 2023-07-22:
PS F:\std-training\intro\hardware-check> cargo run
   Compiling esp-idf-sys v0.33.0
   Compiling wifi v0.1.0 (F:\std-training\common\lib\wifi)
   Compiling rgb-led v0.1.0 (F:\std-training\common\lib\rgb-led)
The following warnings were emitted during compilation:

warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.33.0`

Caused by:
  process didn't exit successfully: `F:\std-training\intro\hardware-check\target\debug\build\esp-idf-sys-95652ea66b616fb0\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
[image]

  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity
  PATH=C:\Users\wyx12\.espressif\esp-idf\v4.4.4\tools;%PATH%

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: Some(
          [
              "F:\\std-training\\intro\\hardware-check\\./sdkconfig.defaults",
          ],
      ),
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v4.4.4",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v4.4.4") }
  E:\APPS\MSYS2\ucrt64\bin\python.exe: No module named pip
  ERROR: Python interpreter at E:\APPS\MSYS2\ucrt64\bin\python.exe doesn't have pip installed. Please check the Getting Started Guides for the steps to install prerequisites for your OS.
  Error: Could not install esp-idf

  Caused by:
      command '"python" "C:\\Users\\wyx12\\.espressif\\esp-idf\\v4.4.4\\tools\\idf_tools.py" "--idf-path" "C:\\Users\\wyx12\\.espressif\\esp-idf\\v4.4.4" "--non-interactive" "install-python-env"' exited with non-zero status code 1
warning: build failed, waiting for other jobs to finish...

N3xed, 2023-07-26:
You need a functioning [python](https://www.python.org/downloads/release/python-3114/) installation including its package manager pip (which should come by default with any new installation). The `python.exe` executable must be in the `PATH` environment variable (`Path` on windows, because case-insensitive).

I can see from the error log, that it's currently using python from your msys2 installation, which is probably not a good idea, and it doesn't have pip which caused the error.

majorzpley, 2023-07-27:
Sorry, After uninstall MSY2 ，i still can not compile the esp-idf-sys part,This is the log from terminal,i don;t known how to fix it.

PS F:\std-training\intro\hardware-check> cargo build
   Compiling esp-idf-sys v0.33.0
The following warnings were emitted during compilation:

warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.33.0`

Caused by:
  process didn't exit successfully: `F:\std-training\intro\hardware-check\target\debug\build\esp-idf-sys-c7a9467380ed8278\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity
  IDF_PYTHON_ENV_PATH=C:\Users\wyx12\.espressif\python_env\idf4.4_py3.10_env
  PATH=C:\Users\wyx12\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\wyx12\.espressif\tools\cmake\3.23.1\bin;C:\Users\wyx12\.espressif\tools\ninja\1.10.2\;C:\Users\wyx12\.espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\wyx12\.espressif\esp-idf\v4.4.4\tools;%PATH%
  Current system platform: win64
  Skipping riscv32-esp-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=C:\Users\wyx12\.espressif\python_env\idf4.4_py3.10_env
  PATH=C:\Users\wyx12\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\wyx12\.espressif\tools\cmake\3.23.1\bin;C:\Users\wyx12\.espressif\tools\ninja\1.10.2\;C:\Users\wyx12\.espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\wyx12\.espressif\esp-idf\v4.4.4\tools;%PATH%
  cargo:rerun-if-changed=F:\std-training\intro\hardware-check\sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_riscv32imc-esp-espidf = None
  CMAKE_riscv32imc_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "F:\\std-training\\intro\\hardware-check\\target\\riscv32imc-esp-espidf\\debug\\build\\esp-idf-sys-281d3f81b0a0f0ca\\out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=C:\\Users\\wyx12\\.espressif\\esp-idf\\v4.4.4\\tools\\cmake\\toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=C:/Users/wyx12/.espressif/python_env/idf4.4_py3.10_env/Scripts/python.exe" "-DCMAKE_INSTALL_PREFIX=F:\\std-training\\intro\\hardware-check\\target\\riscv32imc-esp-espidf\\debug\\build\\esp-idf-sys-281d3f81b0a0f0ca\\out" "-DCMAKE_C_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany"
  -- Checking Python dependencies...
  The following Python requirements are not satisfied:
  click>=7.0
  pyserial>=3.3
  future>=0.15.2
  cryptography>=2.1.4
  pyparsing>=2.0.3,<2.4.0
  pyelftools>=0.22
  idf-component-manager~=1.0
  gdbgui==0.13.2.0; python_version < "3.11"
  pygdbmi<=0.9.0.2; python_version < "3.11"
  python-socketio<5; python_version < "3.11"
  jinja2<3.1; python_version < "3.11"   # See https://github.com/espressif/esp-idf/issues/8760
  itsdangerous<2.1; python_version < "3.11"
  kconfiglib==13.7.1
  reedsolo>=1.5.3,<=1.5.4
  bitstring>=3.1.6,<4
  ecdsa>=0.16.0
  construct==2.10.54
  esp-windows-curses; sys_platform == 'win32'
  Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
  Diagnostic information:
      IDF_PYTHON_ENV_PATH: (not set)
      Python interpreter used: C:\Users\wyx12\.espressif\python_env\idf4.4_py3.10_env\Scripts\python.exe
      Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
      PATH: C:\Users\wyx12\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\wyx12\.espressif\esp-idf\v4.4.4\tools;C:\Users\wyx12\.espressif\tools\ninja\1.10.2\;C:\Users\wyx12\.espressif\tools\cmake\3.23.1\bin;C:\Users\wyx12\.espressif\python_env\idf4.4_py3.10_env\Scripts;F:\std-training\intro\hardware-check\target\debug\deps;F:\std-training\intro\hardware-check\target\debug;C:\Users\wyx12\.rustup\toolchains\nightly-2023-02-28-x86_64-pc-windows-msvc\bin;E:\APPS\Vmware16\bin\;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;E:\APPS\Bandizip\;C:\Program Files\dotnet\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;E:\A [... cut ...]
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    C:/Users/wyx12/.espressif/esp-idf/v4.4.4/tools/cmake/build.cmake:416 (__build_check_python)
    CMakeLists.txt:16 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit code: 1

  build script failed, must exit now', C:\Users\wyx12\.cargo\registry\src\github.com-1ecc6299db9ec823\cmake-0.1.50\src\lib.rs:1098:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

Julien-cpsn, 2023-10-14:
Same error here, any solution?

Julien-cpsn, 2023-10-14:
> Sorry, After uninstall MSY2 ，i still can not compile the esp-idf-sys part,This is the log from terminal,i don;t known how to fix it.
> 
> PS F:\std-training\intro\hardware-check> cargo build Compiling esp-idf-sys v0.33.0 The following warnings were emitted during compilation:
> 
> warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity
> 
> error: failed to run custom build command for `esp-idf-sys v0.33.0`
> 
> Caused by: process didn't exit successfully: `F:\std-training\intro\hardware-check\target\debug\build\esp-idf-sys-c7a9467380ed8278\build-script-build` (exit code: 101) --- stdout cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS cargo:rerun-if-env-changed=MCU cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE cargo:rerun-if-env-changed=ESP_IDF_VERSION cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR cargo:rerun-if-env-changed=IDF_PATH cargo:rerun-if-env-changed=EXTRA-COMPONENTS cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity IDF_PYTHON_ENV_PATH=C:\Users\wyx12.espressif\python_env\idf4.4_py3.10_env PATH=C:\Users\wyx12.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\wyx12.espressif\tools\cmake\3.23.1\bin;C:\Users\wyx12.espressif\tools\ninja\1.10.2;C:\Users\wyx12.espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\wyx12.espressif\esp-idf\v4.4.4\tools;%PATH% Current system platform: win64 Skipping riscv32-esp-elf@esp-2021r2-patch5-8.4.0 (already installed) Skipping cmake@3.23.1 (already installed) Skipping ninja@1.10.2 (already installed) IDF_PYTHON_ENV_PATH=C:\Users\wyx12.espressif\python_env\idf4.4_py3.10_env PATH=C:\Users\wyx12.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\wyx12.espressif\tools\cmake\3.23.1\bin;C:\Users\wyx12.espressif\tools\ninja\1.10.2;C:\Users\wyx12.espressif\python_env\idf4.4_py3.10_env\Scripts;C:\Users\wyx12.espressif\esp-idf\v4.4.4\tools;%PATH% cargo:rerun-if-changed=F:\std-training\intro\hardware-check\sdkconfig.defaults CMAKE_PREFIX_PATH_riscv32imc-esp-espidf = None CMAKE_PREFIX_PATH_riscv32imc_esp_espidf = None TARGET_CMAKE_PREFIX_PATH = None CMAKE_PREFIX_PATH = None CMAKE_riscv32imc-esp-espidf = None CMAKE_riscv32imc_esp_espidf = None TARGET_CMAKE = None CMAKE = None running: "cmake" "F:\std-training\intro\hardware-check\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-281d3f81b0a0f0ca\out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=C:\Users\wyx12\.espressif\esp-idf\v4.4.4\tools\cmake\toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=C:/Users/wyx12/.espressif/python_env/idf4.4_py3.10_env/Scripts/python.exe" "-DCMAKE_INSTALL_PREFIX=F:\std-training\intro\hardware-check\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-281d3f81b0a0f0ca\out" "-DCMAKE_C_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS= -ffunction-sections -fdata-sections -march=rv32imc -mabi=ilp32 -mcmodel=medany" -- Checking Python dependencies... The following Python requirements are not satisfied: click>=7.0 pyserial>=3.3 future>=0.15.2 cryptography>=2.1.4 pyparsing>=2.0.3,<2.4.0 pyelftools>=0.22 idf-component-manager~=1.0 gdbgui==0.13.2.0; python_version < "3.11" pygdbmi<=0.9.0.2; python_version < "3.11" python-socketio<5; python_version < "3.11" jinja2<3.1; python_version < "3.11" # See [espressif/esp-idf#8760](https://github.com/espressif/esp-idf/issues/8760) itsdangerous<2.1; python_version < "3.11" kconfiglib==13.7.1 reedsolo>=1.5.3,<=1.5.4 bitstring>=3.1.6,<4 ecdsa>=0.16.0 construct==2.10.54 esp-windows-curses; sys_platform == 'win32' Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide Diagnostic information: IDF_PYTHON_ENV_PATH: (not set) Python interpreter used: C:\Users\wyx12.espressif\python_env\idf4.4_py3.10_env\Scripts\python.exe Warning: python interpreter not running from IDF_PYTHON_ENV_PATH PATH: C:\Users\wyx12.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\wyx12.espressif\esp-idf\v4.4.4\tools;C:\Users\wyx12.espressif\tools\ninja\1.10.2;C:\Users\wyx12.espressif\tools\cmake\3.23.1\bin;C:\Users\wyx12.espressif\python_env\idf4.4_py3.10_env\Scripts;F:\std-training\intro\hardware-check\target\debug\deps;F:\std-training\intro\hardware-check\target\debug;C:\Users\wyx12.rustup\toolchains\nightly-2023-02-28-x86_64-pc-windows-msvc\bin;E:\APPS\Vmware16\bin;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;E:\APPS\Bandizip;C:\Program Files\dotnet;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;E:\APPS\MSYS2\usr\bin;E:\APPS\Erlang OTP\bin;C:\Program Files\Microsoft SQL Server\150\Tools\Binn;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC [... cut ...] Call Stack (most recent call first): C:/Users/wyx12/.espressif/esp-idf/v4.4.4/tools/cmake/build.cmake:416 (__build_check_python) CMakeLists.txt:16 (idf_build_process)
> 
> thread 'main' panicked at ' command did not execute successfully, got: exit code: 1
> 
> build script failed, must exit now', C:\Users\wyx12.cargo\registry\src\github.com-1ecc6299db9ec823\cmake-0.1.50\src\lib.rs:1098:5 note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

Same error here, any solution?

Vollbrecht, 2023-10-14:
@Julien-cpsn since the other problem was older, how did you setup your toolchain? also how did you generate the project. Did you go through all steps mention in the esp-rust book https://esp-rs.github.io/book/installation/index.html 
and made used the https://github.com/esp-rs/esp-idf-template to make a simple hello-world with all mentiond prerequest? Also what platform you are on etc

Julien-cpsn, 2023-10-14:
@Vollbrecht Hello, thanks for you quick reply

Here are my system versions: 
- Windows 11 22621.2283
- rustc 1.75.0-nightly
- cargo 1.75.0-nightly
- ESP-IDF 5.1

My rustup targets:
- x86_64-pc-windows-msvc (installed)
- riscv32imac-unknown-none-elf (installed)
- riscv32imc-unknown-none-elf (installed)

I may be creating another git issue, but let's try here first.

The problem I am running into is weird.
Sometimes it is the same problem as described above by @majorzpley and sometimes it's this one:
```
C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test>cargo build
   Compiling std v0.0.0 (C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std)
   Compiling semver v1.0.20
   Compiling serde_json v1.0.107
   Compiling cargo-platform v0.1.4
   Compiling envy v0.4.2
   Compiling camino v1.1.6
   Compiling rustc_version v0.4.0
error[E0583]: file not found for module `wait_status`
  --> C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\process\process_unsupported.rs:58:1
   |
58 | mod wait_status;
   | ^^^^^^^^^^^^^^^^
   |
   = help: to create the module `wait_status`, create file "C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\process\wait_status.rs" or "C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\process\wait_status\mod.rs"

error[E0432]: unresolved import `wait_status::ExitStatus`
  --> C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys\unix\process\process_unsupported.rs:59:9
   |
59 | pub use wait_status::ExitStatus;
   |         ^^^^^^^^^^^^^^^^^^^^^^^ no `ExitStatus` in `sys::unix::process::process_inner::wait_status`
   |
help: consider importing this struct instead
   |
59 | pub use crate::process::ExitStatus;
   |         ~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0432]: unresolved import `crate::sys::process::ExitStatus`
  --> C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys_common\process.rs:10:35
   |
10 | use crate::sys::process::{EnvKey, ExitStatus, Process, StdioPipes};
   |                                   ^^^^^^^^^^ no `ExitStatus` in `sys::unix::process`
   |
   = help: consider importing this struct instead:
           crate::process::ExitStatus

   Compiling heapless v0.7.16
   Compiling embuild v0.31.3
   Compiling cargo_metadata v0.15.4
error[E0599]: no function or associated item named `from_inner` found for struct `ExitStatus` in the current scope
    --> C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\os\unix\process.rs:300:30
     |
300  |         process::ExitStatus::from_inner(From::from(raw))
     |                              ^^^^^^^^^^ function or associated item not found in `ExitStatus`
     |
    ::: C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\process.rs:1594:1
     |
1594 | pub struct ExitStatus(imp::ExitStatus);
     | --------------------- function or associated item `from_inner` not found for this struct
     |
     = help: items from traits can only be used if the trait is implemented and in scope
note: `FromInner` defines an item `from_inner`, perhaps you need to implement it
[... 71 lines cut ...]
     |
62   | pub trait AsInner<Inner: ?Sized> {
     | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0599]: no function or associated item named `from_inner` found for struct `ExitStatus` in the current scope
    --> C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\process.rs:1601:21
     |
1594 | pub struct ExitStatus(imp::ExitStatus);
     | --------------------- function or associated item `from_inner` not found for this struct
...
1601 |         ExitStatus::from_inner(imp::ExitStatus::default())
     |                     ^^^^^^^^^^ function or associated item not found in `ExitStatus`
     |
     = help: items from traits can only be used if the trait is implemented and in scope
note: `FromInner` defines an item `from_inner`, perhaps you need to implement it
    --> C:\Users\Julien\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\std\src\sys_common\mod.rs:80:1
     |
80   | pub trait FromInner<Inner> {
     | ^^^^^^^^^^^^^^^^^^^^^^^^^^

Some errors have detailed explanations: E0432, E0583, E0599.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `std` (lib) due to 10 previous errors
warning: build failed, waiting for other jobs to finish...
```

Vollbrecht, 2023-10-15:
yeah this problem is the one related to the issue you pointed out and has nothing to do with the initial problem from thes issue. please just use no compiler newer than 2023-10-13 from the nightly branch. You can install every daily nightly rust compiler via 'rustup install nightly-2023-10-12` . Make sure to also than update the channel to `nightly-<date>` inside rust-toolchain.toml or specify the compiler toolchain when running cargo b

Julien-cpsn, 2023-10-15:
After some check, my error is a little different from @majorzpley's 

here is the trace:
```
   Compiling esp-idf-sys v0.33.2
   Compiling esp-idf-hal v0.41.2
   Compiling esp-idf-svc v0.46.2
   Compiling no-std-net v0.5.0
   Compiling esp32-rust-test v0.1.0 (C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test)
   Compiling embassy-futures v0.1.0
   Compiling embedded-svc v0.25.3
The following warnings were emitted during compilation:

warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.33.2`

Caused by:
  process didn't exit successfully: `C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\target\debug\build\esp-idf-sys-972b151738ea9101\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity
  IDF_PYTHON_ENV_PATH=C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\python_env\idf5.1_py3.11_env
  ESP_IDF_VERSION=5.1
  PATH=C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\tools\cmake\3.24.0\bin;C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\tools\ninja\1.10.2\;C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\python_env\idf5.1_py3.11_env\Scripts;C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\esp-idf\v5.1.1\tools;%PATH%
  IDF_DEACTIVATE_FILE_PATH=C:\Users\Julien\AppData\Local\Temp\tmp_urloj19idf_1656
  Current system platform: win64
  Skipping riscv32-esp-elf@esp-12.2.0_20230208 (already installed)
  Skipping cmake@3.24.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\python_env\idf5.1_py3.11_env
  ESP_IDF_VERSION=5.1
  PATH=C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\tools\cmake\3.24.0\bin;C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\tools\ninja\1.10.2\;C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\python_env\idf5.1_py3.11_env\Scripts;C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\esp-idf\v5.1.1\tools;%PATH%
  IDF_DEACTIVATE_FILE_PATH=C:\Users\Julien\AppData\Local\Temp\tmpov6zq9l4idf_12624
  cargo:rerun-if-changed=C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc_zicsr_zifencei-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_zicsr_zifencei_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_riscv32imc_zicsr_zifencei-esp-espidf = None
  CMAKE_riscv32imc_zicsr_zifencei_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "C:\\Users\\Julien\\Documents\\Programmation\\rust\\esp32-rust-test\\target\\riscv32imc-esp-espidf\\debug\\build\\esp-idf-sys-f9cb6210d5c4efc0\\out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=C:\\Users\\Julien\\Documents\\Programmation\\rust\\esp32-rust-test\\.embuild\\espressif\\esp-idf\\v5.1.1\\tools\\cmake\\toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=C:/Users/Julien/Documents/Programmation/rust/esp32-rust-test/.embuild/espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe" "-DCMAKE_INSTALL_PREFIX=C:\\Users\\Julien\\Documents\\Programmation\\rust\\esp32-rust-test\\target\\riscv32imc-esp-espidf\\debug\\build\\esp-idf-sys-f9cb6210d5c4efc0\\out" "-DCMAKE_C_FLAGS= -march=rv32imc_zicsr_zifencei -ffunction-sections -fdata-sections -march=rv32imc_zicsr_zifencei -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc_zicsr_zifencei -ffunction-sections -fdata-sections -march=rv32imc_zicsr_zifencei -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections -march=rv32imc_zicsr_zifencei -mabi=ilp32 -mcmodel=medany"
  -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.41.0.windows.1")
  -- The C compiler identification is GNU 12.2.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: C:/Users/Julien/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Checking Python dependencies...
  Python requirements are satisfied.
  Constraint file: C:\Users\Julien\Documents\Programmation\rust\esp32-rust-test\.embuild\espressif\espidf.constraints.v5.1.txt
  Requirement files:
[... cut ...]
      esp_supplicant/src/crypto/crypto_mbedtls-bignum.c.obj

    cannot be safely placed under this directory.  The build may not work
    correctly.

  thread 'main' panicked at C:\Users\Julien\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cmake-0.1.50\src\lib.rs:1098:5:

  command did not execute successfully, got: exit code: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Sadly, I really don't find where it comes from...

Vollbrecht, 2023-10-15:
This error you are posting again is another error, its a windows specific error. On windows the path cannot be to long (yeah wtf windows) or it dont work. The only workaround here is to directly user your projects inside a root dir like C:\your-project or use it inside WSL where you dont have such problems. Windows gonna windows.

Julien-cpsn, 2023-10-15:
Ok thanks man, ima go with the WSL workaround !
I'll try to let you updated if anythingh goes wrong

closed 2023-10-31 by ivmarkov