esp-rs/esp-idf-sys #223: Cant build on ARM OSX

skrcka, 2023-07-29:
I was trying to cargo build esp-idf-template and build fails on this package

There seems to be this issue: xtensa-esp32-elf-gcc: error: unrecognized command line option '--target=xtensa-esp32-espidf'
Also it's probably not the only issue, I tried to bypass it and got some other but don't have logs from that.

logs:
[log.txt](https://github.com/esp-rs/esp-idf-sys/files/12207368/log.txt)

N3xed, 2023-08-01:
Well, this is a strange error. CMake shouldn't be setting the offending flag.

My hunch is that there is maybe some environment variable that sets the `--target=xtensa-esp32-esp-idf` flag which CMake incorperates.

I can't really help you more than that, without knowing more about your setup.

ivmarkov, 2023-08-08:
ARM OSX is supported since a long time ago. Are you sure that you have followed all install &build steps correctly?
Please also get rid of any `CC_FLAGS` `RUSTFLAGS` and so on env variables.

ivmarkov, 2023-08-08:
And by the way `--target=xtensa-esp32-espidf` is not a valid GCC target (and probably not even a valid GCC command line flag) - this is a Rustc thing. Please double check all your env vars.

dacut, 2023-08-08:
@ivmarkov I was wondering what we could be logging (maybe in embuild?) to try to debug this kind of stuff easier. Might be too much, or maybe just log all environment variables on error? (Though that might capture secrets...)

N3xed, 2023-08-08:
There is already a good chunk of logging, but if you want to show it you have to run cargo with the `-vv` flag to show build script output.

MikeMitterer, 2023-10-01:
I had the same problem. `env` on the cmdline showed me that I've set CC and CXX. After `unset CC` and `unset CXX` compiling worked. Maybe these vars should be checked in export-esp.sh???

ivmarkov, 2023-10-01:
But sometimes you might **want** to set these. The problem is, these are often set in a global way "for all projects" which is an anti-pattern, **unless** you know what you are doing.

MikeMitterer, 2023-10-07:
@ivmarkov >unless you know what you are doing
The thing is that the toolchain for ESPs is extremely complex, so most people don't know what goes on in all the files and tools used in/for the toolchain. The error message you get from espflash, if CC and CXX is set, is not very intuitive. It took me several hours to find out that the problem is the CC and CXX setting I had in my .zshrc since several years causing the problem... A warning or so would help!

ivmarkov, 2023-10-07:
> @ivmarkov >unless you know what you are doing The thing is that the toolchain for ESPs is extremely complex, so most people don't know what goes on in all the files and tools used in/for the [...]

I have no idea what `espflash` has to do with all of this... (after all, it is just a flasher utility?!), but fear not!
This was committed a couple of days ago: https://github.com/esp-rs/esp-idf-sys/commit/849149bd420165daf049c51f982dbae104fa9f97

ivmarkov, 2023-10-07:
Closing in light of the above ^^^.

closed 2023-10-07 by ivmarkov

MikeMitterer, 2023-10-08:
Hm, OK it's just a tool but in my opinion the tool should also help people to solve (potential) problems or to give them some hints. Rust is well known for its great error messages. Tools should make no exception to this behavior.

ivmarkov, 2023-10-08:
> Hm, OK it's just a tool but in my opinion the tool should also help people to solve (potential) problems or to give them some hints. Rust is well known for its great error messages. Tools should make no exception to this behavior.

My point is, this is the wrong tool for that problem. `espflash` does not know, and should not know about `esp-idf-sys` build internals. The problem should be addressed inside  the `esp-idf-sys` build itself, as done already.

rdeterre, 2024-03-29:
Just in case someone else ends up here looking for a solution to this issue, setting `CRATE_CC_NO_DEFAULTS` while building esp-idf-sys as recommended [here](https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$PsQE2CLb3s65Yq3jtTepnIcr0jNeh6Z3JoDUvxWLJvs?via=matrix.org&via=tchncs.de) made the build work on my system.

`CRATE_CC_NO_DEFAULTS=1 cargo build`