esp-rs/esp-idf-sys #224: Can't build from project

LarryMerino, 2023-07-31:
I've been working for some time without problems with a project that has the following dependencies:
`
[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]

default = ["torrot"]
torrot = []

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33.1", features = ["binstart"] }
esp-idf-hal = { version = "0.41.2", features = [
    "edge-executor",
    "embassy-sync",
    "critical-section",
] }
esp-idf-svc = { version = "0.46.2", features = [
    "nightly",
    "experimental",
    "embassy-time-driver",
    "embassy-time-isr-queue",
] }
embedded-svc = { version = "0.25.1", features = ["nightly", "experimental"] }
embedded-hal = "1.0.0-alpha.10"
# ansi_term has no support since 4 year ago. Look to replace for nu-ansi-term instead
ansi_term = "0.12"
async-channel = "1.9.0"
statig = "0.3.0"
crossbeam-channel = "0.5.8"
futures-lite = "1.13.0"
once_cell = "1.18.0"
regex = { version = "1.8.4", default-features = false, features = ["std", "unicode-perl"] }
esp32-nimble = "0.1.4"
edge-executor = "0.3.0"
embassy-futures = { version = "0.1" }
embassy-sync = { version = "0.1", features = ["std"] }
embassy-time = "0.1.2"
heapless = "0.7.16"
lazy_static = { version = "1.4.0" }
serde = { version = "1", default-features = false }

[build-dependencies]
embuild = "0.31.2"
`
But now when I try to create a new project using the same dependencies I can't build it

This is the error: 

`
error[E0432]: unresolved imports `esp_idf_sys::ble_uuid_any_t`, `esp_idf_sys::ble_uuid_cmp`, `esp_idf_sys::os_mbuf_append`
 --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.1.4/src/server/ble_characteristic.rs:5:19
  |
5 | use esp_idf_sys::{ble_uuid_any_t, ble_uuid_cmp, os_mbuf_append};
  |                   ^^^^^^^^^^^^^^  ^^^^^^^^^^^^  ^^^^^^^^^^^^^^ no `os_mbuf_append` in the root
  |                   |               |
  |                   |               no `ble_uuid_cmp` in the root
  |                   no `ble_uuid_any_t` in the root

error[E0432]: unresolved imports `esp_idf_sys::ble_uuid_any_t`, `esp_idf_sys::ble_uuid_cmp`, `esp_idf_sys::os_mbuf_append`
 --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.1.4/src/server/ble_descriptor.rs:5:19
  |
5 | use esp_idf_sys::{ble_uuid_any_t, ble_uuid_cmp, os_mbuf_append};
  |                   ^^^^^^^^^^^^^^  ^^^^^^^^^^^^  ^^^^^^^^^^^^^^ no `os_mbuf_append` in the root
  |                   |               |
[... 615 lines cut ...]
    |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `esp_idf_sys`

error[E0531]: cannot find unit struct, unit variant or constant `ble_error_codes_BLE_ERR_CONN_TERM_LOCAL` in crate `esp_idf_sys`
   --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.1.4/src/ble_return_code.rs:106:20
    |
106 |       esp_idf_sys::ble_error_codes_BLE_ERR_CONN_TERM_LOCAL => {
    |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `esp_idf_sys`

error[E0531]: cannot find unit struct, unit variant or constant `ble_error_codes_BLE_ERR_CONN_ESTABLISHMENT` in crate `esp_idf_sys`
   --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.1.4/src/ble_return_code.rs:109:20
    |
109 |       esp_idf_sys::ble_error_codes_BLE_ERR_CONN_ESTABLISHMENT => {
    |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `esp_idf_sys`

error[E0425]: cannot find value `ble_hs_cfg` in crate `esp_idf_sys`
  --> /Users/larry/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-nimble-0.1.4/src/ble_security.rs:13:20
   |
13 |       esp_idf_sys::ble_hs_cfg.set_sm_bonding(bonding as _);
   |                    ^^^^^^^^^^ not found in `esp_idf_sys`

continue...
`

I'm using esp-idf-template to create a new project and I can't get it to work with version 4.4 or 5.1
Any idea what I may be doing wrong?

N3xed, 2023-08-01:
Did you enable bluetooth in the sdkconfig as per [the usage section of `esp32-nimble`](https://github.com/taks/esp32-nimble#usage)? Also it seems like [`esp32-nimble` uses esp-idf `v4.4.2`](https://github.com/taks/esp32-nimble/blob/95a462d3e59877c0035137952c812174bef45050/.cargo/config.toml#L35).

This issue doesn't seem to be related to esp-idf-sys, more so [`esp32-nimble`](https://github.com/taks/esp32-nimble).

N3xed, 2023-08-01:
No worries. Closing.

closed 2023-08-01 by N3xed