esp-rs/esp-idf-sys #225: Problems including as a dependency in a library crate

kyp44, 2023-08-01:
Hi, loving the ESP-IDF Rust ecosystem so great work!

I'm writing a board support crate for an ESP32 platform. I need to include `esp-idf-sys` as a dependency, literally only for `EspError`. I am having trouble getting my library crate to compile at all on my local platform (i.e. when try to I compile the crate itself rather than including it as a dependency in a binary crate), but my main concern is being able to have `docs.rs` automatically build the documentation for my crate once I publish it on `crates.io`. I couldn't find any guidance in the `esp-idf-sys` documentation that addresses this problem, so is there an easy way to do this?

It seems like `esp-idf-sys` downloads and builds the ESP-IDF C project as part of it's build, so that it can run `bindgen` on its header files, so maybe this isn't so straightforward. However, is there a way to make `esp-idf-sys` just run `bindgen` on the headers without trying to build and link the actual C library?

If needed, I can provide details on what I've tried and what specific errors I am getting, or otherwise just link to my project, which is already up on GitHub, but not yet published on `crates.io`.

Vollbrecht, 2023-08-01:
related https://github.com/esp-rs/esp-idf-hal/issues/239 
even with this option is not enough in our case unfortunately . 

you can try to manually download the docs.rs docker setup to replicate the issue. The core problem i think currently is that docs.rs doesn't allow any other files downloaded/run outsite the build dir. On default we use the .embuild dir to localy download and install esp-idf and building the bindgen. 

Technically there is an option ( ENV var)  in esp-idf-sys to not install in .embuild but using the build dir. But for some reason i cant get it to work in there build environment. There environment  just seem to ignore that. 

If you want to give it a try - download there docker setup and try it yourself.

Otherwise you always can build the documentation locally and then host it yourself. We use this approach currently if you look at the README of esp-idf-hal/svc on the docu button

kyp44, 2023-08-02:
I appreciate the response. I would definitely just self-host the documentation, however I cannot even build the library by itself locally, so I cannot build the documentation. I can only compile it when its used as a dependency in my main binary project. I tried copying the build script and cargo settings from my main binary project, but I can't get `esp-id-sys` to compile properly in the standalone library. I can provide more details on this if you would be able to quickly identify the problem, how the build system works is a weak point of mine, especially when it involves also compiling C code.

Vollbrecht, 2023-08-02:
check out how we build it in our ci, that should give you a hint how you can build it locally as a library standalone

ivmarkov, 2023-08-03:
> Hi, loving the ESP-IDF Rust ecosystem so great work!
> 
> I'm writing a board support crate for an ESP32 platform. I need to include `esp-idf-sys` as a dependency, literally only for `EspError`.

That sounds very, very weird to the point that I can't believe that. You are saying that you need `EspError` itself, but not ANY of the code that can generate it. 

Sounds to me you are passing on generics where you really need them.

kyp44, 2023-08-03:
Indeed all I _really_ need is the HAL crate (`esp-idf-hal`). I have wrapper functions that setup ESP32 peripherals in a certain way compatible with the chips they are connected to on this particular board (e.g. setting up an `I2cDriver` with a particular baud rate, etc.). Creating these HAL crate peripheral driver structs usually returns a `Result<T, EspError>`, so that my function also needs to return this. Hence I need to have `EspError` in scope to specify this return type. `EspError` comes from `esp-idf-sys`, which is not re-exported from the HAL crate. I'm not sure how this would be done with generics as you suggest, but it might just be that I'm not creative enough.

ivmarkov, 2023-08-04:
Oh, got it. `esp-idf-hal`/`esp-idf-svc` *can* reexport `EspError`, but it is probably not worth it because via those you'll still transitively depend on the `esp-idf-sys` crate so you still need to build it anyway.

As to building your library crate, and thus `esp-idf-sys` and friends, how about just

`cargo build --target xtensa-esp32-espidf -Zbuild-std=std,panic_abort -Zbuild-std-features=panic_immediate_abort`?

dacut, 2023-08-04:
Oooh, we should probably re-export EspError from esp-idf-hal here, just so a crate doesn't have to specify a version of esp-idf-sys. That would allow for later binding (from the binary crate).

I created https://github.com/esp-rs/esp-idf-hal/issues/284 for this.

closed 2023-08-08 by ivmarkov