esp-rs/esp-idf-sys #227: Can't build due to type mismatch

kaarelen, 2023-08-01:
(from "cargo generate esp-rs/esp-idf-template cargo')

cub@archlinux ~/P/rust-esp-std-test (master)> cargo run
   Compiling esp-idf-sys v0.33.1
   Compiling esp-idf-hal v0.41.2
   Compiling esp-idf-svc v0.46.2
   Compiling rust-esp-std-test v0.1.0 (/home/cub/Projects/rust-esp-std-test)
error[E0308]: mismatched types
  --> /home/cub/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.1/src/lib.rs:36:62
   |
36 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /home/cub/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.1/src/lib.rs:38:51
   |
38 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` (lib) due to 2 previous errors

Vollbrecht, 2023-08-01:
if you are using esp-idf version 5.x you need to use the `rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]` flag but if you are using esp-idf version 4.4 you should not use it. this is probably your error

kaarelen, 2023-08-01:
@Vollbrecht Thanks! You right!
Have a good day!
(template should be configured via advanced submenu, because by default it 4.x, regardless to actual idf version)

closed 2023-08-01 by kaarelen

dacut, 2023-08-04:
https://github.com/esp-rs/esp-idf-sys/pull/228 improves the documentation here a bit.