esp-rs/esp-idf-sys #234: Fails to build crate

RoyalFoxy, 2023-08-23:
I have created a project from the template and have not modified a single thing. I attached the log of `cargo build` below.

[log.txt](https://github.com/esp-rs/esp-idf-sys/files/12418343/log.txt)

As for my setup, I'm using a macbook with m2 chip and I went through the rust on esp book multiple times to double and triple check if I have everything installed.

below is a list of what I installed and how

- rust: through rustup downloaded from the official site
- nightly toolchain: with this command `rustup toolchain install nightly --component rust-src`
- riscv targets: with this command
```
rustup target add riscv32imc-unknown-none-elf # For ESP32-C2 and ESP32-C3
rustup target add riscv32imac-unknown-none-elf # For ESP32-C6 and ESP32-H2
```
- llvm compiler infrastructure: with this command `brew install llvm`
- python: with `brew install python`
- git: preinstalled with dev tools
- ldproxy: with `cargo install ldproxy`
- espup: with `cargo install espup`
- espflash: with `cargo install espflash`
- esp compatible rust etc.: with `espup install`

Note that I could not build the std template while the no_std template worked perfectly fine!

RoyalFoxy, 2023-08-23:
Also forgot to mention that I use fish as my shell and I had to manually rewrite the paths... This is what I put into my fish config in the end

```fish
set LIBCLANG_PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib"
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:$PATH"
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:$PATH"
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:$PATH"
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:$PATH"
```

I copied these from the env file that gets created which looks like this on my machine

```sh
export LIBCLANG_PATH="/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib"
export PATH="/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:$PATH"
export PATH="/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:$PATH"
export PATH="/Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:$PATH"
export PATH="/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:$PATH"
```

ivmarkov, 2023-08-23:
Can you try by generating your project in the drive root, i.e. `C:\myproj`?

We are constantly hit by Windows not supporting long-enough paths, and we cannot fix it. But we should fail the build (or warn the user) during the build, that their project path is too long.

ivmarkov, 2023-08-23:
... and possibly ask the users to switch to WSL2 for building on Windows, as annoying as it might be. Including utilizing the native Linux filesystem for the project location (which is a key).

RoyalFoxy, 2023-08-23:
Im on macos...

RoyalFoxy, 2023-08-23:
I just tried to build it regardless within my homes folder which should surely work.

This is the error I ended up with which looks like a different one and one I got too from another test try.

[log.txt](https://github.com/esp-rs/esp-idf-sys/files/12422442/log.txt)

RoyalFoxy, 2023-08-23:
I saw in both this log and the original log this line

`error: unknown target triple 'xtensa', please use -triple or -arch`

ivmarkov, 2023-08-23:
> Im on macos... I dont use windows as my platform of choice for dev anymore

Sorry! I get so much noise around failed builds with just a freshly-generated `esp-idf-template` projects on Windows, that I apparently don't look carefully in the logs anymore. Anyway.

On to your first log:
```
    Include directory
    '/Users/royalfoxy/files/private/programming/esp32/test5/.embuild/espressif/esp-idf/v4.4.5/components/mbedtls/mbedtls/include'
    is not a directory.
```

It seems that somehow the git clone of ESP IDF did result in a partially cloned repo. Are you behind some proxy by any chance? In any case - that's the reason, but why - I have no idea. In any case, this is not normal, but is not a problem in `esp-idf-sys` per-se either. Do you have any global settings in your git tool that do shallow clone by default? Or don't do a proper `git submodule update`?

On to your second log:
```
error: unknown target triple 'xtensa', please use -triple or -arch
```

So... are you trying to build for the ESP32CXX series, or are you trying to buyild for the original ESP32? It seems you are trying to do the latter, as your target seems to be `xtensa-esp32-espidf` and not one of the `riscv32im(a)c-esp-espidf` targets?

For ESP32 (xtensa) you need the `espup` toolchain, as per the README. Moreover, you need the `esp_export.sh` **ran** before the build. I don't like this either, but [people think that's OK](https://github.com/esp-rs/esp-idf-template/issues/130#issuecomment-1662195174).

ivmarkov, 2023-08-23:
> I saw in both this log and the original log this line
> 
> `error: unknown target triple 'xtensa', please use -triple or -arch`

No. It is only in the second log. Check the first log you have attached here. It is a partially cloned git repo.

RoyalFoxy, 2023-08-23:
I tried like 6 different times, 5 of those being std templates so I probably got a log or two confused ^^"

Regarding espup and the esp_export.sh, I have both where espup is just installed and I ran `espup install` and the second is in my shell config and properly appears. And no I don't think my company uses a proxy that would block anything from github... and it may be that submodules don't automatically get initialized when cloning but Im unsure about that.

Oh and yes I am compiling for the original esp32.

RoyalFoxy, 2023-08-23:
Tried doing another `esp install` just to make sure

```
espup install
[2023-08-23T19:46:22Z INFO ] 💽  Installing the Espressif Rust ecosystem
[2023-08-23T19:46:22Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases/latest'
[2023-08-23T19:46:22Z INFO ] 💡  Querying GitHub API: 'https://api.github.com/repos/esp-rs/rust-build/releases'
[2023-08-23T19:46:23Z INFO ] 🔧  Checking Rust installation
[2023-08-23T19:46:23Z INFO ] 🔧  Installing RISC-V targets ('riscv32imc-unknown-none-elf' and 'riscv32imac-unknown-none-elf') for 'nightly' toolchain
[2023-08-23T19:46:23Z WARN ] ⚠️   Previous installation of GCC exists in: '/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208'. Reusing this installation.
[2023-08-23T19:46:23Z WARN ] ⚠️   Previous installation of GCC exists in: '/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208'. Reusing this installation.
[2023-08-23T19:46:23Z WARN ] ⚠️   Previous installation of GCC exists in: '/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208'. Reusing this installation.
[2023-08-23T19:46:23Z WARN ] ⚠️   Previous installation of GCC exists in: '/Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208'. Reusing this installation.
[2023-08-23T19:46:23Z WARN ] ⚠️   Previous installation of LLVM exists in: '/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516'. Reusing this installation.
[2023-08-23T19:46:23Z WARN ] ⚠️   Previous installation of Xtensa Rust 1.71.0.1 exists in: '/Users/royalfoxy/.rustup/toolchains/esp'. Reusing this installation.
[2023-08-23T19:46:24Z INFO ] 🔧  Creating export file
[2023-08-23T19:46:24Z WARN ] 💡  Please, set up the environment variables by running: '. /Users/royalfoxy/export-esp.sh'
[2023-08-23T19:46:24Z WARN ] ⚠️   This step must be done every time you open a new terminal.
[2023-08-23T19:46:24Z INFO ] ✅  Installation successfully completed!
```

and printed the path to the console

```
echo $PATH
/Users/royalfoxy/.bun/bin /Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin /Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin /Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin /Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin /opt/homebrew/bin /opt/homebrew/sbin /Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin /Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin /Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin /Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin /Users/royalfoxy/miniforge3/bin /Users/royalfoxy/bin /Users/RoyalFoxy/bin /Users/RoyalFoxy/.cargo/bin /Users/royalfoxy/.cargo/bin /opt/homebrew/opt/openjdk/bin /usr/local/bin /System/Cryptexes/App/usr/bin /usr/bin /bin /usr/sbin /sbin /var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin /var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin /var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin /Library/Apple/usr/bin /usr/local/MacGPG2/bin /Users/royalfoxy/.bun/bin /Users/royalfoxy/Library/pnpm /Users/royalfoxy/.local/share/nvm/v20.4.0/bin /opt/homebrew/bin /opt/homebrew/sbin
```

The path variable is correct, it's not bash nor zsh. In fish instead of : just plain spaces are used

ivmarkov, 2023-08-23:
> Regarding espup and the esp_export.sh, I have both where espup is just installed and I ran `espup install` and the second is in my shell config and properly appears. 

Fact is, if you see `error: unknown target triple 'xtensa', please use -triple or -arch` it 99.9% means you are using the wrong (system) clang, so your path is somehow not correct.  Since this is your own environment, these problems are difficult to debug, so I would appreciate some help from you too in figuring out where it breaks, by examining the `esp_export.sh` script as to what environment vars it sets and which is the non-working one (should be clang related).

> And no I don't think my company uses a proxy that would block anything from github... and it may be that submodules don't automatically get initialized when cloning but Im unsure about that.

Well it works for you at home, it doesn't work for you at work, what are the chances that the `esp-idf-sys` is broken? It is some sort of flaky networking issue, most likely networking related, isn't it? Also obviously we do `git submodule init` (or whatever the command was), or else it wouldn't work at all, right?

> Oh and yes I am compiling for the original esp32.

Don't want to nit-pick, but I'm asking [because the original description of the bug](https://github.com/esp-rs/esp-idf-sys/issues/234#issue-1863103330) is not mentioning xtensa at all, and only talks about the riscv chips. And your environment (nightly Rust & all) is for these only.

ivmarkov, 2023-08-23:
Hm. What is this? It is not part of the build instructions at all?
> - llvm compiler infrastructure: with this command `brew install llvm

ivmarkov, 2023-08-23:
The content of `export-esp.sh`:
```
export LIBCLANG_PATH="/home/coder/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-15.0.0-20221201/esp-clang/lib"
export PATH="/home/coder/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-2021r2-patch5-8_4_0/xtensa-esp32s2-elf/bin:$PATH"
export PATH="/home/coder/.rustup/toolchains/esp/xtensa-esp32-elf/esp-2021r2-patch5-8_4_0/xtensa-esp32-elf/bin:$PATH"
export PATH="/home/coder/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-2021r2-patch5-8_4_0/xtensa-esp32s3-elf/bin:$PATH"
export PATH="/home/coder/.rustup/toolchains/esp/riscv32-esp-elf/esp-2021r2-patch5-8_4_0/riscv32-esp-elf/bin:$PATH"
```

`export LIBCLANG_PATH="/home/coder/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-15.0.0-20221201/esp-clang/lib"` does not work for you. Not sure why.

RoyalFoxy, 2023-08-23:
Regarding the llvm I installed with brew it's because of this paragraph in the book
[image]

I did have a successful download of the template at work as well but our workspace has a few rather flaky access points and Im at a spot where my laptop can sometimes swtich between them every few seconds...

Ohh well I just installed *everything* the book talked about even the things I didn't need so if I need them later I don't have to go through stuff and look what I already have and what not...

And I'm guessing you saw the export-esp.sh file contents I added in the second comment? I could try using zsh and sourcing the file, then try to do a cargo build, maybe it works and it would give us some insight that it has to do with the fish variables not being declared properly or something even tho I can use them with echo and see that their value is correct...

RoyalFoxy, 2023-08-23:
[image]
I'm speechless...

RoyalFoxy, 2023-08-23:
Sooo I apparantly didn't know that you have to add a special flag to `set` in fish if you want variables that are fully global and not just user based...

I got it running with fish as well by converting this

```fish
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin" $PATH
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin" $PATH
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin" $PATH
set PATH "/Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin" $PATH
set LIBCLANG_PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib"
```

to 

```fish
set -x PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin" $PATH
set -x PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin" $PATH
set -x PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin" $PATH
set -x PATH "/Users/royalfoxy/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin" $PATH
set -x LIBCLANG_PATH "/Users/royalfoxy/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib"
```

the subtle `-x` is important here ._.

RoyalFoxy, 2023-08-23:
Thanks for bringing me onto the right track so I could find out what was causing it! Before I just had a million and one ideas what could be wrong like that I didn't install a proper version of python or llvm or it just didn't compile on apples arm for reason but you pointing out it's likely something with clang path got me thinking that just `set` may not be global. Thanks for the help!

closed 2023-08-24 by ivmarkov