esp-rs/esp-idf-sys #237: Some uart fields and constants do not exist

MCorange99, 2023-08-29:
Im using endavourOS (arch linux based)

While compiling the project i get this error:

```rust
mcorange@eos [18:19:31] [~/@Projects/rust/embedded/minion] [master *]
-> % cargo build
   Compiling esp-idf-hal v0.41.2
   Compiling minion v0.1.0 (/home/mcorange/@Projects/rust/embedded/minion)
error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /home/mcorange/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/uart.rs:1091:25
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                         ^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `UART_NUM_MAX` in this scope
    --> /home/mcorange/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/uart.rs:1091:60
     |
1091 | static REFS: [AtomicU8; UART_NUM_MAX as usize] = [NO_REFS; UART_NUM_MAX as usize];
     |                                                            ^^^^^^^^^^^^ not found in this scope

error[E0560]: struct `esp_idf_sys::uart_config_t` has no field named `source_clk`
   --> /home/mcorange/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.41.2/src/uart.rs:932:9
    |
932 |         source_clk: config.source_clock.into(),
    |         ^^^^^^^^^^ `esp_idf_sys::uart_config_t` does not have this field
    |
    = note: available fields are: `baud_rate`, `data_bits`, `parity`, `stop_bits`, `flow_ctrl` ... and 2 others

Some errors have detailed explanations: E0425, E0560.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `esp-idf-hal` (lib) due to 3 previous errors
```

Heres the package versions if that matters:

```toml
[features]

default = ["std", "hal", "esp-idf-sys/native"]

pio = ["esp-idf-sys/pio"]
all = ["std", "nightly", "experimental", "embassy"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]
std = ["alloc", "esp-idf-sys/std", "esp-idf-sys/binstart", "embedded-svc?/std", "esp-idf-hal?/std", "esp-idf-svc?/std"]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
nightly = ["embedded-svc?/nightly", "esp-idf-svc?/nightly"] # Future: "esp-idf-hal?/nightly"
experimental = ["embedded-svc?/experimental", "esp-idf-svc?/experimental"]
embassy = ["esp-idf-hal?/embassy-sync", "esp-idf-hal?/critical-section", "esp-idf-hal?/edge-executor", "esp-idf-svc?/embassy-time-driver", "esp-idf-svc?/embassy-time-isr-queue"]

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33.2", default-features = false }
esp-idf-hal = { version = "0.41.2", optional = true, default-features = false }
esp-idf-svc = { version = "0.46", optional = true, default-features = false }
embedded-svc = { version = "0.25", optional = true, default-features = false }

[build-dependencies]
embuild = "0.31.2"
```

I have successfully used the normal C toolchain for esp32 (im using esp-wroom-32 dev board)

MCorange99, 2023-08-30:
I have successfully cloned the 3 esp-idf crates and fixed the issues so it compiles:

The fixes:

```C
// /dependencies/esp-idf-sys/src/include/esp-idf/bindings.h
#define ESP_IDF_COMP_DRIVER_ENABLED  //! added on line 1
```

```rs
// /dependencies/esp-idf-hal/src/i2s/pdm.rs
#[cfg(any(esp_idf_soc_i2s_supports_pdm_rx, esp32, esp32s3))]
impl PdmRxClkConfig {
    ...
/// Convert to the ESP-IDF SDK `i2s_pdm_rx_clk_config_t` representation.
#[cfg(not(esp_idf_version_major = "4"))]
#[inline(always)]
pub(super) fn as_sdk(&self) -> i2s_pdm_rx_clk_config_t {
    i2s_pdm_rx_clk_config_t {
        sample_rate_hz: self.sample_rate_hz,
        clk_src: self.clk_src.as_sdk(),
        mclk_multiple: self.mclk_multiple.as_sdk(),
        dn_sample_mode: self.downsample_mode.as_sdk(),
        bclk_div: 8 // no idea if this is correct but i saw it in another file somewhere set as 8
    }
}
```

ivmarkov, 2023-09-08:
What version of ESP IDF are you using?
Also... what versions of `esp-idf-hal` and `esp-idf-svc` are you using? Wondering because I don't think the I2S driver (that you are fixing) was released?

MCorange99, 2023-09-14:
I was using the master branch of esp idf sdk
and the crate versions are 
esp-idf-sys = 0.33.2
esp-idf-hal = 0.41.2
esp-idf-svc = 0.46

ivmarkov, 2023-09-14:
> I was using the master branch of esp idf sdk and the crate versions are esp-idf-sys = 0.33.2 esp-idf-hal = 0.41.2 esp-idf-svc = 0.46

OK. I suggest you wait a week or so for the new crates to be released. It has both the `UART_NUM_MAX` problem addressed (which isn't in the released `esp-idf-hal` 0.41.2), as well as the I2S issue which you fixed in the `master` version of `esp-idf-hal`).

In the interim you can use the unreleased `master` version of all, but you have to add to the list `embedded-svc` as well.

Or just not use the master version of esp idf. I have difficulties imagining why you want to do that.

closed 2023-09-21 by ivmarkov