esp-rs/esp-idf-sys #242: Missing esp_wps.h from the bindings.h

kakko3, 2023-09-23:
Seems like the esp_wps.h is missing from the bindings causing it to be impossible to use WPS connecting for WiFi. I tried [forking the repository](https://github.com/kakko3/esp-idf-sys) and adding the missing header file to bindings. After adding the fork as cargo depencency I get following error?

```
error: failed to select a version for `esp-idf-sys`.
    ... required by package `esp-idf-hal v0.41.2`
    ... which satisfies dependency `esp-idf-hal = "^0.41.2"` (locked to 0.41.2) of package `esp32 v0.1.0 (C:\Coding\esp32)`
versions that meet the requirements `^0.33` (locked to 0.33.2) are: 0.33.2

the package `esp-idf-sys` links to the native library `esp_idf`, but it conflicts with a previous package which links to `esp_idf` as well:
package `esp-idf-sys v0.33.2 (C:\Coding\esp32\esp-idf-sys)`
    ... which satisfies path dependency `esp-idf-sys` of package `esp32 v0.1.0 (C:\Coding\esp32)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the links ='esp-idf-sys' value. For more information, see https://doc.rust-lang.o
rg/cargo/reference/resolver.html#links.

failed to select a version for `esp-idf-sys` which could resolve this conflict
```

Probably doing something wrong as this build stuff is way above my understanding but is there some way to add these bindings manually and have the cargo build succedd?

Vollbrecht, 2023-09-24:
Check out how to correcly use a local or forked git repo. Infos can be found https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#testing-a-bugfix . Make sure you patch esp-idf-sys like this in your Cargo.toml

kakko3, 2023-09-24:
Thanks a lot! That seems to do it. Have to delve into that cargo manual more, such a nice tool.

closed 2023-10-03 by ivmarkov