esp-rs/esp-idf-sys #245: custom build command failed, "The C compiler is not able to compile a simple test program"

TheOnAndOnlyZenomat, 2023-10-03:
Hey,
I have been trying to compile a std program for my esp32-c3. I generate the project with:
1. `cargo generate`
2. name: std
3. mcu: esp32c3
4. advanced: false

Then I just ran `cargo run`

I already tried different esp-idf version, also running it from the environment and went through `espup`. I am happy to try each of these steps again or anything else you need me to do to debug this.
I had to cut some of the log, otherwise the comment would have been to long. I can provide the full log on request.

```
.....
The following warnings were emitted during compilation:

warning: Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

error: failed to run custom build command for `esp-idf-sys v0.33.2`

Caused by:
  process didn't exit successfully: `/home/zeno/code/esp32-riscv/std/target/debug/build/esp-idf-sys-e3a6e9c71127f9ed/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:warning=Configuring first supported MCU 'esp32c3' derived from the build target 'riscv32imc-esp-espidf' supporting MCUs [esp32c3, esp32c2, esp32h2, esp32c5]; explicitly specify an MCU to resolve this ambiguity

  PATH=/home/zeno/code/esp32-riscv/std/.embuild/espressif/esp-idf/v4.4.5/tools:$PATH
  pip 23.2.1 from /usr/lib/python3.11/site-packages/pip (python 3.11)
  created virtual environment CPython3.11.5.final.0-64 in 2409ms
    creator CPython3Posix(dest=/home/zeno/code/esp32-riscv/std/.embuild/espressif/python_env/idf4.4_py3.11_env, clear=False, no_vcs_ignore=False, global=False)
    seeder PipInvoke(download=False, pip=bundle, setuptools=bundle, wheel=bundle)
      added seed packages: pip==23.2.1, setuptools==68.2.0, wheel==0.41.2
    activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring gdbgui: markers 'python_version < "3.11"' don't match your environment
  Ignoring pygdbmi: markers 'python_version < "3.11"' don't match your environment
  Ignoring python-socketio: markers 'python_version < "3.11"' don't match your environment
  Ignoring jinja2: markers 'python_version < "3.11"' don't match your environment
  Ignoring itsdangerous: markers 'python_version < "3.11"' don't match your environment
  Ignoring None: markers 'sys_platform == "win32"' don't match your environment
  Requirement already satisfied: setuptools>=21 in /home/zeno/code/esp32-riscv/std/.embuild/espressif/python_env/idf4.4_py3.11_env/lib/python3.11/site-packages (from -r /home/zeno/code/esp32-riscv/std/.embuild/espressif/esp-idf/v4.4.5/requirements.txt (line 4)) (68.2.0)
  Current system platform: linux-amd64
  Installing riscv32-esp-elf@esp-2021r2-patch5-8.4.0
  Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch5/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch5-linux-amd64.tar.gz
  Destination: /home/zeno/code/esp32-riscv/std/.embuild/espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch5-linux-amd64.tar.gz.tmp
Done
  Extracting /home/zeno/code/esp32-riscv/std/.embuild/espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch5-linux-amd64.tar.gz to /home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0
  Installing cmake@3.23.1
  Downloading https://github.com/Kitware/CMake/releases/download/v3.23.1/cmake-3.23.1-linux-x86_64.tar.gz
  Destination: /home/zeno/code/esp32-riscv/std/.embuild/espressif/dist/cmake-3.23.1-linux-x86_64.tar.gz.tmp
Done
  Extracting /home/zeno/code/esp32-riscv/std/.embuild/espressif/dist/cmake-3.23.1-linux-x86_64.tar.gz to /home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/cmake/3.23.1
  Installing ninja@1.10.2
  Downloading https://dl.espressif.com/dl/ninja-1.10.2-linux64.tar.gz
  Destination: /home/zeno/code/esp32-riscv/std/.embuild/espressif/dist/ninja-1.10.2-linux64.tar.gz.tmp
Done
  Extracting /home/zeno/code/esp32-riscv/std/.embuild/espressif/dist/ninja-1.10.2-linux64.tar.gz to /home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/ninja/1.10.2
  IDF_PYTHON_ENV_PATH=/home/zeno/code/esp32-riscv/std/.embuild/espressif/python_env/idf4.4_py3.11_env
  PATH=/home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:/home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/cmake/3.23.1/bin:/home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/ninja/1.10.2/:/home/zeno/code/esp32-riscv/std/.embuild/espressif/python_env/idf4.4_py3.11_env/bin:/home/zeno/code/esp32-riscv/std/.embuild/espressif/esp-idf/v4.4.5/tools:$PATH
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
[... 449 lines cut ...]
      Change Dir: /home/zeno/code/esp32-riscv/std/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-9640a82b5d01ae28/out/build/CMakeFiles/CMakeTmp
      
      Run Build Command(s):/home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/ninja/1.10.2/ninja cmTC_b747a && [1/2] Building C object CMakeFiles/cmTC_b747a.dir/testCCompiler.c.obj
      FAILED: CMakeFiles/cmTC_b747a.dir/testCCompiler.c.obj 
      /home/zeno/code/esp32-riscv/std/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc   -march=rv32imc  -ffunction-sections -fdata-sections -mabi=ilp32 -mcmodel=medany -pipe -march=native -o CMakeFiles/cmTC_b747a.dir/testCCompiler.c.obj -c /home/zeno/code/esp32-riscv/std/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-9640a82b5d01ae28/out/build/CMakeFiles/CMakeTmp/testCCompiler.c
      Assembler messages:
      Fatal error: -march=native: ISA string must begin with rv32 or rv64
      cc1: error: -march=native: ISA string must begin with rv32 or rv64
      ninja: build stopped: subcommand failed.
      
      

    

    CMake will not be able to correctly generate this project.
  Call Stack (most recent call first):
    CMakeLists.txt:6 (project)

  thread 'main' panicked at /home/zeno/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ivmarkov, 2023-10-03:
Please run `set` in your command prompt and look for `CCFLAGS`, `CXXFLAGS` and similar. Unset **ALL** of those.

ivmarkov, 2023-10-03:
Please let me know if this was the culprit. If that's the case, we'll likely implement functionality which by default stops inheriting these environment variables from the user environment (even though the user is **not**  supposed to define these globally), as this is only causing headaches.

TheOnAndOnlyZenomat, 2023-10-03:
Yep, that was the problem, thank you very much. I would have never gotten to that myself

closed 2023-10-03 by TheOnAndOnlyZenomat

ivmarkov, 2023-10-03:
https://github.com/esp-rs/esp-idf-sys/commit/849149bd420165daf049c51f982dbae104fa9f97