esp-rs/esp-idf-sys #248: error: failed to run custom build command for `esp-idf-sys v0.33.2`

ds2k5, 2023-10-13:
Hi,
using Debian 12 on x86_64

did following:

$ rustup toolchain install nightly --component rust-src
$ cargo install cargo-generate
$ cargo install ldproxy
$ cargo install espflash
$ cargo install cargo-espflash

$ cargo generate --git https://github.com/esp-rs/esp-idf-template cargo

$ cargo build

```
   Compiling embassy-futures v0.1.0
   Compiling esp-idf-sys v0.33.2
   Compiling esp-idf-hal v0.41.2
   Compiling esp-idf-svc v0.46.2
   Compiling test3 v0.1.0 (/home/developer/rust/test3)
   Compiling embedded-svc v0.25.3
error: failed to run custom build command for `esp-idf-sys v0.33.2`

Caused by:
  process didn't exit successfully: `/home/developer/rust/test3/target/debug/build/esp-idf-sys-1bb90064ef145193/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  Submodule path 'components/asio/asio': checked out 'f31694c9f1746ba189a4bcae2e34db15135ddb22'
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
  Submodule path 'components/bt/controller/lib_esp32': checked out '7bb0d445db3414ce96d21c50ba9249125d41480f'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '0cfac1b21ebc995e8e9aa040ab1ab29deee4f580'
  Submodule path 'components/bt/host/nimble/nimble': checked out 'e2d7a766fb3927d008902611b3985e8595864c55'
  Submodule path 'components/cbor/tinycbor': checked out '7c349dbb6b8d76db39383b226d3ebdf59b8ab37d'
  Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
  Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
  Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
  Submodule path 'components/coap/libcoap': checked out '3aa11612c143c9734d72022720f33e12506f7a2c'
  Submodule path 'components/coap/libcoap/ext/tinydtls': checked out '59055b8a935bc53bf69d002fc089ad4bd08851b2'
  Submodule path 'components/esp_phy/lib': checked out '32583f98f229ab9b3c48aa3a10f954fcdddfe4d1'
  Submodule path 'components/esp_wifi/lib': checked out 'a9f72f895dd74bb763f1ff9146203ff57846ab5d'
  Submodule path 'components/esptool_py/esptool': checked out '7b17ea072fbac0f92fc417ddbe34e28afbd8ced0'
  Submodule path 'components/expat/expat': checked out '454c6105bc2d0ea2521b8f8f7a5161c2abd8c386'
  Submodule path 'components/ieee802154/lib': checked out 'f7b5e8059a3bb6f321e79ac3bf2aa4d2a9b93326'
  Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
  Submodule path 'components/libsodium/libsodium': checked out '4f5e89fa84ce1d178a6765b8b46f2b6f91216677'
  Submodule path 'components/lwip/lwip': checked out '4f24c9baf9101634b7c690802f424b197b3bb685'
  Submodule path 'components/mbedtls/mbedtls': checked out '1d7033af30e20ccb2a0c0a114d3a08e372430f34'
  Submodule path 'components/mqtt/esp-mqtt': checked out 'bb9c8af9d552b608dd3aabf9617bde757a538ebe'
  Submodule path 'components/nghttp/nghttp2': checked out '8f7b008b158e12de0e58247afd170f127dbb6456'
  Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '7c91efc1ffda769a5f1a872c646c82b00698f1b8'
  Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'b967ca054f48a36f82d8fcdd32e54ec5144f2751'
  Submodule path 'components/openthread/lib': checked out '9a8d34d8f698cad2c9468468b473e26a3dda51b9'
  Submodule path 'components/openthread/openthread': checked out 'c36c0e77a2465355bcf13bd7dc718d8c9aa6ff64'
  Submodule path 'components/protobuf-c/protobuf-c': checked out 'abc67a11c6db271bedbb9f58be85d6f4e2ea8389'
  Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
  Submodule path 'components/tinyusb/tinyusb': checked out 'c4badd394eda18199c0196ed0be1e2d635f0a5f6'
  Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
  Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
  Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out '36d0642e66ff5b1c7a291873f24c498ca6ffedef'
[... 190 lines cut ...]
    File "/home/developer/rust/test3/.embuild/espressif/esp-idf/v4.4.6/tools/idf_tools.py", line 1594, in action_install_python_env
      import virtualenv  # noqa: F401
      ^^^^^^^^^^^^^^^^^
  ModuleNotFoundError: No module named 'virtualenv'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/home/developer/rust/test3/.embuild/espressif/esp-idf/v4.4.6/tools/idf_tools.py", line 2003, in <module>
      main(sys.argv[1:])
    File "/home/developer/rust/test3/.embuild/espressif/esp-idf/v4.4.6/tools/idf_tools.py", line 1999, in main
      action_func(args)
    File "/home/developer/rust/test3/.embuild/espressif/esp-idf/v4.4.6/tools/idf_tools.py", line 1597, in action_install_python_env
      subprocess.check_call([sys.executable, '-m', 'pip', 'install', '--user', 'virtualenv'],
    File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1.
  Error: Could not install esp-idf

  Caused by:
      command 'IDF_TOOLS_PATH="/home/developer/rust/test3/.embuild/espressif" "python3" "/home/developer/rust/test3/.embuild/espressif/esp-idf/v4.4.6/tools/idf_tools.py" "--idf-path" "/home/developer/rust/test3/.embuild/espressif/esp-idf/v4.4.6" "--non-interactive" "install-python-env"' exited with non-zero status code 1

```

How can I fix this ?

Vollbrecht, 2023-10-13:
This is a debian specific problem. You can't rawly install python packages via pip if its not in a venv or through apt, they enforced this now in debian 12. The esp-idf installtation try's to install virtualenv if its not found, but this failed because its only allowed inside a virtualenv. But because the virtualenv is not created you got this error. 
Solution: `apt install python3-virtualenv`

ds2k5, 2023-10-13:
Hi @Vollbrecht,
thanks for the hint

did the following:

$ sudo apt-get install python3-virtualenv
$ cargo clean
$ rm -rf .embuild
$ cargo build
```

  Successfully created esp32 image.
  Generated /home/developer/rust/test23/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-d0b03021d1cb46d6/out/build/libespidf.bin
  [884/884] cd /home/developer/rust/test23/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-d0b03021d1cb46d6/out/build/esp-idf/esptool_py && /home/developer/rust/test23/.embuild/espressif/python_env/idf5.1_py3.11_env/bin/python /home/developer/rust/test23/.embuild/espressif/esp-idf/v5.1.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/developer/rust/test23/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-d0b03021d1cb46d6/out/build/partition_table/partition-table.bin /home/developer/rust/test23/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-d0b03021d1cb46d6/out/build/libespidf.bin
  libespidf.bin binary size 0x273d0 bytes. Smallest app partition is 0x100000 bytes. 0xd8c30 bytes (85%) free.
  cargo:root=/home/developer/rust/test23/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-d0b03021d1cb46d6/out
  cargo:rerun-if-changed=/home/developer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.2/src/include/esp-idf/bindings.h
  cargo:rustc-env=EMBUILD_GENERATED_BINDINGS_FILE=/home/developer/rust/test23/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-d0b03021d1cb46d6/out/bindings.rs

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.1.1") }
  Cloning into '/home/developer/rust/test23/.embuild/espressif/esp-idf/v5.1.1'...
  Note: switching to 'e088c3766ba440e72268b458a68f27b6e7d63986'.

  You are in 'detached HEAD' state. You can look around, make experimental
  changes and commit them, and you can discard any commits you make in this
  state without impacting any branches by switching back to a branch.

  If you want to create a new branch to retain commits you create, you may
  do so (now or later) by using -c with the switch command. Example:

    git switch -c <new-branch-name>

  Or undo this operation with:

    git switch -

  Turn off this advice by setting config variable advice.detachedHead to false

  Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
  Submodule 'components/bt/controller/lib_esp32' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib_esp32'
  Submodule 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib' (https://github.com/espressif/esp32c2-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib'
  Submodule 'components/bt/controller/lib_esp32c3_family' (https://github.com/espressif/esp32c3-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c3_family'
  Submodule 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib' (https://github.com/espressif/esp32c6-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
  Submodule 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib' (https://github.com/espressif/esp32h2-bt-lib.git) registered for path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib'
  Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
  Submodule 'components/cmock/CMock' (https://github.com/ThrowTheSwitch/CMock.git) registered for path 'components/cmock/CMock'
  Submodule 'components/esp_coex/lib' (https://github.com/espressif/esp-coex-lib.git) registered for path 'components/esp_coex/lib'
  Submodule 'components/esp_phy/lib' (https://github.com/espressif/esp-phy-lib.git) registered for path 'components/esp_phy/lib'
  Submodule 'components/esp_wifi/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
  Submodule 'components/heap/tlsf' (https://github.com/espressif/tlsf.git) registered for path 'components/heap/tlsf'
[... 68 lines cut ...]
  From https://github.com/espressif/esp-mqtt
   * branch            dffabb067fb3c39f486033d2e47eb4b1416f0c82 -> FETCH_HEAD
  From https://github.com/espressif/esp-thread-lib
   * branch            c9c17a0b6634af016111c9817f0be1fe0e7112f6 -> FETCH_HEAD
  From https://github.com/espressif/openthread
   * branch            5beae143700db54c6e9bd4b15a568abe2f305723 -> FETCH_HEAD
  From https://github.com/pellepl/spiffs
   * branch            0dbb3f71c5f6fae3747a9d935372773762baf852 -> FETCH_HEAD
  From https://github.com/ThrowTheSwitch/Unity
   * branch            7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 -> FETCH_HEAD
  Using esp-idf v5.1.1 at '/home/developer/rust/test23/.embuild/espressif/esp-idf/v5.1.1'
  Compiler supported targets: xtensa-esp32-elf

[... cut ...]
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at /home/developer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/ir/context.rs:530:15:
  libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Vollbrecht, 2023-10-13:
did you install via espup ? make sure your env variables are not poluted and that you source the export script `export-esp.sh`

ds2k5, 2023-10-13:
>did you install via espup ?

what packages ?

ds2k5, 2023-10-13:
$ espup install
$ chmod u+x ~/export.sh
$ ~/export.sh
$ cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
$ cd project1
$ cargo build

....
error as above

Vollbrecht, 2023-10-13:
thats not how you source env file on debian - please also check out https://github.com/esp-rs/esp-idf-hal/issues/319 for more info regarding this specific problem

ds2k5, 2023-10-13:
thanks - will check tomorrow

ivmarkov, 2023-10-13:
> $ chmod u+x ~/export-esp.sh
> ~/export-esp.sh

Yeah, to cut some chasing in the (long) other thread @Vollbrecht  mentioned, you need to do, **exactly**:
`. ~/export-esp.sh`

And I'm so sick and tired of this problem :(

ivmarkov, 2023-10-13:
@SergioGasquez **Enjoy your vacation**, but maybe another nail on the "export" script. :)
We need the symlink approach, I think.

closed 2023-10-13 by ivmarkov

ds2k5, 2023-10-15:
$ . ~/export-esp.sh
$ caro clean
$ rm -rf .embuild
$ cargo build

   Compiling esp-idf-svc v0.46.2
   Compiling embedded-can v0.4.1
   Compiling no-std-net v0.5.0
   Compiling embassy-time v0.1.4
   Compiling project1 v0.1.0 (/home/developer/rust/project1)
   Compiling embassy-futures v0.1.0
   Compiling embedded-svc v0.25.3
    Finished dev [optimized + debuginfo] target(s) in 4m 03s

thanks a lot to @ALL !

@ivmarkov 
> And I'm so sick and tired of this problem :(

Sorry about that!