esp-rs/esp-idf-sys #249: Fresh install of Ubuntu 22.04 LTS - CMake error Some Python dependencies must be installed.

sjames, 2023-10-15:
Hi,

I followed the esp-rs build instructions and have installed the necessary dependencies on a fresh Ubuntu 22.04 LTS install.  The build for esp-idf-sys fails with...

```
 -- Checking Python dependencies...
  -- Configuring incomplete, errors occurred!
  See also "/home/sojan/playground/ccontroller/target/riscv32imac-esp-espidf/debug/build/esp-idf-sys-184886213eed6816/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.1.1") }
  Using esp-idf v5.1.1 at '/home/sojan/playground/ccontroller/.embuild/espressif/esp-idf/v5.1.1'
  ERROR: /home/sojan/playground/ccontroller/.embuild/espressif/espidf.constraints.v5.1.txt doesn't exist. Perhaps you've forgotten to run the install scripts. Please check the installation guide for more information.
  CMake Error at /home/sojan/playground/ccontroller/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/build.cmake:363 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /home/sojan/playground/ccontroller/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/build.cmake:498 (__build_check_python)
    CMakeLists.txt:16 (idf_build_process)

  thread 'main' panicked at /home/sojan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

```
I'm unable to figure out which python dependency needs to be installed.  

**Info:**
Python version: Python 3.10.12
Ubuntu 22.04.3 LTS
ESP_IDF_VERSION = "v5.1.1"

Vollbrecht, 2023-10-15:
make sure your system has `python3 python3-pip python3-venv` installed on your system.

sjames, 2023-10-15:
Hi @Vollbrecht. Thank you for the reply.

I did check that these three are installed. 

Here is a Docker file with which I can reproduce the issue.  The build breaks if you change ubuntu:20.04 to ubuntu:22.04

```
# There are a few Dockerfile restrictions when using Github Actions
# See: https://docs.github.com/en/actions/creating-actions/dockerfile-support-for-github-actions

#FROM debian:bullseye-slim
FROM ubuntu:20.04
ENV DEBIAN_FRONTEND=noninteractive
ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8
ARG CONTAINER_USER=esp
ARG CONTAINER_GROUP=esp
ARG NIGHTLY_VERSION=nightly-2023-02-28
ARG ESP_IDF_VERSION=v5.1.1
ARG ESP_BOARD=esp32c6

RUN apt-get update \
    && apt-get install -y git curl ninja-build llvm-dev libclang-dev \
    clang unzip python3 python3-pip pkg-config libssl-dev \
    && apt-get install python3-venv cmake sudo wget vim -y \
    && apt-get clean -y \
    && rm -rf /var/lib/apt/lists/* /tmp/library-scripts

RUN wget https://github.com/Kitware/CMake/releases/download/v3.24.1/cmake-3.24.1-Linux-x86_64.sh \
	  -q -O /tmp/cmake-install.sh \
	&& chmod u+x /tmp/cmake-install.sh \
	&& sudo mkdir /opt/cmake-3.24.1 \
	&& sudo /tmp/cmake-install.sh --skip-license --prefix=/opt/cmake-3.24.1 \
	&& rm /tmp/cmake-install.sh \
	&& sudo ln -s /opt/cmake-3.24.1/bin/* /usr/local/bin

RUN adduser --disabled-password --gecos "" ${CONTAINER_USER}
USER ${CONTAINER_USER}
WORKDIR /home/${CONTAINER_USER}

RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- \
    --default-toolchain ${NIGHTLY_VERSION} -y --profile minimal \
    --component rust-src,clippy

ENV PATH=${PATH}:$HOME/.cargo/bin

RUN ARCH=$($HOME/.cargo/bin/rustup show | grep "Default host" | sed -e 's/.* //') && \
    curl -L "https://github.com/esp-rs/espflash/releases/latest/download/cargo-espflash-${ARCH}.zip" -o "${HOME}/.cargo/bin/cargo-espflash.zip" && \
    unzip "${HOME}/.cargo/bin/cargo-espflash.zip" -d "${HOME}/.cargo/bin/" && \
    rm "${HOME}/.cargo/bin/cargo-espflash.zip" && \
    chmod u+x "${HOME}/.cargo/bin/cargo-espflash" && \
    curl -L "https://github.com/esp-rs/espflash/releases/latest/download/espflash-${ARCH}.zip" -o "${HOME}/.cargo/bin/espflash.zip" && \
    unzip "${HOME}/.cargo/bin/espflash.zip" -d "${HOME}/.cargo/bin/" && \
    rm "${HOME}/.cargo/bin/espflash.zip" && \
    chmod u+x "${HOME}/.cargo/bin/espflash" && \
    curl -L "https://github.com/esp-rs/esp-web-flash-server/releases/latest/download/web-flash-${ARCH}.zip" -o "${HOME}/.cargo/bin/web-flash.zip" && \
    unzip "${HOME}/.cargo/bin/web-flash.zip" -d "${HOME}/.cargo/bin/" && \
    rm "${HOME}/.cargo/bin/web-flash.zip" && \
    chmod u+x "${HOME}/.cargo/bin/web-flash" && \
    curl -L "https://github.com/esp-rs/embuild/releases/latest/download/ldproxy-${ARCH}.zip" -o "${HOME}/.cargo/bin/ldproxy.zip" &&  \
    unzip "${HOME}/.cargo/bin/ldproxy.zip" -d "${HOME}/.cargo/bin/" && \
    rm "${HOME}/.cargo/bin/ldproxy.zip" && \
    chmod u+x "${HOME}/.cargo/bin/ldproxy" && \
    GENERATE_VERSION=$(git ls-remote --refs --sort="version:refname" --tags "https://github.com/cargo-generate/cargo-generate" | cut -d/ -f3- | tail -n1) &&  \
    GENERATE_URL="https://github.com/cargo-generate/cargo-generate/releases/latest/download/cargo-generate-${GENERATE_VERSION}-${ARCH}.tar.gz" &&  \
    curl -L "${GENERATE_URL}" -o "${HOME}/.cargo/bin/cargo-generate.tar.gz" &&  \
    tar xf "${HOME}/.cargo/bin/cargo-generate.tar.gz" -C "${HOME}/.cargo/bin/" &&  \
    rm "${HOME}/.cargo/bin/cargo-generate.tar.gz" && \
    chmod u+x "${HOME}/.cargo/bin/cargo-generate"

RUN mkdir -p ${HOME}/.espressif/frameworks/ \
    && git clone --branch ${ESP_IDF_VERSION} -q --depth 1 --shallow-submodules \
    --recursive https://github.com/espressif/esp-idf.git \
    ${HOME}/.espressif/frameworks/esp-idf \
    && python3 ${HOME}/.espressif/frameworks/esp-idf/tools/idf_tools.py install cmake \
    && ${HOME}/.espressif/frameworks/esp-idf/install.sh ${ESP_BOARD} \
    && rm -rf .espressif/dist \
    && rm -rf .espressif/frameworks/esp-idf/docs \
    && rm -rf .espressif/frameworks/esp-idf/examples \
    && rm -rf .espressif/frameworks/esp-idf/tools/esp_app_trace \
    && rm -rf .espressif/frameworks/esp-idf/tools/test_idf_size

ENV IDF_TOOLS_PATH=${HOME}/.espressif
RUN echo "source ${HOME}/.espressif/frameworks/esp-idf/export.sh > /dev/null 2>&1" >> ~/.bashrc

CMD "/bin/bash"

```

ivmarkov, 2023-10-16:
>RUN mkdir -p ${HOME}/.espressif/frameworks/ \
    && git clone --branch ${ESP_IDF_VERSION} -q --depth 1 --shallow-submodules \
    --recursive https://github.com/espressif/esp-idf.git \
    ${HOME}/.espressif/frameworks/esp-idf \
    && python3 ${HOME}/.espressif/frameworks/esp-idf/tools/idf_tools.py install cmake \
    && ${HOME}/.espressif/frameworks/esp-idf/install.sh ${ESP_BOARD} \
    && rm -rf .espressif/dist \
    && rm -rf .espressif/frameworks/esp-idf/docs \
    && rm -rf .espressif/frameworks/esp-idf/examples \
    && rm -rf .espressif/frameworks/esp-idf/tools/esp_app_trace \
    && rm -rf .espressif/frameworks/esp-idf/tools/test_idf_size
>
> ENV IDF_TOOLS_PATH=${HOME}/.espressif
> RUN echo "source ${HOME}/.espressif/frameworks/esp-idf/export.sh > /dev/null 2>&1" >> ~/.bashrc

Why are you doing this?

sjames, 2023-10-16:
I just reused the Dockerfile from one of the Rust-on ESP32 training material.   I face the issue on my Ubuntu install and just wanted to try and replicate it in a Docker environment.

sjames, 2023-10-18:
@Vollbrecht .  My mistake.  I had a stale .embuild folder in my build.  Removing this and rebuilding solved the problem.

closed 2023-10-18 by ivmarkov