esp-rs/esp-idf-sys #250: error: -march=: ISA string must begin with rv32 or rv64

sayanee, 2023-10-16:
I'm a beginner in Rust embedded. I'm trying to `cargo build` a [simple blinky program on ESP32-C3](https://github.com/shanemmattner/ESP32-C3_Rust_Tutorials/blob/main/Tutorials/p0-output/src/main.rs) (Thanks to Shane Mattner).

## Pre-requisites

I'm on macOS 12.7 with Python `3.12.0`.

I used [esp-idf-template](https://github.com/esp-rs/esp-idf-template#prerequisites) to generate the initial boilerplate. 

```
👨 ››› ~/Desktop
$ cargo generate esp-rs/esp-idf-template cargo
⚠️   Favorite `esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template.git
🤷   Project Name: blinky
🔧   Destination: /Users/sayanee/Desktop/blinky ...
🔧   project-name: blinky ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c3
✔ 🤷   Configure advanced template options? · false
🔧   Moving generated files into: `/Users/sayanee/Desktop/blinky`...
🔧   Initializing a fresh Git repository
✨   Done! New project created /Users/sayanee/Desktop/blinky
```

## Code 

<details>
<summary>The blinky code in `src/main.rs` </summary>

```rs
use esp_idf_hal::{delay::FreeRtos, gpio::PinDriver, peripherals::Peripherals};
use esp_idf_sys as _;
use esp_println::println;

fn main() {
    esp_idf_sys::link_patches();

    let peripherals = Peripherals::take().unwrap();
    let mut led_pin = PinDriver::output(peripherals.pins.gpio8).unwrap();

    loop {
        led_pin.set_low().unwrap();
        println!("LED ON");
        FreeRtos::delay_ms(1000);

        led_pin.set_high().unwrap();
        println!("LED OFF");
        FreeRtos::delay_ms(1000);
    }
}
```
</details>

## Config 

<details>
<summary>`Cargo.toml`</summary>

```toml
[package]
name = "blinky"
version = "0.1.0"
authors = ["Sayanee <sayanee@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.66"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
esp-idf-sys = { version = "0.32", features = ["binstart"] }
esp-idf-hal = "0.40"
esp-println = { version = "0.3.1", features = ["esp32c3"] }

[build-dependencies]
embuild = "0.30.4"
```
</details>

<details> 
<summary>And `.cargo/config.toml` has:</summary>

```toml
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
runner = "espflash --monitor"
rustflags = ["-C", "default-linker-libraries"]
```
</details>

## Error

<details>
<summary>My error when I run `cargo build`:</summary>

```
 /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
          /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: failed to merge target specific data of file /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/lib/rv32imc/ilp32/no-rtti/libc.a(lib_a-atexit.o)
          /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
          /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: failed to merge target specific data of file /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/lib/rv32imc/ilp32/no-rtti/libc.a(lib_a-__atexit.o)
          collect2: error: ld returned 1 exit status

          Stack backtrace:
             0: anyhow::backtrace::capture::Backtrace::capture
             1: anyhow::error::<impl anyhow::Error>::msg
             2: ldproxy::main
             3: std::sys_common::backtrace::__rust_begin_short_backtrace
             4: std::rt::lang_start::{{closure}}
             5: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/core/src/ops/function.rs:284:13
                std::panicking::try::do_call
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:502:40
                std::panicking::try
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:466:19
                std::panic::catch_unwind
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panic.rs:142:14
                std::rt::lang_start_internal::{{closure}}
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/rt.rs:148:48
                std::panicking::try::do_call
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:502:40
                std::panicking::try
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:466:19
                std::panic::catch_unwind
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panic.rs:142:14
                std::rt::lang_start_internal
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/rt.rs:148:20
             6: _main

error: could not compile `blinky` (bin "blinky") due to previous error
```
</details>

I'm really not sure what I am doing wrong. Apologies if this is not the right place to post this issue, but I thought the rust lang forum would be too generic for this error.

MabezDev, 2023-10-16:
Thanks for the bug report!

Do you have a value for `ESP_IDF_VERSION` inside `.cargo/config.toml`?

Vollbrecht, 2023-10-16:
can you also please clarify what prompts you choose when using `cargo generate esp-rs/esp-idf-template cargo` Because if you are just generated the project (in the couple last days), it should not have this dependency versions in it you are showing us inside your Cargo.toml

sayanee, 2023-10-17:
Thanks everyone for your replies!

---

> Can you also please clarify what prompts you choose when using `cargo generate esp-rs/esp-idf-template cargo`

I regenerated the project with `cargo generate esp-rs/esp-idf-template cargo` and the simplest options:

- Project Name: blinky
- Which MCU to target? ›esp32c3
- Configure advanced template options? ›false
---
> Do you have a value for `ESP_IDF_VERSION` inside `.cargo/config.toml`?

The generated `.cargo/config.toml` has `ESP_IDF_VERSION = "v4.4.6"`

---
> You can also try `cargo update` in the root of your binary crate.

I ran `cargo update && cargo clean && cargo build` in the folder `blinky` if that's what you mean by "root of your binary crate". The error is the same.

--- 

As a side note, I did manage to `cargo build` successfully before reformatting my MacBook, so I suspect I'm indeed missing some prerequisites.

Also, `cargo build` fails at the very last step, so maybe it's link / `ldproxy` issue?

```
 Compiling embedded-can v0.4.1
 Compiling blinky v0.1.0 (/Users/sayanee/Desktop/blinky)
 Compiling esp-idf-sys v0.32.1
 Compiling esp-idf-hal v0.40.1
 Building [=======================> ] 162/168: esp-idf-sys(build)                                                                  
 Building [=======================> ] 167/168: blinky(bin)
```

I tried the[ original code from Shan Mattner](https://github.com/shanemmattner/ESP32-C3_Rust_Tutorials/tree/main/Tutorials/p0-output) with just `cargo build`. Same error.

It must be something really simple 😅

sayanee, 2023-10-17:
Ok I'm attaching more error logs, because it was a long one:

- `error: linking with ldproxy failed: exit status: 1`
-   `Mis-matched ISA version for 'i' extension. 2.0 vs 2.1`
- `-march=: ISA string must begin with rv32 or rv64`

<details>
<summary> Error log details</summary>

```
 Compiling void v1.0.2
   Compiling embedded-hal v0.2.7
   Compiling embedded-hal-nb v1.0.0-alpha.1
   Compiling embedded-can v0.4.1
   Compiling blinky v0.1.0 (/Users/sayanee/Desktop/blinky)
   Compiling esp-idf-sys v0.32.1
   Compiling esp-idf-hal v0.40.1
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/Users/sayanee/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/bin:/Users/sayanee/.rvm/gems/ruby-3.2.2/bin:/Users/sayanee/.rvm/gems/ruby-3.2.2@global/bin:/Users/sayanee/.rvm/rubies/ruby-3.2.2/bin:/Users/sayanee/.pyenv/shims:/Users/sayanee/.pyenv/versions/3.12.0/lib/python3.12/site-packages:/Users/sayanee/.rvm/bin:/Users/sayanee/.gem/bin:/Users/sayanee/.oh-my-zsh/custom/plugins/git-open:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/sayanee/.cargo/bin:/Users/sayanee/Documents/scripts:/Users/sayanee/Library/Python/3.9/bin:/Users/sayanee/.rvm/bin:/opt/X11/bin:/usr/local/sbin"
...
 = note: [ldproxy] Running ldproxy
          Error: Linker /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: error: esp-idf/esp_ringbuf/libesp_ringbuf.a(ringbuf.c.obj): Mis-matched ISA version for 'i' extension. 2.0 vs 2.1

...

 /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
          /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: failed to merge target specific data of file /Users/sayanee/Desktop/blinky/.embuild/espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/lib/rv32imc/ilp32/no-rtti/libc.a(lib_a-__atexit.o)
          collect2: error: ld returned 1 exit status

          Stack backtrace:
             0: anyhow::backtrace::capture::Backtrace::capture
             1: anyhow::error::<impl anyhow::Error>::msg
             2: ldproxy::main
             3: std::sys_common::backtrace::__rust_begin_short_backtrace
             4: std::rt::lang_start::{{closure}}
             5: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/core/src/ops/function.rs:284:13
                std::panicking::try::do_call
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:502:40
                std::panicking::try
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:466:19
                std::panic::catch_unwind
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panic.rs:142:14
                std::rt::lang_start_internal::{{closure}}
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/rt.rs:148:48
                std::panicking::try::do_call
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:502:40
                std::panicking::try
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:466:19
                std::panic::catch_unwind
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panic.rs:142:14
                std::rt::lang_start_internal
                       at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/rt.rs:148:20
             6: _main

error: could not compile `blinky` (bin "blinky") due to previous error
```
</details>

ivmarkov, 2023-10-17:
The current released version of `esp-idf-sys` is 0.33.2, you are running 0.32.1. There is no way you can link your project with the newer rustc you are using and this older version of your sys crate. Update its version, as well as the versions of the embedded-svc, hal and svc crates and rebuild. `cargo update` only works for automatic updates across patch releases; your crates are older than that.

closed 2023-10-17 by ivmarkov