esp-rs/esp-idf-sys #252: Error: Too long output directory

achieve-dream1221, 2023-10-19:
create a project by ` cargo generate esp-rs/esp-idf-template cargo`
and then, run `cargo build`
# error: failed to run custom build command for `esp-idf-sys v0.33.3`                                                                                            

Caused by:
  process didn't exit successfully: `E:\rust\esp32-std-project\target\debug\build\esp-idf-sys-7beb8c60ac8a9a8b\build-script-build` (exit code: 1)
  --- stderr
  Error: Too long output directory: `\\?\E:\rust\esp32-std-project\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-168dae4d88221cf9\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!
warning: build failed, waiting for other jobs to finish...
[image]

Vollbrecht, 2023-10-19:
On windows please try to put your projects in root project like C:\your-project to make the path as short as possible or use WSL as it has not such a problem. This is a windows specific problem we cant influence.

achieve-dream1221, 2023-10-20:
> this works for me [image]
> 
> in project [image]
> 
> you may also need to "cargo clean" and delete the ".embuild" folder

thank you so much, it worked!

Julien-cpsn, 2023-10-20:
> this works for me [image]
> 
> in project [image]
> 
> you may also need to "cargo clean" and delete the ".embuild" folder

Just for precision : this line goes into the `.cargo/config.toml` file

xobs, 2023-10-24:
I think this workaround should be enabled by default if it isn't fatal. Note that the recommended workaround doesn't work, so the documentation should be updated to reflect this approach instead: https://esp-rs.github.io/book/misc/troubleshooting.html#long-path-names

ivmarkov, 2023-10-28:
> I think this workaround should be enabled by default if it isn't fatal. Note that the recommended workaround doesn't work, so the documentation should be updated to reflect this approach instead: https://esp-rs.github.io/book/misc/troubleshooting.html#long-path-names

*What* workaround should be enabled by default if it isn't fatal? If you mean `ESP_IDF_PATH_ISSUES=warn` - this is BAD, because - depending on what bits and pieces you use from ESP IDF - your build might fail in weird ways (partial `git` clones, strange build errors). This workaround is a bit "use at your own risk".

And I'm not enabling it by default no matter what, as we were flooded by "I have a weird build failure on Windows" errors before.

As for the trick described in the ESP book, I don't know why it is there, given that - last time I checked - it does NOT work for `esp-idf-sys`. Again, the error message is:
{quote}
Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). ***Note that tricks like Windows subst do NOT work!***
{quote}

(Emphasis mine.)

Otherwise, thanks to you and the rest of the folks trying to help! Unfortunately I'm not very optimistic we can improve the "too long Windows path" issue handling beyond what we have and what we suggest.

ivmarkov, 2023-10-28:
UPDATE: For the "subst" trick - I've posted a question in the Matrix chat as I'm not sure who was ever successfully using that. But other than that, not sure what else to do here. Closing :)

closed 2023-10-28 by ivmarkov

Tastaturtaste, 2023-12-05:
> UPDATE: For the "subst" trick - I've posted a question in the Matrix chat as I'm not sure who was ever successfully using that. But other than that, not sure what else to do here. Closing :)

I just successfully used that trick. What error did you @ivmarkov get when trying it out? Did you have [windows long path support](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry#enable-long-paths-in-windows-10-version-1607-and-later) set up correctly?

ivmarkov, 2024-01-03:
@Tastaturtaste 
> I just successfully used that trick. What error did you @ivmarkov get when trying it out? Did you have [windows long path [...]

* This trick (SUBST) does not work, plain and simple. I have in the meantime tested it.
* The link you've pasted w.r.t. regedits mentions that there are TWO issues that need to be fulfilled for that to work. The regedit is just one of them.

archifo, 2024-02-03:
Does it mean that it is definitively impossible to use Rust for espressif on windows ? Sorry but I'm quite angry.
Isn't it worth trying to found a real workaround ? like putting temporary files flatly in %TMP% ? Or any better idea?
I've never seen this kind of problem with other chipsets supporting Rust...

ivmarkov, 2024-02-03:
> Does it mean that it is definitively impossible to use Rust for espressif on windows ? 

It is possible.
In the original bug report, as well as throughout the thread that follows, as well as in the error message that the build spits out, the solution is given:
 
"Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows subst do NOT work!"

> Sorry but I'm quite angry. 

Espressif is currently investing (as in hired, full time developers) in the baremetal crates, not the ones which are based on top of ESP-IDF. Either because they consider the baremetal solutions more important, or because they think the ESP-IDF crates are in the meantime relatively mature, or both. Speculating here a bit, but that's the status quo.

Therefore, the effort put into `esp-idf-sys/hal/svc` is based on mine (and others) free time. As in good will and so on.

> Isn't it worth trying to found a real workaround ? like putting temporary files flatly in %TMP% ? Or any better idea? 

Not sure for that suggestion. Might or might not work. If we go this route, we have to put not just build artefacts, but also the ESP IDF git repo itself in something like `/tmp/<short-tmp-dir>`.

Another solution would be to make sure that *ALL* tools involved in building the `esp-idf-sys` crate itself as well as the ESP-IDf itself understand long paths, and if not - fix them. Perhaps possible, but definitely not easy.

You also need to realize, that the "please use a short project path" restriction *is coming from ESP IDF itself*. It is just that the cargo build system is making it worse (as in the project path needs to be even shorter).

> I've never seen this kind of problem with other chipsets supporting Rust...

Espressif's ESP IDF is unique in its class in that it provides you a Rust STD environment. On an MCU. Therefore, I think this "short project path" constraint on pure Windows (it does not exist iwith WSL2's native file system) is a small price to pay for that. Surely annoying, but tolerable. But then - IMO.  Perhaps if you or other folks find this really inconvenient, you can work towards a solution?

N3xed, 2024-02-03:
> Does it mean that it is definitively impossible to use Rust for espressif on windows ? Sorry but I'm quite angry. Isn't it worth trying to found a real workaround ? like putting temporary files [...]

Well, you can always give your hands on trying to create a workaround for this. I originally wrote this compilation infrastructure on Windows, where we ran into _command-line_ length limitations (see https://github.com/esp-rs/esp-idf-sys/pull/7#issuecomment-909164445 and https://github.com/espressif/esp-idf/issues/7864), which was fixed by using response files and preventing the linker to expand the response files while running the jobs.

Then, someone ran into Windows file path limitations (see https://github.com/esp-rs/esp-idf-sys/issues/23). Which seems to be caused by gcc no supporting long paths on Windows (see my comment https://github.com/esp-rs/esp-idf-sys/issues/23#issuecomment-956238333). There may be other tools that we're using that also don't support this (e.g. I seem to remember there were some issues with git and too long paths as well).

The original workarounds I posted are:

> Possible workarounds:
> 
> * Use [`subst`](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/subst) to assign a virtual drive letter to a path (should work as pointed out [here](https://stackoverflow.com/questions/7866387/git-mingw32-and-long-path-file-names-under-windows)).
> 
> * Create a directory junction (with [`mklink /j`](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/mklink)) to shorten the path (not tested).
> 
> * Moving the project to a directory with a shorter path.

Where point one seems not to work. But maybe point 2 could? The issue is also with the system headers (e.g. for newlib) which reside in the `.embuild` directory. So I think only moving your project to a shorter path may not actually help if your `.embuild` folder where the toolchain is stored is also at a long path. But almost certainly the root cause is still GCC not supporting long paths.

I've been pretty inactive, and am also not building stuff with esp-idf-sys, so I've not looked into workarounds further. But again if anyone wants improvements in this regard, you'd (or anyone) have to it yourself.

Tastaturtaste, 2024-02-03:
Regarding the gcc side of the problem there [is this tracked bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=107974). According to one of the comments the fix was simply to add a manifest to enable long path support on windows. It seems nobody has pursued this further.

RingCanary, 2024-08-30:
> > Does it mean that it is definitively impossible to use Rust for espressif on windows ?
> 
> It is possible. In the original bug report, as well as throughout the thread that follows, as well as in the error message that the build spits out, the solution is given:
> 
> "Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows subst do NOT work!"
> 
> > Sorry but I'm quite angry.
> 
> Espressif is currently investing (as in hired, full time developers) in the baremetal crates, not the ones which are based on top of ESP-IDF. Either because they consider the baremetal solutions more important, or because they think the ESP-IDF crates are in the meantime relatively mature, or both. Speculating here a bit, but that's the status quo.
> 
> Therefore, the effort put into `esp-idf-sys/hal/svc` is based on mine (and others) free time. As in good will and so on.
> 
> > Isn't it worth trying to found a real workaround ? like putting temporary files flatly in %TMP% ? Or any better idea?
> 
> Not sure for that suggestion. Might or might not work. If we go this route, we have to put not just build artefacts, but also the ESP IDF git repo itself in something like `/tmp/<short-tmp-dir>`.
> 
> Another solution would be to make sure that _ALL_ tools involved in building the `esp-idf-sys` crate itself as well as the ESP-IDF itself understand long paths, and if not - fix them. Perhaps possible, but definitely not easy.
> 
> You also need to realize, that the "please use a short project path" restriction _is coming from ESP IDF itself_. It is just that the cargo build system is making it worse (as in the project path needs to be even shorter).
> 
> > I've never seen this kind of problem with other chipsets supporting Rust...
> 
> Espressif's ESP IDF is unique in its class in that it provides you a Rust STD environment. On an MCU. Therefore, I think this "short project path" constraint on pure Windows (it does not exist with WSL2's native file system) is a small price to pay for that. Surely annoying, but tolerable. But then - IMO. Perhaps if you or other folks find this really inconvenient, you can work towards a solution?

It's better to use WSL2 as recommended and please refer [this](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) to bind the usb ports to the hosted linux distro -- things works like a charm :)

ivmarkov, 2024-08-30:
> It's better to use WSL2 as recommended and please refer [this](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) to bind the usb ports to the hosted linux distro -- things works like a [...]

You are preaching to the wrong choir. I never said it does not work or that WSL2 is not a good workaround. If you would like to help, a paragraph to the README of esp-idf-template dedicated to Windows+WSL2, including some instructions what to install so that `espflash` works from inside WSL2 would be really, really appreciated.

RingCanary, 2024-08-31:
Happy to help, will definitely do so over this weekend @ivmarkov.

RingCanary, 2024-09-01:
update : https://github.com/esp-rs/esp-idf-template/pull/232

Hecatron, 2025-07-28:
For anyone else running into this problem under windows
you can try setting both the `ESP_IDF_TOOLS_INSTALL_DIR` and `ESP_IDF_PATH_ISSUES` options
This will override / place the espressif files under `C:\Users\<username>\.espressif`

I think this also has the added benifit of not needing to recompile the espressif files between different projects (assuming the same config is used)
and you don't need WSL as well

```yaml
[env]
# This places the esp-idf-sys build dir under
# Windows: C:\Users\<username>\.espressif
# Linux: ~/.espressif
# This avoids path too long issues - https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md
ESP_IDF_TOOLS_INSTALL_DIR="global"
ESP_IDF_PATH_ISSUES="warn"

sergt1978, 2025-08-13:
c:\j>cargo build
warning: unused manifest key: env
....
error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `c:\j\target\debug\build\esp-idf-sys-acf273d75f9a8565\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: Some(
          "sdkconfig",
      ),
      esp_idf_sdkconfig_defaults: Some(
          [
              "sdkconfig.defaults",
              "sdkconfig.defaults.ble",
          ],
      ),
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2.2") }
  Error: Could not install esp-idf

  Caused by:
      0: command '"git" "clone" "--jobs=16" "--recursive" "--depth" "1" "--shallow-submodules" "--branch" "v5.2.2" "https://github.com/espressif/esp-idf.git" "C:\\Users\\sergt\\.espressif\\esp-idf\\v5.2.2"' failed to start
      1: program not found
warning: build failed, waiting for other jobs to finish...

это не работает для меня. 
Поддержку длинных имен включил,  раздел env в манифест добавил

sergt1978, 2025-08-13:
everything about [env] has been transferred to the appropriate file, thank you very much
for python 3.13, what is GIT, please tell me, I'll put it.
But so far, the problem of installing esp-idf has not gone away.

    Checking uncased v0.9.10
    Checking embassy-time-driver v0.1.0
    Checking embassy-futures v0.1.1
error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `c:\j\target\debug\build\esp-idf-sys-acf273d75f9a8565\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: Some(
          "sdkconfig",
      ),
      esp_idf_sdkconfig_defaults: Some(
          [
              "sdkconfig.defaults",
              "sdkconfig.defaults.ble",
          ],
      ),
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2.2") }
  Error: Could not install esp-idf

  Caused by:
      0: command '"git" "clone" "--jobs=16" "--recursive" "--depth" "1" "--shallow-submodules" "--branch" "v5.2.2" "https://github.com/espressif/esp-idf.git" "C:\\Users\\sergt\\.espressif\\esp-idf\\v5.2.2"' failed to start
      1: program not found
warning: build failed, waiting for other jobs to finish...

sergt1978, 2025-08-13:
[image]
It seemed to work after installing git and python.
It only hangs for a very long time, does the Internet really need to check and compile after all the downloaded stuff... A strange scheme  
If a person does not have constant access to the network, they will not be able to do anything.

sergt1978, 2025-08-13:
[image]
It seemed to work after installing git and python.
It only hangs for a very long time, does the Internet really need to check and compile after all the downloaded stuff... A strange scheme  
If a person does not have constant access to the network, he will not be able to do anything
, there has been no movement for 40 minutes, 500 MB of Internet has been downloaded.
A strange work process