esp-rs/esp-idf-sys #254: Cargo binstall version for releases please?

noahbliss, 2023-10-21:
Hello! 

One major caveat of using this crate is the long initial build time. Many other rust projects have circumvented this limitation by offering a cargo binstall crate as part of the github releases. Perhaps this would be possible? More info on binstall here: https://crates.io/crates/cargo-binstall

Vollbrecht, 2023-10-21:
You have to understand that this crate is an entrypoint into the esp-idf FreeRTOS. If not present it downloads it, it configures your project - and it will recompile it depending on the sdkconfig settings you set. The majority of the time is spent there usually - not doing the rust compile itself. Its like building the linux kernel but in small, and if you use different feature-sets in the downstream hal / svc crates it will also compile depending on that. So you wont just make a bininstall of this and call it good. Hope that clarify it a bit.

noahbliss, 2023-10-21:
It does. Thanks. I guess a more general question would be is there any way to markedly speed up this process for users who only leverage the most common options?

Vollbrecht, 2023-10-21:
The first thing you should notice is that usually for any given project,  esp-idf-sys should only compile once, every other incremental build would not touch esp-idf-sys, thous making it not the pain point anymore. 

There are options to use a globally installed esp-idf, so it would not redownload it for every project. This is in general a speed positive, but give up on the fully self-contained project. So the user would now be tasked with updating the esp-idf version and keep it all in sync, witch is a source for errors for many people.

Also other options would be to use a systemwide cache for your builds like sccache. While this can give a nice speedoost across projects, cache invalidation errors can be very painfull.

In that light, its convenience/robustness vs speed. Compiling esp-idf and using it with rust is already a complex task, so our main goal is to make it simpler and less error prone first even if it might be a tat slower.

noahbliss, 2023-10-21:
My specific use-case is with the container that is available on dockerhub. If we're talking about containers only, maybe this allows some concessions on portability? Just a thought. In any case thanks for the reply, I'll sit and wait a little happier knowing I at least asked. Haha

N3xed, 2023-10-21:
> It does. Thanks. I guess a more general question would be is there any way to markedly speed up this process for users who only leverage the most common options?

There are a couple ways to speed up the initial build:

1. Use a cached/already present esp-idf tools directory and esp-idf repository with [`ESP_IDF_TOOLS_INSTALL_DIR`](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir) and/or [`IDF_PATH`](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#idf_path-idf_path-native-builder-only).
2. Set [`esp_idf_components`](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_components-esp_idf_components-native-builder-only) to the list of esp-idf components that are actually being used.
3. Cache the esp-idf cmake build that esp-idf-sys does, for example with `ccache` (see the [esp-idf docs](https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-guides/build-system.html#idf-py-options).

I think that's about it.

closed 2023-10-21 by noahbliss