esp-rs/esp-idf-sys #256: ESP32-2432S028R / ili9341 - compile error:    | xtensa-esp32-elf

ds2k5, 2023-11-09:
Hello,
here is what I did

**cargo generate --git https://github.com/esp-rs/esp-idf-template cargo
cd esp32display
cargo add mipidsi
cargo add esp-idf-svc
cargo add esp-idf-sys
cargo add display-interface

cargo check**

[output.txt](https://github.com/esp-rs/esp-idf-sys/files/13306030/output.txt)

[output_backtrace.txt](https://github.com/esp-rs/esp-idf-sys/files/13306055/output_backtrace.txt)

What did I wrong ?

the source code:

```
fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello, world!");

    // create a DisplayInterface from SPI and DC pin, with no manual CS control
    let di = SPIInterfaceNoCS::new(spi, dc);
    // create the ili9341 display driver from the display interface and optional RST pin
    let mut display = Builder::ili9341(di)
        .init(&mut delay, Some(rst));
    // clear the display to black
    display.clear(Rgb666::BLACK).unwrap();

}

```

Vollbrecht, 2023-11-09:
did you install the esp toolchain via espup? did you source the generated file at the end to setup your env var pointing to the installed esp-clang version?

Vollbrecht, 2023-11-09:
also not directly releated, if you generate a new project wicht cargo generate you dont need to directly depend on esp-idf-sys and esp-idf-hal anymore. They are both fully re-exported through esp-idf-svc. For example getting to timer.rs in hal is esp-idf-svc::hal::timer::*; etc

ds2k5, 2023-11-09:
@Vollbrecht 

>did you install the esp toolchain via espup?

yes 

>did you source the generated file at the end to setup your env var pointing to the installed esp-clang version?

Thanks for the hint..... 

I forgot to run:    . ~/export-esp.sh

I am so sorry!

----

. ~/export-esp.sh
cargo clean
rm -rf .embuild
cargo check

now another error.... but not with esp-idf-sys

closed 2023-11-17 by ivmarkov