esp-rs/esp-idf-sys #263: Windows: "Error: Too long output directory" on path with 9 characters

xobs, 2023-11-20:
I'm trying to build on Windows, and got the error to "Shorten your project path down to no more than 10 characters". My path is `E:\fp\prs`, which is 5 characters, 6 if you include the path separator, 9 if you add in the drive letter.

Is the error message incorrect or just misleading? If I cut it down to `E:\fp\pr` then it seems to be happy, which is 8 characters all-told, and only 4 characters of meaningful path.

ivmarkov, 2023-11-20:
That's all we can do for Windows. On your own risk, you can turn this error into a warning. Or migrate to WSL2.

closed 2023-11-20 by ivmarkov

xobs, 2023-11-20:
But 9 characters is less than 10 characters, yet it still emits the warning. Is there an off-by-one error somewhere in the detector? Or should the warning be changed to "...no more than 9 characters"?

ivmarkov, 2023-11-20:
Are you doing `subst` or similar tricks?

xobs, 2023-11-20:
No, this is on my E: drive which is just a normal partition.

ivmarkov, 2023-11-20:
Great. What directory does it output in this message: 
`Too long output directory: <HERE>`

xobs, 2023-11-20:
It says this:

```
Caused by:
  process didn't exit successfully: `E:\fp\prs\target\debug\build\esp-idf-sys-69aeb28bf0abd57b\build-script-build` (exit code: 1)
  --- stderr
  Error: Too long output directory: `\\?\E:\fp\prs\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-791cc6c77d505d35\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!
```

ivmarkov, 2023-11-20:
Apparently the path gets expanded into a [Windows Device Path by some of the Rust APIs](https://ikrima.dev/dev-notes/win-internals/win-path-formats/). How and why - a bit unclear. And *assuming* the expansion happens, and the path is expanded precisely to DevidcePath, yes there seems to be an off by 1 error.

I'll fix.

Guddiny, 2024-01-17:
Same issue. Did you fix it?

ivmarkov, 2024-01-18:
https://github.com/esp-rs/esp-idf-sys/commit/50f6bfdac0a6f7eab363d231aecd837849085761

Not released yet though.

wumao, 2024-02-20:
warning: unused config key `http.check-config` in `C:\rust\cargo\config`
   Compiling esp-idf-sys v0.34.0
error: failed to run custom build command for `esp-idf-sys v0.34.0`

Caused by:
  process didn't exit successfully: `E:\sp32demo\target\debug\build\esp-idf-sys-b156624b7e3f52e8\build-script-build` (exit code: 1)
  --- stderr
  Error: Too long output directory: `\\?\E:\sp32demo\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-572e777190dc7832\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks

wumao, 2024-02-25:
@ivmarkov  still not work on windows 11.    esp32-s3
PS E:\rust\device> cargo build
warning: unused config key `http.check-config` in `C:\rust\cargo\config`
   Compiling esp-idf-sys v0.34.1
error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `E:\rust\device\target\debug\build\esp-idf-sys-a53b120d8fe38da9\build-script-build` (exit code: 1)
  --- stderr
  Error: Too long output directory: `\\?\E:\rust\device\target\xtensa-esp32s3-espidf\debug\build\esp-idf-sys-a753a1ef795c9d59\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!

ivmarkov, 2024-02-25:
> @ivmarkov still not work on windows 11. esp32-s3 PS E:\rust\device> cargo build warning: unused config key `http.check-config` in `C:\rust\cargo\config` Compiling esp-idf-sys v0.34.1 error: failed [...]

Did you actually read the error message?