esp-rs/esp-idf-sys #267: What ESP_IDF version do I have?

coder0xff, 2023-12-17:
Hi. I'm running into the same problem as #238, but things were working fine yesterday and I don't know what's changed. To recap:

```
$ cargo generate esp-rs/esp-idf-template cargo
...
$ cargo build
...
error[E0308]: mismatched types
  --> /home/coder0xff/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.7/src/lib.rs:50:62
   |
50 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /home/coder0xff/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.7/src/lib.rs:52:51
   |
52 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` (lib) due to 2 previous errors
```

#238 mentions needing esp-idf 5.0 or greater installed. How do I find out what version I have installed? Or what version it's trying to use? How do I update my version?

I remembered that espup and export-esp.sh was part of figuring out the initial setup, so I was able to get this much:

```
$ espup --version
espup 0.10.0
$ cat ~/export-esp.sh
export LIBCLANG_PATH="/home/coder0xff/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.4-20231113/esp-clang/lib"
export PATH="/home/coder0xff/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:$PATH"
```

Sorry if those aren't helpful. Thanks for taking a look.

Vollbrecht, 2023-12-17:
If you created your project with the template and cargo-generate, than the version is determined by .cargo/config.toml inside ESP_IDF_VERSION field. 

If you are using esp-idf v5 you need to pass additional compiler flags duo to the 2038 time problem that is fixed inside esp-idf 5 e.g passing `    "--cfg","espidf_time64",`  Check the difference in the esp-template in version v4.4.6 and 5.1.2

coder0xff, 2023-12-17:
Ok, yes. I checked that the generated project has `ESP_IDF_VERSION = "v5.1.1"` and `rustflags = [ "--cfg",  "espidf_time64"]` so that seems correct. Do you have any advice on what I can check next?

Vollbrecht, 2023-12-17:
make sure that you don't override that mention env variable manual or have others env variables set from some local / old esp-idf install you did manually.A list of all env variables that affect esp-idf-sys can be found [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp-idf-configuration).

 Also you can clear the .embuild dir in your project by removing it. It will download and set it up from scratch automatically.

coder0xff, 2023-12-24:
I exited my terminal session and did a new generation. You were probably right about an environment variable being the cause. Thanks for the help @Vollbrecht

closed 2023-12-24 by coder0xff