esp-rs/esp-idf-sys #269: [QUESTION] Safe wrapper for IDF component

Jozott00, 2023-12-22:
I would like to write a safe wrapper for an IDF component (golioth-sdk). 

My first approach was to write a `-sys` crate using the `esp-idf-sys` dependency and add the additional component in the Cargo.toml via
```
[[package.metadata.esp-idf-sys.extra_components]]
component_dirs = ["vendor/port/esp_idf/components"]
bindings_header = "src/bindings.h"
bindings_module = "golioth"
```

Then I wrote an sdk library that provides a safe wrapper around the `-sys` crate. This all works like a charm.

Unfortunately, if a crate wants to use the sdk library, it must add the `[[package.metadata.esp-idf-sys.extra_components]]` configuration to its root crate (as mentioned in the docs). Otherwise, `esp-idf-sys` will not generate the bindings for the Golioth component even if it is configured in the `-sys` crate. 

What would be the right way to write a safe wrapper for an idf component? Would I have to write a custom build script with `embuild`? 

Dependency structure:
`esp-idf-sys <- golioth-sdk-sys <- golioth-sdk <- <some-project>`

I appreciate any help on this topic!

N3xed, 2023-12-23:
This is because of esp-idf-sys will honor only **direct dependencies'** Cargo.toml of the root crate, as described [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#extra-esp-idf-components).
This limitation was introduced so that a user sees directly which dependencies can use extra esp-idf components, since the feature allows to depend on arbitrary code not in the crate graph of cargo.

The solution is simple, a user of the wrapper must also depend on the -sys library. So with your example, `golioth-sdk-sys` *and* `golioth-sdk` must be depended on.

Jozott00, 2023-12-26:
Alright, thank you for the clarification!

closed 2023-12-26 by Jozott00