esp-rs/esp-idf-sys #272: Failing to add mdns remote component

PatOConnor43, 2023-12-31:
Hi 👋 I'm trying to use esp_idf_svc::mdns module but I'm having trouble actually getting the build process to pull the remote component. I'm using an esp32s3 and my project seems to be building fine when I don't import this module in `src/main.rs`.

I'm familiar with this build option:
https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#remote-components-idf-component-registry

My Cargo.toml file looks like this:
```toml
[package]
name = "esp32s2-server2"
version = "0.1.0"
authors = ["Pat O'Connor <PatOConnor43@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[package.metadata.esp-idf-sys.extra_components]
remote_component = { name = "espressif/mdns", version = "1.2" }

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "hal", "esp-idf-sys/native"]

pio = ["esp-idf-sys/pio"]
all = ["std", "nightly", "experimental"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]
std = ["alloc", "esp-idf-sys/std", "esp-idf-sys/binstart", "embedded-svc?/std", "esp-idf-hal?/std", "esp-idf-svc?/std"]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
nightly = ["embedded-svc?/nightly", "esp-idf-svc?/nightly"] # Future: "esp-idf-hal?/nightly"
experimental = ["embedded-svc?/experimental", "esp-idf-svc?/experimental"]
#embassy = ["esp-idf-hal?/embassy-sync", "esp-idf-hal?/critical-section", "esp-idf-svc?/embassy-time-driver", "esp-idf-svc?/embassy-time-isr-queue"]

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33.7", default-features = false }
esp-idf-hal = { version = "0.42.5", optional = true, default-features = false }
esp-idf-svc = { version = "0.47.3", optional = true, default-features = false }
embedded-svc = { version = "0.26.4", optional = true, default-features = false}
anyhow = "1.0.72"
futures = "0.3.28"
heapless = { version = "0.8", default-features = false }

[build-dependencies]
embuild = "0.31.2"

```

I don't know if this is necessary but my `build.rs` file looks like this:
```rust
fn main() -> Result<(), Box<dyn std::error::Error>> {
    embuild::espidf::sysenv::output();
    Ok(())
}
```

Is there something else I need to do to get the build to download mdns? Thanks for all the hard work on this project!

Vollbrecht, 2023-12-31:
your syntax is a bit wrong in the Cargo.toml. You need double `[]` around the definition like the following -> 
```toml
[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "espressif/mdns", version = "1.2" }
```
this is similar to how a bin and a lib definitions are written. So essentially its a array-of-tables of TOML configs so its technically possible to write more than one of this definitions. For more info look [here](https://doc.rust-lang.org/cargo/reference/cargo-targets.html)

PatOConnor43, 2023-12-31:
Thanks for the response. I did try that as well. It seems to have the same result. Here's what the output of `cargo run --release` looks like:
```
cargo run --release
   Compiling esp32s2-server2 v0.1.0 (/Users/patrickoconnor/workspace/esp32s2-server2)
error[E0432]: unresolved import `esp_idf_svc::mdns`
  --> src/main.rs:3:5
   |
3  | use esp_idf_svc::mdns;
   |     ^^^^^^^^^^^^^^^^^ no `mdns` in the root
   |
note: found an item that was configured out
  --> /Users/patrickoconnor/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.47.3/src/lib.rs:70:9
   |
70 | pub mod mdns;
   |         ^^^^

```

Vollbrecht, 2023-12-31:
you may need to run a `cargo-clean` to make sure the bindgens are build correctly. You can check that the flag is correctly set and included with ` cargo rustc -- --print cfg | grep esp_idf_comp_espressif__mdns_enabled`

PatOConnor43, 2023-12-31:
Short story: it seems to be fixed now. Thanks!

Long story: I ended up running `cargo clean`, `rm -r .embuild`, and `git submodule deinit --all`. The error I seemed to be getting was a problem cloning the `esp-idf` repo from espressif. The git error included `invalid index-pack output`. After searching around that seemed to be caused by a network issue. So I restarted my computer, ran the commands above, then ran `cargo run --release` and everything worked as expected. I'm now able to build while importing `mdns`.

Thanks for the help!

closed 2023-12-31 by PatOConnor43