esp-rs/esp-idf-sys #273: How to allocate a vector or array in PSRAM

enelson1001, 2024-01-03:
I have a ESP32-S3 WROOM board with 8MB of PSRAM.  How can I use the PSRAM for a vector or an array?

Vollbrecht, 2024-01-04:
First make sure you read the esp-idf [docu](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html) around PSRAM. 

After you configured your sdkconfig correctly, you see different methods described in that documentation. Depending on that you can directly use normal rust methods to put things on the heap, though if you want a little bit more control over it you have to directly intercept here.

enelson1001, 2024-01-05:
Thank you for the link to document.

So I added the following to sdkconfig.defaults
```
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y
```
and during bootup I see the following which tells me I have the 8MB PSRAM available for heap allocation
```
I (797) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
```
I added the following to main to test PSRAM allocation
```
log::info!("========== Placing vector in PSRAM ==========");
    let mut large_vec: Vec<u32> = Vec::with_capacity(500 * 1024 / 4);

    for i in 0..(500 * 1024 / 4) {
        large_vec.push((i & 0xff) as u32);
    }

    log::info!("large vec size = {} bytes", large_vec.len() * 4);
    log::info!("large vec address = {:p}", large_vec.as_ptr());
    log::info!("large vec[..100] = {:?}", &large_vec[..100]);
```
And the results
```
I (889) std_psram_test: ========== Placing vector in PSRAM ==========
I (909) std_psram_test: large vec size = 512000 bytes
I (919) std_psram_test: large vec address = 0x3c070958
0x3c070958 - _ext_ram_bss_end
    at ??:??
I (919) std_psram_test: large vec[..100] = [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99]
```
In ESP-IDF Programming Guide section "Support for External RAM" it states 
```
During the ESP-IDF startup, external RAM is mapped into the data address space, starting at address 0x3D000000 (byte-accessible). The length of this region is the same as the SPI RAM size (up to the limit of 32 MB).
```

My question is why is the large_vec pointer address at 0x3c070958 instead of 0x3d000000 ?

Vollbrecht, 2024-01-06:
First is maybe to try to understand the memory model of the data structure you are talking about. Maybe something like [this](https://cheats.rs/#data-structures) helps. Also possible reading the [source](https://doc.rust-lang.org/src/alloc/vec/mod.rs.html#396) might help. 

After that you should ask yourself, "do i ask the rust program really what i want to know" ? To me it looks like you are not asking the right question to your rust program, e.g you are not asking it to print the address of the allocated memory rather asking what address the pointer lives on.

enelson1001, 2024-01-06:
OK I looked at the source at lines 312 to 322.  This seems to tell me that ptr holds the start address to the location where the vector is on the heap.  The function as_ptr states it returns a raw pointer to vectors buffer.
When I look at the log info printed to terminal it shows the following
```
I (910) std_psram_test: large vec ptr address = 0x3c070958
0x3c070958 - _ext_ram_bss_start
    at ??:??
```
So I assume the words ``` 0x3c070958 - _ext_ram_bss_start ``` means the vector is located in PSRAM.

For now I am going to assume the ESP-IDF documentation is not correct when it stated PSRAM starting address is 0x3D000000.

I will close this issue.

closed 2024-01-06 by enelson1001