esp-rs/esp-idf-sys #275: backtrace (of rust code) is incomplete on esp32c6

t-moe, 2024-01-09:
I'm trying to obtain a backtrace for a simple rust program when it panics.

Code:

```rust
fn c() {
    panic!("oh no");
}

fn b() {
    c();
}

fn a() {
    b();
}

fn main() {
    a();
}
```

I'm using the esp32c6 here, together with `espflash run --monitor`.
I have found that the amount of info I can get from the backtrace varies a lot when using `esp-idf-sys` compared to when using `esp-hal`.

`rust --version -v`:
```
rustc 1.77.0-nightly (e51e98dde 2023-12-31)
binary: rustc
commit-hash: e51e98dde6a60637b6a71b8105245b629ac3fe77
commit-date: 2023-12-31
host: x86_64-unknown-linux-gnu
release: 1.77.0-nightly
LLVM version: 17.0.6
```

--------------------------

# esp-hal // no_std Case

Based on `cargo generate esp-rs/esp-template` I have the following setup.

`main.rs`:
```rust
#![no_std]
#![no_main]

use esp32c6_hal::prelude::*;
use esp_backtrace as _;

fn c() {
    panic!("oh no");
}

fn b() {
    c();
}

 fn a() {
    b();
}

#[entry]
fn main() -> ! {
    a();
    loop {};
}

```

[... 871 lines cut ...]
0xa5a5a5a5 - s_rtc_timer_retain_mem
    at ??:??
0xa5a5a5a5 - s_rtc_timer_retain_mem
    at ??:??
0xa5a5a5a5 - s_rtc_timer_retain_mem
    at ??:??
0xa5a5a5a5 - s_rtc_timer_retain_mem
    at ??:??
0xa5a5a5a5 - s_rtc_timer_retain_mem
    at ??:??
0xa5a5a5a5 - s_rtc_timer_retain_mem
    at ??:??

```
</details>

------

# Conclusion / Summary

The output of the esp-idf-version does not contain any of my functions (a,b,c). It does contain main, buried under a lot of noise.

I dont care a lot about the noise, but I would really like to see my functions (a,b,c).

# Please wait with answering, I'll add some comments that describe what I've tried and the results

t-moe, 2024-01-09:
# Experiment 1:  Add `esp_backtrace` to get more/alternative output.

New dependencies in `Cargo.toml`:
```esp-backtrace = { version = "0.10.0", features = ["print-jtag-serial", "esp32c6"] }```

Modified function `c` in `main.rs`:
```rust
fn c() {
    let bt = esp_backtrace::arch::backtrace();
    for a in bt {
        if let Some(a) = a {
            println!("esp_backtrace: 0x{:x}", a);
        }
    }

    panic!("oh no");
}
```

Additional output compared to before (raw, without espflash monitor):
```
esp_backtrace: 0x4200028a
esp_backtrace: 0x4201e1ae
esp_backtrace: 0x42000270
```

`espflash run --monitor` enhanced:
```
esp_backtrace: 0x4200028a
0x4200028a - std::rt::lang_start::{{closure}}
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:166
esp_backtrace: 0x4201e1ae
0x4201e1ae - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:284
esp_backtrace: 0x42000270
0x42000270 - std::rt::lang_start
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:172

```

This does not really add much value on its own. The functions in question are still missing.

t-moe, 2024-01-09:
# Experiment 3: Set `RUST_BACKTRACE=1`

I've added `std::env::set_var("RUST_BACKTRACE", "1");`  to main.rs, the rest is the same as in Experiment 2.

<details>
<summary>main.rs
</summary>

```rust
#[inline(never)]
fn c() {
    let bt = esp_backtrace::arch::backtrace();
    for a in bt {
        if let Some(a) = a {
            println!("esp_backtrace: 0x{:x}", a);
        }
    }

    panic!("oh no");
}

#[inline(never)]
fn b() {
    c();
}
#[inline(never)]
fn a() {
    b();
}

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();
    std::env::set_var("RUST_BACKTRACE", "1");
    a();
}

```
</details>

<details>
<summary>
espflash enhanced output:
</summary>

```
esp_backtrace: 0x42000232
0x42000232 - backtrace_std::main
    at /tmp/backtrace-std/src/main.rs:24
esp_backtrace: 0x4200026e
0x4200026e - main
    at ??:??
esp_backtrace: 0x42000040
0x42000040 - std::sys_common::backtrace::__rust_begin_short_backtrace
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:161
esp_backtrace: 0x420002ec
0x420002ec - std::rt::lang_start::{{closure}}
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:166
esp_backtrace: 0x4201e8e8
0x4201e8e8 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:284
esp_backtrace: 0x420002d2
0x420002d2 - std::rt::lang_start
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:172
thread 'main' panicked at src/main.rs:11:5:
oh no
stack backtrace:

abort() was called at PC 0x420444b1 on core 0
0x420444b1 - abort_return<int>
[... 612 lines cut ...]
    at ??:??
0x00000000 - core::array::iter::_$LT$impl$u20$core..iter..traits..collect..IntoIterator$u20$for$u20$$u5b$T$u3b$$u20$N$u5d$$GT$::into_iter::h1f11533074ef3cf2
    at ??:??
4080f490: 0x4080f420 0x00000000 0x4080f4c8 0x00000000 0x4080f4b0 0x0000000a 0x0000000a 0x420002d2
0x4080f420 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x00000000 - core::array::iter::_$LT$impl$u20$core..iter..traits..collect..IntoIterator$u20$for$u20$$u5b$T$u3b$$u20$N$u5d$$GT$::into_iter::h1f11533074ef3cf2
    at ??:??
0x4080f4c8 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x00000000 - core::array::iter::_$LT$impl$u20$core..iter..traits..collect..IntoIterator$u20$for$u20$$u5b$T$u3b$$u20$N$u5d$$GT$::into_iter::h1f11533074ef3cf2
    at ??:??
0x4080f4b0 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x0000000a - core::array::iter::_$LT$impl$u20$core..iter..traits..collect..IntoIterator$u20$for$u20$$u5b$T$u3b$$u20$N$u5d$$GT$::into_iter::h1f11533074ef3cf2
    at ??:??
0x0000000a - core::array::iter::_$LT$impl$u20$core..iter..traits..collect..IntoIterator$u20$for$u20$$u5b$T$u3b$$u20$N$u5d$$GT$::into_iter::h1f11533074ef3cf2
    at ??:??
0x420002d2 - std::rt::lang_start
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:172

```
</details>

Same results as in experiment 2.

t-moe, 2024-01-09:
# Experiment 4: Enable DWARF by setting `CONFIG_ESP_SYSTEM_USE_EH_FRAME=y`

I've added `CONFIG_ESP_SYSTEM_USE_EH_FRAME=y`  to sdkconfig.defaults, the rest ist the same as in experiment 3.

<details>
<summary>sdkconfig.defautls
</summary>

```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000

CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT=n
CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT=y

CONFIG_ESP_SYSTEM_USE_EH_FRAME=y
# Needs `"-C", "force-unwind-tables=yes"` rustc flags to work properly

```
</details>

<details>
<summary>
raw serial monitor output:
</summary>

```
esp_backtrace: 0x42000232
esp_backtrace: 0x4200026e
esp_backtrace: 0x42000040
esp_backtrace: 0x420002ec
esp_backtrace: 0x4201e8e8
esp_backtrace: 0x420002d2
thread 'main' panicked at src/main.rs:11:5:
oh no
stack backtrace:

abort() was called at PC 0x42044d2d on core 0
Backtrace: 0x40800eba:0x4080f1a0 0x4080413e:0x4080f1a0 0x40807e6c:0x4080f1b0 0x42044d2e:0x4080f220 0x4201edae:0x4080f230 0x4200ab30:0x4080f240 0x42038d74:0x4080f2b0 0x42020414:0x4080f300 0x4200aa0e:0x4080f320 0x4200b026:0x4080f370 0x4200aee8:0x4080f3d0 0x4200b66a:0x4080f460 0x4200b4de:0x4080f4d0 0x4200abca:0x4080f4f0 0x4200b2ec:0x4080f500 0x4203d740:0x4080f520 0x42000210:0x4080f540 0x42000220:0x4080f5e0 0x42000230:0x4080f5f0 0x4200026c:0x4080f600 0x4200003e:0x4080f610 0x420002ea:0x4080f620 0x4201e8e6:0x4080f630 0x420002d0:0x4080f650 0x4204d0e8:0x4080f660 0xfffffffe:0x4080f680
Backtrace ended abruptly: cannot find DWARF information for instruction at address 0xfffffffe

```
</details>

<details>
<summary>
espflash enhanced output:
</summary>

```
esp_backtrace: 0x42000232
0x42000232 - backtrace_std::main
    at /tmp/backtrace-std/src/main.rs:24
esp_backtrace: 0x4200026e
0x4200026e - main
    at ??:??
esp_backtrace: 0x42000040
0x42000040 - std::sys_common::backtrace::__rust_begin_short_backtrace
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:161
esp_backtrace: 0x420002ec
0x420002ec - std::rt::lang_start::{{closure}}
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:166
esp_backtrace: 0x4201e8e8
0x4201e8e8 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:284
esp_backtrace: 0x420002d2
0x420002d2 - std::rt::lang_start
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:172
thread 'main' panicked at src/main.rs:11:5:
oh no
[... 95 lines cut ...]
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:284
0x4080f630 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x420002d0 - std::rt::lang_start
    at /home/timo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:165
0x4080f650 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x4204d0e8 - main_task
    at /home/timo/esp/esp-idf/components/freertos/app_startup.c:208
0x4080f660 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0xfffffffe - s_rtc_timer_retain_mem
    at ??:??
0x4080f680 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
Backtrace ended abruptly: cannot find DWARF information for instruction at address 0xfffffffe
0xfffffffe - s_rtc_timer_retain_mem
    at ??:??

```
</details>

**the panic output now contains main, a, b and c, and is much more concise.  :+1:**

(note: It ends with the error `Backtrace ended abruptly: cannot find DWARF information for instruction at address 0xfffffffe`)

t-moe, 2024-01-09:
# Conclusion

Coming from esp-hal, I find the backtraces on esp-idf-sys extremely painful to handle. They contain a lot of noise, and unless i enable DWARF generation, they will not contain any of my functions in question or only some of them. Surely there must be a way to get a clear and concise backtrace? 

How can we improve the situation?

When printing with esp_backtrace, why dont I see my rust functions a,b and c, when using esp-idf-sys instead of esp-hal. Have my frame pointers suddenly disappeared?

Thank you!

ivmarkov, 2024-01-09:
Your original experiment, in the original issue report:
> `espflash run --monitor` enhanced output: contains main, but neither a, b or c and is really verbose

This is because the panic output from ESP IDF contains a stack memory dump, that - unluckily - is not formatted with regular byte-oriented hexdump i.e. `0a 0c 22 34` but with clusters of 4 bytes each where each such cluster **looks** like (but isn't necessarily) a 32bit value.

My guess is that `espflash` blindly applies stacktrace decoding on the memory dump, assuming that each such 4-byte cluster is a properly decoded 32 bit word. Which seems not to be the case. Hence the noise.

ivmarkov, 2024-01-09:
> Experiment 1: Add esp_backtrace to get more/alternative output.
> Experiment 2: Add #[inline(never)]

> esp_backtrace contains main now, but not a,b,c

I don't have explanation for that. That's weird, because from LLVM's POV, the two targets look exactly the same.
**Maybe following question: what are your optimization settings? (opt-level)? 0, 1, 2 or 3?** It could be, that with opt-level != 0, even if you say "inline(never)", stuff is still sometimes inlined, though then I don't have an explanation why the `b` symbol then pops up in the stack memory dump below.

> panic output contains main, a and b but is still very noisy and not really helpful

Sure. As per above, espflash crawls though the stack memory and blindly applies decoding to everything which looks like a 32 bit word. Including to stuff which is is definitely not an IP (instruction pointer), but data.

**Suggestion for improvement**: the noise might get reduced if the decoding is applied by `espflash` only on 32bit words which start from `0x4000000`. If I'm not mistaken, ESP programs start from this address.

ivmarkov, 2024-01-09:
> Experiment 3: Set RUST_BACKTRACE=1

There **IS** a difference! :)

Snippet from experiment 2:
```
thread 'main' panicked at src/main.rs:11:5:
oh no
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x4202ac81 on core 0
```

Snippet from experiment 3:
```
thread 'main' panicked at src/main.rs:11:5:
oh no
stack backtrace:

abort() was called at PC 0x420444b1 on core 0
```

Where originally it used to say
```
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

now it says:
```
stack backtrace:
```

`stack backtrace:` is coming from the Rust stack walker in the `panic` crate. It is "just" that the stack trace generated by the `panic` crate is completely empty. Well, apparently the thing does not work with riscv frame pointers enabled (unlike `esp_backtrace`!). It does work with `xtensa` frame pointers enabled though. Weird. Maybe we need to upstream a patch into it!

igrr, 2024-01-09:
> espflash crawls though the stack memory and blindly applies decoding to everything which looks like a 32 bit word. Including to stuff which is is definitely not an IP (instruction pointer), but data.

Another idea could be to run GDB and actually decode the backtrace based on the stack and the registers, like https://github.com/espressif/esp-idf-panic-decoder/blob/master/esp_idf_panic_decoder/gdb_panic_server.py does.

t-moe, 2024-01-09:
> **UPDATE**: Seems you are running with level "z" (optimize for size, but still inline). What happens if you change this to 0?
> **UPDATE 2**: Not clear from the no_std `Cargo.toml`, but it seems you are running with opt-level 0 (in the absence of anything else specified)? If so, this is not apples-to-apples comparison, and would explain the differences in the output from `esp_backtrace`

Aaah, good catch. :tada: 
 This at least fixes the esp_backtrace case when used in combination with `std::panic::set_hook`

ivmarkov, 2024-01-09:
> Conclusion
> Coming from esp-hal, I find the backtraces on esp-idf-sys extremely painful to handle. They contain a lot of noise, and unless i enable DWARF generation, they will not contain any of my [...]

Let's unpack this a bit for the sake of being precise:
- Situation only an issue with riscv chips, not xtensa (ok, pretty bad as everything new is riscv anyway)
- Noisiness comes from the unfortunate combination of  `espflash monitor` eagerly applying decoding on everything which looks like a memory address + the raw memory dump generated by ESP IDF on riscv*. Not specific to ESP IDF, or `esp-idf-sys` per se

> How can we improve the situation?

Let's put the "noisiness" problem aside for the moment.
Let's look at the three supported stack-walking schemes:

1. DWARF based (solved for me)
- I think this works. Thanks again to @Vollbrecht for `"-C", "force-unwind-tables=yes"`

**Action item:** document this approach better somewhere

2. riscv frame-pointer based 
- I think that can be made to work for Rust code
- I don't think we can make it work for GCC, as I have been unable to force riscv frame pointers with GCC
- I'm not sure what would happen with stack walking if you have a mixed Rust/C stack, where some frames have frame pointers, and others - don't. If this confuses the stack walker, "I think that can be made to work for Rust code" would be an exaggeration for code that calls C and then the C code calls back Rust code

**Action item**: We can try to upstream the `esp_backtrace` code in Rust's `unwind` crate. But only if "it is worth it", i.e. mixed Rust/C stacks can be at least partially decoded (at least the Rust part)

3. GDB decoding of the memory stack frame
- Will "just work" for ESP IDF with default settings
- Needs us to resurrect the project that wrapped the `idf.py` tooling (`cargo espidf ...`)
OR
- Needs us to enhance `espflash` with what amounts to a port of the Python code that @igrr mentioned above. Seems to me a non-trivial effort

**Action item**: Work on the first or second option

> When printing with esp_backtrace, why dont I see my rust functions a,b and c, when using esp-idf-sys instead of esp-hal. Have my frame pointers suddenly disappeared?

Different optimization settings, as per above?

**The noisiness problem**

I think we should postpone this for a while, until we figure out whether we need to concentrate on (2) or (3).

t-moe, 2024-01-09:
Oops. I was a bit too eagerly creating https://github.com/esp-rs/espflash/issues/543, seeing your conclusion here ^^. 

Can you take over, and create the remaining issues/tasks?

Vollbrecht, 2024-01-09:
i think for gdb and stuff as long as you are not talking about the gdb-stub over uart rather connecting external gdb server, we already have a working solution with `probe-rs`. So it think we don't need espflash for that

ivmarkov, 2024-01-09:
And given that the whole stack memory + registers' dump is so ESP IDF specific, perhaps it would be easier and more natural to just find a way to call `idf.py monitor` with a cargo sub-command like `cargo espidf`. Especially given that we have the ESP IDF and all its tooling (we can also force the GDB tool to always be auto-installed by `esp-idf-sys`) in the `.embuild` directory.

Which brings back the question/trick from some time ago which is thast we have to have at least one successful build of `esp-idf-sys` so that we get as an output from it all build settings, *including the location of ESP IDF and its tooling, as utilized by the project*. This task was in the context of `cargo espidf menuconfig` but it is basically the same for the future `cargo espidf monitor` that we are discussing here.

ivmarkov, 2024-01-09:
> Oops. I was a bit too eagerly creating [esp-rs/espflash#543](https://github.com/esp-rs/espflash/issues/543), seeing your conclusion here ^^.
> 
> Can you take over, and create the remaining issues/tasks based on your action items?

I think we can keep it simple and have just this task/issue for now. I would only create a separate task for the documentation of the DWARF "trick" in the Rust ESP book.