esp-rs/esp-idf-sys #276: Integration with `bevy_ecs` - linker failure `pthread_setcancelstate is not implemented`

mrchantey, 2024-01-11:
I'm looking to use [bevy_ecs](https://crates.io/crates/bevy_ecs) with `esp-idf` but am getting linker errors when I add the `bevy_ecs` dependency. I'm using the `esp-idf-template` and have tried with both Windows and the dev container.

The errors are all along the lines of `pthread_attr_init is not implemented and will always fail`

`bevy_ecs` is the lightweight ecs architecture that drives the bevy engine, I thought it might 'just work' with esp-idf because it officially supports wasm which is also a restricted environment.

my setup

```
default std esp-idf-template
device: esp32-c3
ESP_IDF_VERSION = "v5.1.1"

[dependencies]
bevy_ecs = { version = "0.12.1" }
esp-idf-svc = { version = "0.47.3", default-features = false }
```

main.rs

```rust
use bevy_ecs::prelude::World;

fn main() {
	esp_idf_hal::sys::link_patches();
	let _world = World::new();
}
```

error

```
error: linking with `ldproxy` failed: exit code: 1

  = note: [ldproxy] Running ldproxy
          Error: Linker SOME_PATH/.rustup/toolchains/esp/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe failed: exit code: 1
          STDERR OUTPUT:

SOME_PATH/.embuild/espressif/esp-idf/v5.1.1/components/pthread/pthread.c:769: warning: pthread_attr_init is not implemented and will always fail

SOME_PATH/riscv32-esp-elf/bin/ld.exe: SOME_PATH/riscv32-esp-elf/lib/rv32imc_zicsr_zifencei/ilp32/no-rtti\libnosys.a(pthread.o): in function `pthread_attr_init':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/libgloss/libnosys/pthread.c:102: multiple definition of `pthread_attr_init'; esp-idf\pthread\libpthread.a(pthread.c.obj): 
SOME_PATH/.embuild/espressif/esp-idf/v5.1.1/components/pthread/pthread.c:756: first defined here

... The same for lots of other similar definitions, ie pthread_attr_setdetachstate
```

MabezDev, 2024-01-11:
The `pthread_attr_*` functions in newlib are not implemented in the espidf newlib port. It's probably quite a lot of work to add them, maybe a better idea would be to find out why bevy_ecs depends on these functions, and see if it can be cfg'd away for `target_os = "espidf"`. I'm not at all familiar with bevy, so I probably cannot help here, sorry :(.

MabezDev, 2024-01-11:
Maybe a start would be to disable the multithreading feature: https://github.com/bevyengine/bevy/blob/c4e479a2d4f2ca1d7ce2637e4da1a4bb5cbed133/crates/bevy_ecs/Cargo.toml#L14

MabezDev, 2024-01-11:
Huh, I guess I might have found how wasm32 works: https://github.com/bevyengine/bevy/blob/c4e479a2d4f2ca1d7ce2637e4da1a4bb5cbed133/crates/bevy_ecs/src/query/par_iter.rs#L112-L125. On wasm32 it defaults to single thread operation, maybe they would accept a PR which adds `target_os = "espidf"` too.

mrchantey, 2024-01-11:
Thanks yeah looks like it should also go for that operation when `multi-threading` is disabled, but still get the error even with super explicit feature disabling:
```cargo.toml
[dependencies]
bevy_ecs = { version = "0.12.1", default-features = false, features = [] }
bevy_tasks = { version = "0.12.1", default-features = false, features = [] }
```

mrchantey, 2024-01-11:
Ok it looks like this is a duplicate of [esp-idf-template/178](https://github.com/esp-rs/esp-idf-template/issues/178)

WORKAROUND: Remove -C default-linker-libraries from .cargo/config.toml.

^^ I can confirm this was the solution for my issue too.

closed 2024-01-11 by mrchantey