esp-rs/esp-idf-sys #278: Misaligned struct pointer from esp camera

ViktorWb, 2024-01-19:
Hi! I activated esp-camera using

```toml
[package.metadata.esp-idf-sys]
extra_components = [
  { component_dirs = ["component"], bindings_header = "bindings.h", bindings_module = "cam" },
]
```

in my Cargo.toml. This worked just fine. Then I tried to do:

```rust
esp_idf_sys::cam::esp_camera_init(&camera_config)
let fb = esp_idf_sys::cam::esp_camera_fb_get();
std::slice::from_raw_parts((*fb).buf, (*fb).len)
```

Which also works fine, but only in release mode. When running in debug mode I get

`misaligned pointer dereference: address must be a multiple of 0x8 but is 0x3f801fc4`

The issue is that the `fb` variable is a pointer to a 4-bit aligned struct, while rust requires an 8-bit aligned struct. The struct in question is the `camera_fb_t`. My understanding is that the alignment checks are skipped in release mode, but that it can cause undefined behaviour.

In the generated bindings, the `camera_fb_t` struct has an alignment of 8 and is defined as

```rust
#[doc = " @brief Data structure of camera frame buffer"]
#[repr(C)]
#[derive(Debug, Copy, Clone)]
pub struct camera_fb_t {
    #[doc = "< Pointer to the pixel data"]
    pub buf: *mut u8,
    #[doc = "< Length of the buffer in bytes"]
    pub len: usize,
    #[doc = "< Width of the buffer in pixels"]
    pub width: usize,
    #[doc = "< Height of the buffer in pixels"]
    pub height: usize,
    #[doc = "< Format of the pixel data"]
    pub format: pixformat_t,
    #[doc = "< Timestamp since boot of the first DMA buffer of the frame"]
    pub timestamp: timeval,
}
```

ivmarkov, 2024-01-20:
> The issue is that the fb variable is a pointer to a 4-bit aligned struct, while rust requires an 8-bit aligned struct. The struct in

What architecture is that? riscv? xtensa? In any case, either of these requiring 8 byte aligned struct sounds impossible, as these are 32bit CPUs.

ivmarkov, 2024-01-20:
> The issue is that the fb variable is a pointer to a 4-bit aligned struct, while rust requires an 8-bit aligned struct. 
You sure it is about the `fb` variable? It could actually be the `(*fb).buf` variable. On which line **exactly** do you get the falure?

ivmarkov, 2024-01-20:
> In the generated bindings, the camera_fb_t struct has an alignment of 8 and is defined as

Where do you see this? All I see is `#[repr(C)]` which might just as well mean 4 byte aligned struct?

ivmarkov, 2024-01-20:
Also good to mention:
* ESP IDF version
* Rust version

ivmarkov, 2024-01-20:
Looking [here](https://www.reddit.com/r/rust/comments/r6a16i/pointer_to_reference_alignment/) it seems *the alignment depends on what members the structure or array has*. So it can't be the `buf` thing, as it has `u8`s = alignment 1.

`camera_fb_t` on the other hand, has `timestamp` of type `timeval`. `timeval` is interesting, because - putting aside `pixformat_t` which I don't know what it is - `timeval` is likely the biggest member of the struct, which would dictate the alignment of the struct. now:
* In ESP IDF < 5, `timeval` used to be 4 bytes
* In ESP IDF >= 5, `timeval` is now 8 bytes (hence why we have the pesky `espidf_time64` flag when building under ESP IDF 5.

Strange. It looks to me as some sort of impossible miscompilation happened, where `camera_fb_t` at runtime is with 4 bytes timeval, while Rust thinks it is with 8 bytes timeval. What ESP IDF version are you using again?

ViktorWb, 2024-01-20:
Thank you for your input! The following code

```rust
let fb: *mut esp_idf_sys::cam::camera_fb_t = esp_idf_sys::cam::esp_camera_fb_get();
println!("Alignment: {}", std::mem::align_of::<esp_idf_sys::cam::camera_fb_t>());
println!("Pointer: {:p}", fb);
let deref = *fb;
```

prints

```
Alignment: 8
Pointer: 0x3f801fc4
```

And then panics on the line `let deref = *fb` (0x3f801fc4 is not a multiple of 8).

In my .cargo/config.toml I have `ESP_IDF_VERSION = "v5.1.1"`, and in my .embuild/esp-idf I have a folder called v5.1.1, and running `git describe` in that folder gives `v5.1.1`. I can't really find anything with version 4.

ivmarkov, 2024-01-21:
Can you try with ESP IDF V4.4? I'm not suggesting a permanent downgrade, just terying to figure out if `timeval` is somehow the root of it all (it is 32 bit in 4.4).

ivmarkov, 2024-01-21:
`v4.4.6`. And you need to remove `--cfg espidf_time64` from `.cargo/config.toml` - and then full rebuild.

igrr, 2024-01-21:
Does Rust generally require that all data types are naturally aligned (i.e. aligned to their own size) or is this just the way the Xtensa target was defined?

For example, way GCC is configured on Xtensa is that 2- and 4-byte types are naturally aligned, while larger types are aligned to 4-byte boundaries.

ivmarkov, 2024-01-21:
> Does Rust generally require that all data types are naturally aligned (i.e. aligned to their own size) or is this just the way the Xtensa target was defined?

Great question. I don't know but I *assume* rustc would generally follow the alignment specified in the target. Which would mean that there is probably something wrong [here](https://github.com/esp-rs/rust/blob/esp-1.75.0.0/compiler/rustc_target/src/spec/targets/xtensa_esp32_espidf.rs#L8) w.r.t. types > 32bit for the ESP IDF target, as well as [here](https://github.com/esp-rs/rust/blob/esp-1.75.0.0/compiler/rustc_target/src/spec/targets/xtensa_esp32_none_elf.rs#L8) - for the baremetal ESP32 target (and likely the same for s2 and s3). 

According to [docu](https://stackoverflow.com/questions/67888518/how-do-i-understand-the-data-layout-strings-of-rust-compiler-targets), these are LLVM alignments, actually.

Let me try to decipher this to figure out what it does say for types > 32 bits (if anything).

> For example, the way GCC is configured on Xtensa is that 2- and 4-byte types are naturally aligned, while larger types are aligned to 4-byte boundaries.

ViktorWb, 2024-01-21:
On `v.4.46`, the alignment of `camera_fb_t` becomes `4`, and the problem appears to be gone.

In the  document [Type layout](https://doc.rust-lang.org/reference/type-layout.html):

> Most primitives are generally aligned to their size, although this is platform-specific behavior. In particular, on x86 u64 and f64 are only aligned to 32 bits.

ViktorWb, 2024-01-21:
Turns out there are a bunch of different `time_t` definitions, and I was hoping that a discrepancy somewhere could have caused the issue. But all of the below gives `8`.

```rust
println!("i64 alignment: {}", std::mem::align_of::<i64>());
println!("time_t alignment: {}", std::mem::align_of::<esp_idf_sys::time_t>());
println!("esp32-camera time_t alignment: {}", std::mem::align_of::<esp_idf_sys::cam::time_t>());
println!("esp OS time_t alignment: {}", std::mem::align_of::<std::os::espidf::raw::time_t>());
println!("esp_idf_svc time_t alignment: {}", std::mem::align_of::<esp_idf_svc::sys::time_t>());
println!("ffi long long alignment: {}", std::mem::align_of::<std::ffi::c_longlong>());
```

ivmarkov, 2024-01-21:
> The timeval struct is defined as
> 
> ```rust
> #[repr(C)]
> #[derive(Debug, Default, Copy, Clone)]
> pub struct timeval {
>     pub tv_sec: time_t,
>     pub tv_usec: suseconds_t,
> }
> ```
> 
> Turns out there are a bunch of different `time_t` definitions, and I was hoping that a discrepancy somewhere could have caused the issue. But all of the below gives `8`.
> 
> ```rust
> println!("i64 alignment: {}", std::mem::align_of::<i64>());
> println!("time_t alignment: {}", std::mem::align_of::<esp_idf_sys::time_t>());
> println!("esp32-camera time_t alignment: {}", std::mem::align_of::<esp_idf_sys::cam::time_t>());
> println!("esp OS time_t alignment: {}", std::mem::align_of::<std::os::espidf::raw::time_t>());
> println!("esp_idf_svc time_t alignment: {}", std::mem::align_of::<esp_idf_svc::sys::time_t>());
> println!("ffi long long alignment: {}", std::mem::align_of::<std::ffi::c_longlong>());
> ```

Did you re-test this with ESP IDF < 5? On ESP IDF < 5 it should give you 4, not 8.

ivmarkov, 2024-01-21:
See [here](https://github.com/rust-lang/libc/blob/40741baa1d892518fd3c39795e962058ff558fb9/src/unix/newlib/mod.rs#L49). ESP IDF < 4 does not have `espidf_time64` defined when compiling with rustc.

ViktorWb, 2024-01-21:
> Did you re-test this with ESP IDF < 5? On ESP IDF < 5 it should give you 4, not 8.

Indeed for `v.4.4.6` it gives

```
i64 alignment: 8
time_t alignment: 4
esp32-camera time_t alignment: 4
esp OS time_t alignment: 4
esp_idf_svc time_t alignment: 4
ffi long long alignment: 8
```

ivmarkov, 2024-01-21:
@igrr You are totally right!

In [this spec](https://github.com/esp-rs/rust/blob/esp-1.75.0.0/compiler/rustc_target/src/spec/targets/xtensa_esp32_espidf.rs#L8) we have for i64: `i64:64`, and for i128: `i128:128`. I think we should instead have - for compat with GCC - 
* i64: `i64:32`
* i128: `i128:32`

@MabezDev I think we should fix this in all `xtensa` targets.

ivmarkov, 2024-01-21:
@ViktorWb Can you try compiling **on ESP IDF 5+** with this JSON target instead
(I generated it with `rustc -Z unstable-options --print target-spec-json --target xtensa-esp32-espidf` and then applied the fix from above ^^^):

```json
{
  "arch": "xtensa",
  "cpu": "esp32",
  "crt-objects-fallback": "false",
  "data-layout": "e-m:e-p:32:32-i64:32-i128:32-n32",
  "emit-debug-gdb-scripts": false,
  "env": "newlib",
  "linker": "xtensa-esp32-elf-gcc",
  "linker-flavor": "gnu-cc",
  "llvm-target": "xtensa-none-elf",
  "max-atomic-width": 32,
  "os": "espidf",
  "panic-strategy": "abort",
  "relocation-model": "static",
  "target-family": [
    "unix"
  ],
  "target-pointer-width": "32",
  "vendor": "espressif"
}
```

Here, [the relevant info](https://docs.rust-embedded.org/embedonomicon/custom-target.html) w.r.t. custom targets.

ViktorWb, 2024-01-21:
Sure! I am getting some errors.

I can describe my process.

I created a file called `xtensa-esp32-espidf.json` in the root of my project, and added the JSON data you provided. Then, I changed `.cargo/config.toml` from

```
[build]
target = "xtensa-esp32-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = ["--cfg", "espidf_time64"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32"
ESP_IDF_VERSION = "v5.1.1"
```
to
```
[build]
target = "xtensa-esp32-espidf.json"

[target.xtensa-esp32-espidf.json]
linker = "ldproxy"
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = "--cfg espidf_time64"

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32"
ESP_IDF_VERSION = "v5.1.1"
```

(with `rustflags = ["--cfg", "espidf_time64"]` I am getting error `error: expected a string, but found a array for `rustflags``).

Getting the following:

```
error[E0308]: mismatched types
  --> /home/viktor/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.7/src/lib.rs:50:62
   |
50 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /home/viktor/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-sys-0.33.7/src/lib.rs:52:51
   |
52 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.33.7: The linker arguments force the usage of `ldproxy` but the linker used by cargo is different. Please set the linker to `ldproxy` in your cargo config or set `force_ldproxy` to `false`.
```

ivmarkov, 2024-01-21:
Alright! I'll PR the change to the Rust fork delivered with `rustup`. Until there is a new release of the toolchain with `rustup`, please use the JSON target.

ViktorWb, 2024-01-21:
Fantastic, thank you!

ivmarkov, 2024-01-21:
Closing as we now have [this PR opened](https://github.com/esp-rs/rust/pull/211)

closed 2024-01-21 by ivmarkov

ivmarkov, 2024-01-21:
Just closed my PR because - as it turned out - there is an earlier (and more correct in that it treats the `f64` type too) PR  [here](https://github.com/esp-rs/rust/pull/195).