esp-rs/esp-idf-sys #279: could not identify the root crate and `ESP_IDF_SYS_ROOT_CRATE` not specified

nOrOuzi1993, 2024-01-29:
I'm using workspace and when I want to build one of my binary packages from workspace directory I'll face to these errors: 
`
Compiling e200 v0.1.0 (/dc_fw/platforms/esp32/xxxxx)
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.33.7: could not identify the root crate and `ESP_IDF_SYS_ROOT_CRATE` not specified.

error: failed to run custom build command for `esp-idf-sys v0.33.7`

Caused by:
  process didn't exit successfully: `/dc_fw/target/release/build/esp-idf-sys-ed2a604075017141/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:warning=could not identify the root crate and `ESP_IDF_SYS_ROOT_CRATE` not specified

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: None,
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: Unsupported target 'x86_64-unknown-linux-gnu'
make: *** [Makefile:5: Build_E200_ESP] Error 101
make: *** [Makefile:23: cbee200] Error 2
`
If I won't put this crate to workspace directory everything will work correctly.
I tried to add "export ESP_IDF_SYS_ROOT_CRATE=xxxxx" and getting following errors:
`   Compiling esp-idf-sys v0.33.7
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.33.7: could not read build config from manifest metadata: Matching variant not found

error: failed to run custom build command for `esp-idf-sys v0.33.7`

Caused by:
  process didn't exit successfully: `/dc_fw/target/release/build/esp-idf-sys-ed2a604075017141/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:warning=could not read build config from manifest metadata: Matching variant not found

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: Some(
          [
              "/dc_fw/platforms/esp32/E200/sdkconfig.defaults",
          ],
      ),
      mcu: None,
      native: NativeConfig {
          esp_idf_version: None,
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: Some(
          "xxxxx",
      ),
  }
  Error: Unsupported target 'x86_64-unknown-linux-gnu'
`

nOrOuzi1993, 2024-02-02:
> Where is your `.cargo/config.toml`? I think it must be in the workspace root, and that might be the reason why it fails for you. As in here: https://github.com/ivmarkov/ruwm/tree/master/.cargo

yes, thank you, `config.toml `was problem but one of them. I have multiple platforms on this repository and don't want to put esp32 files on root directory, so I have to change `sdkconfig` and others path too.

closed 2024-02-02 by nOrOuzi1993

1cedsoda, 2024-05-30:
> yes, thank you, `config.toml `was problem but one of them. I have multiple platforms on this repository and don't want to put esp32 files on root directory, so I have to change `sdkconfig` and [...]

Can you explain how you solved this?