esp-rs/esp-idf-sys #285: Trying to set CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=1 not working

empirephoenix, 2024-02-17:
Hi,
I'm currently trying to get ota to work correctly regarding rollbacks.

I can flash a new image easily enough and in the end the update process calls esp_ota_set_boot_partition() to the new partition and then call esp_restart()

The esp start correctly new, but reports ESP_OTA_IMG_UNDEFINED using this rather crude test code, if I understood the idf documentation correctly, it should instead show ESP_OTA_IMG_PENDING_VERIFY?

`    let mut ota_state: esp_ota_img_states_t = 0;
    unsafe {
        let running_partition = esp_ota_get_running_partition();
        println!("Partition address is {}", (*running_partition).address);

        esp_ota_get_state_partition(running_partition, &mut ota_state);
        if ota_state == esp_ota_img_states_t_ESP_OTA_IMG_NEW {
            println!("Partition state is {}", "ESP_OTA_IMG_NEW");
        } else if ota_state == esp_ota_img_states_t_ESP_OTA_IMG_PENDING_VERIFY {
            println!("Partition state is {}", "ESP_OTA_IMG_PENDING_VERIFY");
        } else if ota_state == esp_ota_img_states_t_ESP_OTA_IMG_VALID {
            println!("Partition state is {}", "ESP_OTA_IMG_VALID");
        } else if ota_state == esp_ota_img_states_t_ESP_OTA_IMG_INVALID {
            println!("Partition state is {}", "ESP_OTA_IMG_INVALID");
        } else if ota_state == esp_ota_img_states_t_ESP_OTA_IMG_ABORTED {
            println!("Partition state is {}", "ESP_OTA_IMG_ABORTED");
        } else if ota_state == esp_ota_img_states_t_ESP_OTA_IMG_UNDEFINED {
            println!("Partition state is {}", "ESP_OTA_IMG_UNDEFINED");
        } else {
            println!("Partition state is {}", ota_state);
        }
    }`

In the project I have a sdkconfig.defaults, that is apparently used correctly apart from the ignored rollback, since I increase the stack to 20kb and also have a check for it directly in main, so that any firmware compiled wrong should be rolled back instantly
`
    if esp_idf_sys::CONFIG_MAIN_TASK_STACK_SIZE < 20000 {
        error!(
            "stack too small: {} bail!",
            esp_idf_sys::CONFIG_MAIN_TASK_STACK_SIZE
        );
        return Ok(());
    }`
   
I do not know, if the bootloader needs to be manually reflashed if I change that option nor if I already flash it,
initially for the first upload I use 
`cargo espflash flash --monitor --partition-table partitions.csv`
the image for the ota is build using:
`cargo espflash save-image --chip esp32 image.bin`

Do I need to to something else or did I not understand how the rollback does work? Any help would be appreciated.

Vollbrecht, 2024-02-18:
did you try with "CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y" i think you should use "=y" or "=n"

empirephoenix, 2024-02-19:
Duh, thats it, it now reports ESP_OTA_IMG_NEW thank you :)

closed 2024-02-19 by empirephoenix