esp-rs/esp-idf-sys #287: esp-idf-sys build started fail on arm mac os

mbv, 2024-02-21:
esp-idf-sys build started fail on arm mac os

esp-idf-sys `v0.34.0`
target `esp32`

<details>
  <summary>Log</summary>

```
        Fresh core v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
       Fresh unicode-ident v1.0.12
       Fresh memchr v2.7.1
       Fresh rustc-std-workspace-core v1.99.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-core)
       Fresh aho-corasick v1.1.2
       Fresh regex-syntax v0.8.2
       Fresh compiler_builtins v0.1.103
       Fresh libc v0.2.150
       Fresh proc-macro2 v1.0.78
       Fresh regex-automata v0.4.5
       Fresh bitflags v1.3.2
       Fresh alloc v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/alloc)
       Fresh adler v1.0.2
       Fresh memchr v2.5.0
       Fresh quote v1.0.35
       Fresh rustc-demangle v0.1.23
       Fresh glob v0.3.1
       Fresh errno v0.3.8
       Fresh crossbeam-utils v0.8.19
       Fresh rustc-std-workspace-alloc v1.99.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-alloc)
       Fresh unwind v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/unwind)
       Fresh panic_abort v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_abort)
       Fresh syn v2.0.50
       Fresh bitflags v2.4.2
       Fresh syn v1.0.109
       Fresh gimli v0.28.0
       Fresh miniz_oxide v0.7.1
       Fresh hashbrown v0.14.3
       Fresh std_detect v0.1.5 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/stdarch/crates/std_detect)
       Fresh object v0.32.1
       Fresh panic_unwind v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_unwind)
       Fresh serde_derive v1.0.197
       Fresh rustix v0.38.31
       Fresh addr2line v0.21.0
       Fresh crossbeam-epoch v0.9.18
       Fresh bstr v1.9.0
       Fresh same-file v1.0.6
       Fresh minimal-lexical v0.2.1
       Fresh rustversion v1.0.14
       Fresh thiserror-impl v1.0.57
       Fresh std v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
       Fresh globset v0.4.14
       Fresh crossbeam-deque v0.8.5
       Fresh walkdir v2.4.0
       Fresh nom v7.1.3
       Fresh nix v0.26.4
       Fresh cvt v0.1.2
       Fresh libloading v0.8.1
       Fresh proc_macro v0.0.0 (/Users/mbv/.rustup/toolchains/esp/lib/rustlib/src/rust/library/proc_macro)
       Fresh home v0.5.9
       Fresh ryu v1.0.17
       Fresh heck v0.4.1
       Fresh lazy_static v1.4.0
       Fresh itoa v1.0.10
       Fresh either v1.10.0
       Fresh thiserror v1.0.57
       Fresh which v4.4.2
       Fresh serde_json v1.0.114
       Fresh strum_macros v0.24.3
       Fresh clang-sys v1.7.0
       Fresh fs_at v0.1.10
[... 278 lines cut ...]
      Run Build Command(s):/Users/mbv/test-build-app/.embuild/espressif/tools/ninja/1.10.2/ninja cmTC_4bab5 && [1/2] Building C object CMakeFiles/cmTC_4bab5.dir/testCCompiler.c.obj
      FAILED: CMakeFiles/cmTC_4bab5.dir/testCCompiler.c.obj 
      /Users/mbv/test-build-app/.embuild/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections --target=xtensa-esp32-espidf -o CMakeFiles/cmTC_4bab5.dir/testCCompiler.c.obj -c /Users/mbv/test-build-app/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cdeb9cbd12aa4ef9/out/build/CMakeFiles/CMakeTmp/testCCompiler.c
      xtensa-esp32-elf-gcc: error: unrecognized command-line option '--target=xtensa-esp32-espidf'
      ninja: build stopped: subcommand failed.
      
      

    

    CMake will not be able to correctly generate this project.
  Call Stack (most recent call first):
    /Users/mbv/test-build-app/.embuild/espressif/esp-idf/v5.1.2/tools/cmake/project.cmake:448 (__project)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/mbv/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

</details>

NE4Y, 2024-02-23:
I have the same problem.

Vollbrecht, 2024-02-24:
2 weeks ago the `cc` crate got some updates, and since then there were made multiple releases that fixed problems on macos, but it may as well still be broken. To test the hypothesis could one of you test it with the following.

add this in your Cargo.toml
```toml
[patch.crates-io]
cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }
```
than run `cargo update` to fetch the git dependencys, and then run `cargo tree -f "{p} {f}" | grep cc -B 6` to check if the `cc` dependency is acutally the 1.0.83 version.

Vollbrecht, 2024-02-24:
if `cc` is still the culprit with any version from `1.0.84` to freshly released  `1.0.87` we need to raise a issue on the `cc` crate here

mbv, 2024-02-25:
@Vollbrecht Thank you, `1.0.83` works, but all subsequent versions are broken, the last one I checked was `1.0.88`.

freew01f, 2024-02-26:
thank u very much

> 2 weeks ago the `cc` crate got some updates, and since then there were made multiple releases that fixed problems on macos, but it may as well still be broken. To test the hypothesis could one of [...]

thkans

dtimm, 2024-02-26:
Building with `CRATE_CC_NO_DEFAULTS=1` gets past this issue for me. This stops the script from adding the `--target=xtensa-esp32-espidf` flag that upsets `xtensa-esp-elf-gcc`.

kristofferpeterhansel, 2024-02-28:
> add this in your Cargo.toml
> 
> ```toml
> [patch.crates-io]
> cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }
> ```

Just `cc = "=1.0.83"` also works to force a specific version

meditant, 2024-03-05:
[patch.crates-io]
cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }

Not work for me.

cargo clean
cargo build
warning: Patch `cc v1.0.83 (https://github.com/rust-lang/cc-rs?tag=1.0.83#bfc3ba41)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling compiler_builtins v0.1.103
   Compiling core v0.0.0 (/Users/fr/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.150
   Compiling memchr v2.5.0
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling std v0.0.0 (/Users/fr/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling libc v0.2.153
   Compiling memchr v2.7.1
   Compiling cfg-if v1.0.0
   Compiling serde v1.0.197
   Compiling regex-syntax v0.8.2
   Compiling crossbeam-utils v0.8.19
   Compiling glob v0.3.1
   Compiling syn v1.0.109
   Compiling rustix v0.38.31
   Compiling bitflags v1.3.2
   Compiling bitflags v2.4.2
   Compiling log v0.4.21
   Compiling rustversion v1.0.14
   Compiling same-file v1.0.6
   Compiling serde_json v1.0.114
   Compiling thiserror v1.0.57
   Compiling minimal-lexical v0.2.1
   Compiling walkdir v2.5.0
   Compiling libloading v0.8.2
   Compiling cvt v0.1.2
   Compiling either v1.10.0
   Compiling lazy_static v1.4.0
   Compiling bindgen v0.63.0
   Compiling anyhow v1.0.80
   Compiling itoa v1.0.10
   Compiling ryu v1.0.17
   Compiling home v0.5.9
   Compiling heck v0.4.1
   Compiling fastrand v2.0.1
   Compiling peeking_take_while v0.1.2
   Compiling shlex v1.3.0
   Compiling normpath v1.2.0
   Compiling rustc-hash v1.1.0
   Compiling lazycell v1.3.0
   Compiling cc v1.0.89
   Compiling camino v1.1.6
   Compiling semver v1.0.22
   Compiling fnv v1.0.7
   Compiling autocfg v1.1.0
   Compiling ident_case v1.0.1
   Compiling clang-sys v1.7.0
   Compiling embedded-io-async v0.6.1
   Compiling heapless v0.8.0
   Compiling core-foundation-sys v0.8.6
   Compiling embassy-time-driver v0.1.0
   Compiling unicode-xid v0.2.4
   Compiling once_cell v1.19.0
   Compiling aho-corasick v1.1.2
   Compiling bstr v1.9.1
   Compiling nom v7.1.3
   Compiling quote v1.0.35
[... 192 lines cut ...]

  thread 'main' panicked at /Users/fr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  stack backtrace:
     0: _rust_begin_unwind
     1: core::panicking::panic_fmt
     2: cmake::fail
               at /Users/fr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5
     3: cmake::run
               at /Users/fr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1076:9
     4: cmake::Config::build
               at /Users/fr/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:813:13
     5: build_script_build::native::cargo_driver::build
               at ./build/native/cargo_driver.rs:528:5
     6: build_script_build::native::build
               at ./build/native.rs:17:9
     7: build_script_build::main
               at ./build/build.rs:51:24
     8: core::ops::function::FnOnce::call_once
               at /Users/espressif/action-runner/esp-rs/_work/rust-build/rust-build/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...

Vollbrecht, 2024-03-05:
look at the output : `Compiling cc v1.0.89` you did not apply the patch correctly, you probably forgot to run `cargo update`

samutigro, 2024-03-08:
> 2 weeks ago the `cc` crate got some updates, and since then there were made multiple releases that fixed problems on macos, but it may as well still be broken. To test the hypothesis could one of [...]

@Vollbrecht this worked for me!! Thank you! I was struggling with it since 10 days

NobodyXu, 2024-03-08:
Can you please open an issue in rust-lang/cc-rs ?

clouds56, 2024-03-10:
FYI, https://github.com/rust-lang/cc-rs/issues/1005

NobodyXu, 2024-03-10:
It seems that setting `cc::Build::no_default_flags` to false (can be set by environment variables CRATE_CC_NO_DEFAULTS) fix this problem.

I think updating the build script to disable default flags would probably fix this.

clouds56, 2024-03-16:
I did some investigation, the break change is introduced by https://github.com/rust-lang/cc-rs/pull/709.
After the PR, the could detect `cc` in PATH is actually `clang` in macOS, and then add some default flags for clang in build.
I think we should ether 
1. tell `cc-rs` we do using gcc (something like `TARGET_CC="riscv32-esp-elf-gcc"` and `TARGET_CXX="riscv32-esp-elf-g++"`) 
2. or like @NobodyXu said set `CRATE_CC_NO_DEFAULTS=1`

Further more, I checked CMakeCache.txt in the demo esp-idf project, related CFLAGS shows like:
```
//C Compiler Base Flags
CMAKE_C_FLAGS:STRING='-march=rv32imc_zicsr_zifencei '

//Flags used by the C compiler during DEBUG builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the C compiler during MINSIZEREL builds.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the C compiler during RELEASE builds.
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the C compiler during RELWITHDEBINFO builds.
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//Path to a program.
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND

//Linker Base Flags
CMAKE_EXE_LINKER_FLAGS:STRING='-nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs '

//Flags used by the linker during DEBUG builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
```

this is CMakeCache.txt in `esp-idf-sys` when `CC=clang` (this is default `cc-rs=v1.0.84`, and build failed):
```
//C Compiler Base Flags
CMAKE_C_FLAGS:STRING=-march=rv32imc_zicsr_zifencei  -ffunction-sections -fdata-sections --target=riscv32imc_zicsr_zifencei-esp-espidf

//Flags used by the C compiler during DEBUG builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the C compiler during MINSIZEREL builds.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the C compiler during RELEASE builds.
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the C compiler during RELWITHDEBINFO builds.
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//Path to a program.
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND

//Linker Base Flags
CMAKE_EXE_LINKER_FLAGS:STRING='-nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs '

//Flags used by the linker during DEBUG builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
```

and when `TARGET_CC=riscv32-esp-elf-gcc` (build successfully):
```
//C Compiler Base Flags
CMAKE_C_FLAGS:STRING=-march=rv32imc_zicsr_zifencei  -ffunction-sections -fdata-sections -mabi=ilp32

//Flags used by the C compiler during DEBUG builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the C compiler during MINSIZEREL builds.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the C compiler during RELEASE builds.
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the C compiler during RELWITHDEBINFO builds.
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//Path to a program.
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND

//Linker Base Flags
CMAKE_EXE_LINKER_FLAGS:STRING='-nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs '

//Flags used by the linker during DEBUG builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
```
and with `CRATE_CC_NO_DEFAULTS=1` (build successfully):
```
//C Compiler Base Flags
CMAKE_C_FLAGS:STRING='-march=rv32imc_zicsr_zifencei '

//Flags used by the C compiler during DEBUG builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the C compiler during MINSIZEREL builds.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the C compiler during RELEASE builds.
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the C compiler during RELWITHDEBINFO builds.
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//Path to a program.
CMAKE_DLLTOOL:FILEPATH=CMAKE_DLLTOOL-NOTFOUND

//Linker Base Flags
CMAKE_EXE_LINKER_FLAGS:STRING='-nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs '

//Flags used by the linker during DEBUG builds.
CMAKE_EXE_LINKER_FLAGS_DEBUG:STRING=
```

Related code position:
// detect_family called from here
https://github.com/esp-rs/esp-idf-sys/blob/3b9fb74db678baa30de8cca722b7c4bc4e00437c/build/native/cargo_driver.rs#L527-L528

// TARGET_CC checked at here
https://github.com/rust-lang/cc-rs/blob/77a54a990259b85a1385495594aa6dd188c7c4e8/src/lib.rs#L2697-L2699

NobodyXu, 2024-03-16:
> I did some investigation, the break change is introduced by rust-lang/cc-rs#709.

If that is the case, then rust-lang/cc-rs#1000 should fix this.

It fixes the compiler detection logic by using C macros instead of `--version`

NobodyXu, 2024-03-17:
@clouds56 Can you have a try of rust-lang/cc-rs#1000 , to see if it fixed the issue please?

clouds56, 2024-03-18:
> @clouds56 Can you have a try of [rust-lang/cc-rs#1000](https://github.com/rust-lang/cc-rs/pull/1000) , to see if it fixed the issue please?

I've tried current main branch (at commit `f36d6a7ed06033c38021ef65e2b7f1da38932024`)
It failed at a different point, (seems passed the basic compiler test which failed prior to cc-rs#1000)
```
  CMake Error at /Users/clouds/.local/opt/esp-idf/CMakeLists.txt:234 (message):
    Failed to determine sizeof(time_t)
```

And the CMakeCache.txt still uses `--target`
```
//C Compiler Base Flags
CMAKE_C_FLAGS:STRING=-march=rv32imc_zicsr_zifencei  -ffunction-sections -fdata-sections --target=riscv32imc_zicsr_zifencei-esp-espidf
```

You might also interest in the CMakeError.log
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /Users/clouds/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc 
Build flags: -march=rv32imc_zicsr_zifencei;;-ffunction-sections;-fdata-sections;-mabi=ilp32
Id flags:  

The output was:
No such file or directory
```

Firstyear, 2024-04-14:
Same issue here, downgrade to 1.0.83 resolves.

zrll12, 2024-04-21:
I tried adding `cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }`, but still got en error
```
error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `/Users/zrll/Documents/proj/test-esp32-std/target/debug/build/esp-idf-sys-5943b24c3f5971f2/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: None,
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Error: Unsupported target 'aarch64-apple-darwin'
```

Vollbrecht, 2024-04-21:
@zrll12 that seams unrelated to to issue, you are building against your host target and not against the specific esp targets. Either you forgot to directly pass the target as an argument or if you are using the .cargo/config.toml files it is missing the correct configuration.

joebnb, 2024-12-12:
```
    Blocking waiting for file lock on build directory
   Compiling esp-idf-sys v0.35.0
error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `/home/esp/lllix_esp32s3/target/debug/build/esp-idf-sys-57d579c3ba5cd32f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.11.2
  pip 24.3.1 from /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
  Requirement already satisfied: pip in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (24.3.1)
  Requirement already satisfied: setuptools in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (71.0.0)
  Collecting setuptools
    Using cached setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
  Using cached setuptools-75.6.0-py3-none-any.whl (1.2 MB)
  Installing collected packages: setuptools
    Attempting uninstall: setuptools
      Found existing installation: setuptools 71.0.0
      Uninstalling setuptools-71.0.0:
        Successfully uninstalled setuptools-71.0.0
  Successfully installed setuptools-75.6.0
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
  Collecting setuptools (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 3))
    Using cached https://dl.espressif.cn/pypi/setuptools/setuptools-71.0.0-py3-none-any.whl (908 kB)
  Requirement already satisfied: packaging in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 4)) (24.2)
  Requirement already satisfied: click in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 7)) (8.1.7)
  Requirement already satisfied: pyserial in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 8)) (3.5)
  Requirement already satisfied: cryptography in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 9)) (39.0.2)
  Requirement already satisfied: pyparsing in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 10)) (3.0.9)
  Requirement already satisfied: pyelftools in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 11)) (0.29)
  Requirement already satisfied: idf-component-manager in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 12)) (2.1.1)
  Requirement already satisfied: esp-coredump in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 13)) (1.12.0)
  Requirement already satisfied: esptool in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 14)) (4.7.0)
  Requirement already satisfied: esp-idf-kconfig in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 15)) (1.5.0)
  Requirement already satisfied: esp-idf-monitor in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 16)) (1.5.0)
  Requirement already satisfied: esp-idf-size in /home/esp/lllix_esp32s3/.embuild/espressif/python_env/idf5.1_py3.11_env/lib/python3.11/site-packages (from -r /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/requirements/requirements.core.txt (line 17)) [... cut ...]
  Call Stack (most recent call first):
    /home/esp/lllix_esp32s3/.embuild/espressif/esp-idf/v5.1.4/tools/cmake/project.cmake:604 (idf_build_process)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/esp/.cargo/registry/src/mirrors.ustc.edu.cn-4affec411d11e50f/cmake-0.1.52/src/lib.rs:1115:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

still have error

joebnb, 2024-12-12:
im run build under official devContainer

noreplydev, 2025-01-11:
Same here, mac os on arm chip.

ivmarkov, 2025-01-11:
> Same here, mac os on arm chip.

If you have issues, pasting here the full build output would be much more helpful.

This issue (with cmake-rs) is fixed since some time, so upgrading to the latest released esp-idf-* crates should resolve this problem.

And then I can finally close it.

noreplydev, 2025-01-11:
Yeah, sure. I'm using `cargo generate esp-rs/esp-idf-template cargo` to generate the project  and this is the output running a release build: 

```
    Updating crates.io index
     Locking 183 packages to latest compatible versions
      Adding bitflags v1.3.2 (latest: v2.7.0)
      Adding cargo_metadata v0.18.1 (latest: v0.19.1)
      Adding embassy-time-driver v0.1.0 (latest: v0.2.0)
      Adding embedded-hal v0.2.7 (latest: v1.0.0)
      Adding globwalk v0.8.1 (latest: v0.9.1)
      Adding heck v0.4.1 (latest: v0.5.0)
      Adding itertools v0.13.0 (latest: v0.14.0)
      Adding linux-raw-sys v0.4.15 (latest: v0.7.0)
      Adding nb v0.1.3 (latest: v1.1.0)
      Adding remove_dir_all v0.8.4 (latest: v1.0.0)
      Adding strum v0.24.1 (latest: v0.26.3)
      Adding strum v0.25.0 (latest: v0.26.3)
      Adding strum_macros v0.24.3 (latest: v0.26.4)
      Adding strum_macros v0.25.3 (latest: v0.26.4)
      Adding syn v1.0.109 (latest: v2.0.96)
      Adding thiserror v1.0.69 (latest: v2.0.11)
      Adding thiserror-impl v1.0.69 (latest: v2.0.11)
      Adding wasi v0.11.0+wasi-snapshot-preview1 (latest: v0.14.0+wasi-0.2.3)
      Adding which v4.4.2 (latest: v7.0.1)
      Adding windows-core v0.52.0 (latest: v0.59.0)
      Adding windows-sys v0.52.0 (latest: v0.59.0)
      Adding windows-targets v0.52.6 (latest: v0.53.0)
      Adding windows_aarch64_gnullvm v0.52.6 (latest: v0.53.0)
      Adding windows_aarch64_msvc v0.52.6 (latest: v0.53.0)
      Adding windows_i686_gnu v0.52.6 (latest: v0.53.0)
      Adding windows_i686_gnullvm v0.52.6 (latest: v0.53.0)
      Adding windows_i686_msvc v0.52.6 (latest: v0.53.0)
      Adding windows_x86_64_gnu v0.52.6 (latest: v0.53.0)
      Adding windows_x86_64_gnullvm v0.52.6 (latest: v0.53.0)
      Adding windows_x86_64_msvc v0.52.6 (latest: v0.53.0)
   Compiling compiler_builtins v0.1.123
   Compiling core v0.0.0 (/Users/noreplydev/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.158
   Compiling memchr v2.5.0
   Compiling proc-macro2 v1.0.93
   Compiling unicode-ident v1.0.14
   Compiling std v0.0.0 (/Users/noreplydev/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling libc v0.2.169
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling serde v1.0.217
   Compiling bitflags v2.7.0
   Compiling crossbeam-utils v0.8.21
   Compiling regex-syntax v0.8.5
   Compiling glob v0.3.2
error: failed to run custom build command for `serde v1.0.217`

Caused by:
  process didn't exit successfully: `/Users/noreplydev/dev/rust/self-esp/target/release/build/serde-b148b6d5cbebfa90/build-script-build` (signal: 9, SIGKILL: kill)
warning: build failed, waiting for other jobs to finish...

```

ivmarkov, 2025-01-11:
Your failure has nothing to do with the problem described in this issue.

Seems like something is killing your rustc build process? As in sigkill 9? I can only speculate what this is...

ivmarkov, 2025-01-11:
Specifically, it fails in the serde build script?

noreplydev, 2025-01-11:
fixed, was related with another problem on my environment that leads on builds failing only on release mode. thanks

closed 2025-01-12 by ivmarkov