esp-rs/esp-idf-sys #292: ...no 'IDF_PYTHON_ENV_PATH' item，Debian 12, std mode, esp-idf-template, idf master

xiaguangbo, 2024-03-11:
## problem

Multiple attempts to rebuild from scratch failed
[image]

## log

`
xiaguangbo@debian:/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2$ cargo build
   Compiling esp-idf-sys v0.34.1
error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `/media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/target/debug/build/esp-idf-sys-4f4c73272e66742b/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.11.2
  pip 24.0 from /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages/pip (python 3.11)
  Requirement already satisfied: pip in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (24.0)
  Requirement already satisfied: setuptools in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (69.1.1)
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
  Requirement already satisfied: setuptools in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 3)) (69.1.1)
  Requirement already satisfied: packaging in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 4)) (24.0)
  Requirement already satisfied: click in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 7)) (8.1.7)
  Requirement already satisfied: pyserial in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 8)) (3.5)
  Requirement already satisfied: cryptography in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 9)) (42.0.5)
  Requirement already satisfied: pyparsing in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 10)) (3.1.2)
  Requirement already satisfied: pyelftools in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 11)) (0.30)
  Requirement already satisfied: idf-component-manager in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 12)) (1.5.2)
  Requirement already satisfied: esp-coredump in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 13)) (1.10.0)
  Requirement already satisfied: esptool in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 14)) (4.8.dev1)
  Requirement already satisfied: esp-idf-kconfig in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 15)) (2.1.0)
  Requirement already satisfied: esp-idf-monitor in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 16)) (1.4.0)
  Requirement already satisfied: esp-idf-nvs-partition-gen in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 17)) (0.1.1)
  Requirement already satisfied: esp-idf-size in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 18)) (1.1.1)
  Requirement already satisfied: esp-idf-panic-decoder in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 19)) (1.0.1)
  Requirement already satisfied: pyclang in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 20)) (0.4.2)
  Requirement already satisfied: construct in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 21)) (2.10.70)
  Requirement already satisfied: freertos_gdb in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from -r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 24)) (1.0.2)
  Requirement already satisfied: cffi>=1.12 in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from cryptography->-r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/esp-idf/master/tools/requirements/requirements.core.txt (line 9)) (1.16.0)
  Requirement already satisfied: requests<3 in /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espressif/python_env/idf5.3_py3.11_env/lib/python3.11/site-packages (from idf-component-manager->-r /media/xiaguangbo/linux_data/project/xtempctrl/esp32c2/.embuild/espres [... cut ...]
                                                                           PYTHON_VENV_DIR_TEMPLATE.format(idf_version,
                                                                                                           PYTHON_VER_MAJOR_MINOR))

    python_exe, subdir = get_python_exe_and_subdir()
    idf_python_export_path = os.path.join(idf_python_env_path, subdir)
    virtualenv_python = os.path.join(idf_python_export_path, python_exe)

    return idf_python_env_path, idf_python_export_path, virtualenv_python, idf_version
`

xiaguangbo, 2024-03-12:
Now using the default version of esp-idf can be, the network is good situation is very smooth

cchance27, 2024-04-05:
having this on OSX as well when i'm using master

Vollbrecht, 2024-04-05:
> having this on OSX as well when i'm using master

We are aware that the esp-idf master branch cannot be build. They changed the upstream tools install scripts in a breaking way. For the current time you can only use released versions of esp-idf. I am working on a replacement but its not finished yet.
Also in general we don't recommend to use esp-idf master for enduser projects as the api is unstable and the bindgen output can break on day to day basis.

xiaguangbo, 2024-04-09:
Recommended use of stable

closed 2024-04-09 by xiaguangbo