esp-rs/esp-idf-sys #295: tskNO_AFFINITY missing with esp-idf v5.2.1

torkleyy, 2024-03-20:
esp-idf-sys version: 0.34.1
esp-idf: 5.2.1

After upgrading esp-idf to 5.2.1 (from 5.1.3), the `tskNO_AFFINITY` symbol is no longer available. Not quite sure why, might be a bindgen issue because the header (freertos/task.h) is in `bindings.h`.

Vollbrecht, 2024-03-20:
Given the wired cfg i think what is happening here is they have some change on esp-idf master that they carry forward, but they seam to backport the tskNO_AFFINITY in every actual release.

Thanks for the find anyway. Do you want to create a small PR fixing it ?

Vollbrecht, 2024-06-21:
we have a workaround [here](https://github.com/esp-rs/esp-idf-template/issues/168#issuecomment-2182267993) though that still is maybe broken https://github.com/esp-rs/esp-idf-hal/issues/417 but closing this issue for now

closed 2024-06-21 by Vollbrecht