esp-rs/esp-idf-sys #296: Using custom components introduces a requirement of specifying all components

torkleyy, 2024-04-04:
As documented [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_components-esp_idf_components-native-builder-only), extra components require inclusion into `esp_idf_components`, but if `esp_idf_compoents` is present it acts as a filter for all built components. Interestingly, this doesn't apply to remotely managed components.

Ideally, extra components specified in `Cargo.toml` should be included automatically.

Vollbrecht, 2024-04-05:
Can you give concrete example, on how you structured your `Cargo.toml` and what component you were using one? A custom self-written component or one that you sourced yourself from the idf-component library? 

Overall its a two step process, we need a) building the component in the first place and make it link into the esp-idf library. b) generating "bindings" to be included so they are available in rust through esp-idf-sys.  

When you say extra components should be included automatically - what part do you want to be included automatically? For us the be able to build the component, we need a known format - that works with esp-idf and that we can than run through cmake.

torkleyy, 2024-04-11:
> Can you give concrete example, on how you structured your Cargo.toml and what component you were using one? A custom self-written component or one that you sourced yourself from the idf-component library?

A custom self-written one:

```toml
[[package.metadata.esp-idf-sys.extra_components]]
bindings_header = "extra/include/bindings.h"
component_dirs = ["extra"]
```

If I do just that, the "extra" component won't be built - I have to do this:

```toml
esp_idf_components = ["extra"]
```

Now it will build the "extra" component, but none of the other components 🙃 
So I ended up having to do this:

```toml
esp_idf_components = [
    "xtensa", "esp_ringbuf", "efuse", "esp_mm", "driver", "esp_pm", "mbedtls", "esp_bootloader_format", "esp_app_format", "bootloader_support", "bootloader", "esptool_py", "partition_table", "esp_partition", "app_update", "spi_flash", "pthread", "esp_system", "esp_rom", "hal", "log", "heap", "soc", "esp_hw_support", "freertos", "newlib", "cxx", "esp_common", "esp_timer", "app_trace", "esp_event", "nvs_flash", "esp_phy", "vfs", "lwip", "esp_netif_stack", "esp_netif", "wpa_supplicant", "esp_coex", "esp_wifi", "bt", "unity", "cmock", "console", "http_parser", "esp-tls", "esp_adc", "esp_eth", "esp_gdbstub", "esp_hid", "tcp_transport", "esp_http_client", "esp_http_server", "esp_https_ota", "esp_https_server", "esp_psram", "esp_lcd", "protobuf-c", "protocomm", "esp_local_ctrl", "espcoredump", "wear_levelling", "sdmmc", "idf_test", "ieee802154", "json", "mqtt", "nvs_sec_provider", "openthread", "perfmon", "spiffs", "ulp", "usb", "wifi_provisioning", "espressif__mdns", "main", "extra"
]
```

torkleyy, 2024-04-11:
Yes.

ivmarkov, 2024-05-01:
@torkleyy using `esp_idf_components` has nothing to do with "extra" compone ts. These are ALL esp-idf components, and unless you have a very specific use case, you should not touch it at all.

I would rather concentrate / dig into the direction of what is wrong with the actual, original "extra" co.po ents syntax that you tried to use and it did not work for you. There is so.ethi g wrong there in how you specify it for sure. I know it works actually.

ivmarkov, 2024-05-01:
In other words, inclusion of your extra components in the `esp_idf_components` list will happen automatically, you don't need to touch it / override it at all.

torkleyy, 2024-05-01:
Okay, I can try and create a reproducible example.

Vollbrecht, 2024-06-21:
i am closing this for now as i don't think its a problem. If you come up with a reproducible example feel free to reopen it. Thanks!

closed 2024-06-21 by Vollbrecht