esp-rs/esp-idf-sys #300: Wrapper for RTC Slow memory

weiying-chen, 2024-04-14:
Is there a wrapper for this?

```
[RTC Slow Memory](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/memory-types.html#rtc-slow-memory)

Global and static variables used by code which runs from RTC memory must be placed into RTC Slow memory. For example [deep sleep](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/deep-sleep-stub.html) variables can be placed here instead of RTC FAST memory, or code and variables accessed by the [ULP Coprocessor Programming](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/ulp.html).

The attribute macro named RTC_NOINIT_ATTR can be used to place data into this type of memory. The values placed into this section keep their value after waking from deep sleep.
```

I searched all over Github, but I couldn't find anything.

Vollbrecht, 2024-04-14:
You don't need any wrapper for it since you can directly specify where you want stuff to be linked.

Here is a small example
```rust
#[no_mangle]
#[link_section = ".rtc.data"]
static mut RTC_VARIABLE: u32 = 0;

fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Current RTC Value!: {}", unsafe { RTC_VARIABLE });
    unsafe { RTC_VARIABLE += 1 };

    FreeRtos::delay_ms(2_000);
    unsafe {
        esp_deep_sleep(2_000_000);
    }
}

```
Keep in mind that this example only illustrate the point. Please use something more appropriate that a `static mut` raw pointer here as its an easy footgun

closed 2024-04-14 by Vollbrecht

weiying-chen, 2024-04-14:
Thanks for the reply!