esp-rs/esp-idf-sys #316: How to set `esp-idf-sys` to rebuild when a C component changes?

Maldus512, 2024-06-14:
I successfully set up an extra C component in my Rust project. I noticed however that `cargo build` does not rebuilds `esp-idf-sys` when any of the C sources is modified.
I try adding a pertinent `rerun-if-changed` directive to `build.rs` but I don't think that `esp-idf-sys` recognizes that the C component has changed (not sure about any of this, forgive my ignorance).

Is there a way to rebuild `esp-idf-sys` depending on an extra C component?

Vollbrecht, 2024-06-14:
adding a remote component can't be detected by cargo as its not changing anything that it know about in the past.  That always need a `cargo clean -p esp-idf-sys`  and then `cargo b`

the second case is now what you are wishing for - monitor an already added component from cargo. That is still a bit tricky but not completely hopeless. We don't currently support that but there were changes in the "bindgen" dependency that allows for tracking of c/c++ header files and act if they changed. You can look at the PR that added it in bindgen [here](https://github.com/rust-lang/rust-bindgen/pull/2653/files). But that would not monitor changes you do in any .c file. For that i don't now about a solution. Since you only want to rebuild esp-idf-sys i would go with the `cargo clean -p esp-idf-sys` option as it only then would rebuild esp-idf-sys and dont recompile the rest.

Maldus512, 2024-06-14:
Thank you, I understand.

Vollbrecht, 2024-06-21:
closing this for now, as we don't plan on implementing automatic detection, and a sufficient "workaround" was presented, If someone wants to pick up from here, feel free to reopen it up.

closed 2024-06-21 by Vollbrecht