esp-rs/esp-idf-sys #319: Specifying components in `package.metadata.esp-idf-sys` fails the build

juliankrieger, 2024-06-27:
When generating a new blank template and adding any number of components in `package.metadata.esp-idf-sys`, the build fails with 

```
[...]
[813/860] Linking C static library esp-idf/xtensa/libxtensa.a
  [814/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_log.c.obj
  [815/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_sm_lgcy.c.obj
  [816/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_mbuf.c.obj
  [817/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_store.c.obj
  [818/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_l2cap_coc.c.obj
  [819/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_cfg.c.obj
  [820/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_eddystone.c.obj
  [821/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_att_clt.c.obj
  [822/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_att_cmd.c.obj
  [823/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_sm_cmd.c.obj
  [824/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_startup.c.obj
  [825/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_l2cap_sig.c.obj
  [826/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c.obj
  [827/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_gattc.c.obj
  [828/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_uuid.c.obj
  [829/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_flow.c.obj
  [830/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_misc.c.obj
  [831/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_sm_sc.c.obj
  [832/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_l2cap.c.obj
  [833/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c.obj
  [834/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_adv.c.obj
  [835/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_hci_util.c.obj
  [836/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_hs_hci.c.obj
  [837/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_gatts.c.obj
  [838/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_gattc_cache.c.obj
  [839/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/port/src/nvs_port.c.obj
  [840/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/npl/freertos/src/nimble_port_freertos.c.obj
  [841/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_gap.c.obj
  [842/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c.obj
  [843/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/store/config/src/ble_store_nvs.c.obj
  [844/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/src/ble_gattc_cache_conn.c.obj
  [845/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/nimble/host/store/ram/src/ble_store_ram.c.obj
  [846/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/nimble_port.c.obj
  [847/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/endian.c.obj
  [848/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/mem.c.obj
  [849/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/os_msys_init.c.obj
  [850/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/btc/profile/esp/blufi/nimble_host/esp_blufi.c.obj
  [851/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/os_mempool.c.obj
  [852/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/esp-hci/src/esp_nimble_hci.c.obj
  [853/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c.obj
  [854/860] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/os_mbuf.c.obj
  [855/860] Linking C static library esp-idf/bt/libbt.a
  [856/860] Generating ld/sections.ld
  [857/860] Building C object CMakeFiles/libespidf.elf.dir/project_elf_src_esp32s3.c.obj
  [858/860] Linking CXX executable libespidf.elf
  FAILED: libespidf.elf 
  : && /Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-a67f383aaf2931fa/out/build/libespidf.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld CMakeFiles/libespidf.elf.dir/project_elf_src_esp32s3.c.obj -o libespidf.elf -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/esp-idf/v5.2.2/components/soc/esp32s3/ld   -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/esp-idf/v5.2.2/components/esp_rom/esp32s3/ld   -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-a67f383aaf2931fa/out/build/esp-idf/esp_system/ld   -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/esp-idf/v5.2.2/components/esp_phy/lib/esp32s3   -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/esp-idf/v5.2.2/components/esp_coex/lib/esp32s3   -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/esp-idf/v5.2.2/components/esp_wifi/lib/esp32s3   -L/Users/wintermute/Dev/Work/EduLock/software/experiments/esp32/.embuild/espressif/esp-idf/v5.2.2/components/bt/controller/lib_esp32c3_family/esp32s3 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbed [... cut ...]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.49", default-features = false }

[build-dependencies]
embuild = "0.32.0"

[package.metadata.esp-idf-sys]
esp_idf_components = ["bt"]
```

Is this related to `binstart` in some way?

Vollbrecht, 2024-06-27:
while that technically should cross import all non specified components recursively through REQUIRES or PRIV_REQUIRES flag, it could create a case where for example "technically" bt doesn't require the "main" component and so for. So in that case you could miss something here that we assume to always be there.

I would in your shoes check out what "default" components we use and manual include the ones that seams pretty substatial to explicitly included by you. For example on a s3 the default list on a "hello-world" is something like "bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa"

You can try to add them all and add your bt component and try to delete stuff from that list and check when it breaks. You can find the list of included components in the "output" file that is create in the buildoutput inside the esp-idf-sys build dir

cchandler, 2024-09-05:
I had the exact same issue as the OP. I did @Vollbrecht's suggestion and checked the default component list in `./target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-9954d550c2403695/output` (esp-idf-sys=0.35). There's a line prefixed with `-- Components:` that has the entire list.

Adding that entire list fixed it for me and then I was able to winnow it down to just "main" to resolve the problem. In my case I also have extra_components in the `[package.metadata.esp-idf-sys]`.

ivmarkov, 2024-09-05:
> and then I was able to winnow it down to just "main" to resolve the problem. 

What does "just main" mean exactly for your concrete case?

cchandler, 2024-09-05:
@ivmarkov Here's the section from Cargo.toml. If I omit `main` from the `esp_idf_components` I get the OP's error in a cleanroom build.

```
[package.metadata.esp-idf-sys]
esp_idf_components = ["usb", "usb_host_hid", "main"]
esp_idf_comp_usb_enabled = true
mcu = "esp32s3"
extra_components = [
    { component_dirs = [ "idf-extra-components/usb/usb_host_hid/" ], bindings_header = "idf-extra-components/usb/usb_host_hid/include/usb/hid_host.h", bindings_module = "hid_host", manifest_dir = "idf-extra-components/usb/usb_host_hid" }
]
```