esp-rs/esp-idf-sys #320: windows failed to run custom build command for `esp-idf-sys v0.34.1`

mengyan258, 2024-06-29:
to configure：
```
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1"
ESP_IDF_TOOLS_INSTALL_DIR = { value = "global"}

# Workaround for https://github.com/esp-rs/esp-idf-template/issues/174 until
# https://github.com/esp-rs/esp-idf-hal/pull/387 gets released and the template
# updated.
CRATE_CC_NO_DEFAULTS = "1"
IDF_PATH = "C:\\Users\\梦魇\\.espressif\\esp-idf"

[patch.crates-io]
cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }
```
log：
```
error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `D:\p\esp\target\debug\build\esp-idf-sys-06597401d0a9be6e\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "C:\\Users\\梦魇\\.espressif\\esp-idf",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  fatal: not a git repository (or any of the parent directories): .git
  fatal: not a git repository (or any of the parent directories): .git
  Error: 'C:\Users\梦魇\.espressif\esp-idf' is not a git respository

  Caused by:
      0: command '"git" "rev-parse" "--show-toplevel"' exited with non-zero status code 128
      1: fatal: not a git repository (or any of the parent directories): .git

  Stack backtrace:
     0: std::backtrace_rs::backtrace::dbghelp64::trace
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\..\..\backtrace\src\backtrace\dbghelp64.rs:91
     1: std::backtrace_rs::backtrace::trace_unsynchronized
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
     2: std::backtrace::Backtrace::create
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\backtrace.rs:331
     3: std::backtrace::Backtrace::capture
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\backtrace.rs:296
     4: anyhow::context::ext::impl$0::ext_context<enum2$<embuild::cmd::CmdError>,anyhow::Error>
               at C:\Users\梦魇\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\anyhow-1.0.86\src\backtrace.rs:27
     5: anyhow::context::impl$0::context<alloc::string::String,enum2$<embuild::cmd::CmdError>,anyhow::Error>
               at C:\Users\梦魇\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\anyhow-1.0.86\src\context.rs:54
     6: embuild::git::Repository::open<ref$<std::path::PathBuf> >
               at C:\Users\梦魇\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\embuild-0.31.4\src\git.rs:52
     7: build_script_build::native::cargo_driver::build
               at .\build\native\cargo_driver.rs:204
     8: build_script_build::native::build
               at .\build\native.rs:17
     9: build_script_build::main
               at .\build\build.rs:51
    10: core::ops::function::FnOnce::call_once<enum2$<core::result::Result<tuple$<>,anyhow::Error> > (*)(),tuple$<> >
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95\library\core\src\ops\function.rs:250
    11: std::sys::backtrace::__rust_begin_short_backtrace<enum2$<core::result::Result<tuple$<>,anyhow::Error> > (*)(),enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95\library\std\src\sys\backtrace.rs:155
    12: std::rt::lang_start::closure$0<enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95\library\std\src\rt.rs:159
    13: std::rt::lang_start_internal
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\rt.rs:141
    14: std::rt::lang_start<enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95\library\std\src\rt.rs:158
    15: main
    16: invoke_main
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    17: __scrt_common_main_seh
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    18: BaseThreadInitThunk
    19: RtlUserThreadStart
```

Vollbrecht, 2024-06-29:
If you are providing "ESP_IDF_TOOLS_INSTALL_DIR = { value = "global"}" you should not manually provide a IDF_PATH = "C:\\Users\\梦魇\\.espressif\\esp-idf" env variable. 

Did you manually clone into .espressif\esp-idf dir? if not i would suggest you to remove the ,espressif dir and just rerun the build. It should automatically repopulate that dir in this case. And as the error suggest, whatever lives in IDF_PATH needs to be a git repository.

Vollbrecht, 2024-06-29:
also you dont need to use the patch.crates-io if you are using the `
CRATE_CC_NO_DEFAULTS = "1"` option

mengyan258, 2024-06-29:
> 此外，如果您正在使用该选项，则无需使用 patch.crates-io` CRATE_CC_NO_DEFAULTS = "1"`

按找你给的方式，重新修改了配置：
```
[build]
target = "riscv32imc-esp-espidf"

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = [ "--cfg",  "espidf_time64"] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c3"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1"
ESP_IDF_TOOLS_INSTALL_DIR = { value = "global"}

# Workaround for https://github.com/esp-rs/esp-idf-template/issues/174 until
# https://github.com/esp-rs/esp-idf-hal/pull/387 gets released and the template
# updated.
#CRATE_CC_NO_DEFAULTS = "1"
#IDF_PATH = "C:\\Users\\梦魇\\.espressif\\esp-idf"

[patch.crates-io]
cc = { git = "https://github.com/rust-lang/cc-rs", tag = "1.0.83" }
```
Cleared cache and recompiled, still reporting an error
```
error: failed to run custom build command for `esp-idf-sys v0.34.1`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_RELEASE_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `D:\p\esp\target\release\build\esp-idf-sys-733c207e6cbec726\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
[... 140 lines cut ...]
     3: std::backtrace::Backtrace::capture
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\backtrace.rs:296
     4: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
     5: <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
     6: embuild::git::Repository::clone_ext
     7: embuild::git::sdk::RemoteSdk::open_or_clone
     8: embuild::espidf::Installer::install
     9: build_script_build::native::cargo_driver::build::{{closure}}::{{closure}}
    10: build_script_build::native::cargo_driver::build
    11: build_script_build::native::build
    12: <build_script_build::BindgenCallbacks as bindgen::callbacks::ParseCallbacks>::int_macro::{{closure}}
    13: core::ops::function::FnOnce::call_once
    14: std::sys::backtrace::__rust_begin_short_backtrace
    15: std::rt::lang_start::{{closure}}
    16: std::rt::lang_start_internal
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\rt.rs:141
    17: std::rt::lang_start
    18: main
    19: invoke_main
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    20: __scrt_common_main_seh
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    21: BaseThreadInitThunk
    22: RtlUserThreadStart
```

Vollbrecht, 2024-06-29:
the error here " fatal: unable to access 'https://github.com/espressif/esp-idf.git/': Failed to connect to github.com port 443 after 21109 ms: Couldn't connect to server" indicates that you cannot clone from github.com.

You have two options. Either make sure that you can directly clone from github via your current internet connection, or if that is not possible try to git clone the esp-idf repository's version you want to use from another git mirror that is accessible by your connection.

 In the later case you can use the IDF_PATH variable to point to such manual clone, but you need to make sure that its actually a git repository and not just a directory containing ESP-IDF.

mengyan258, 2024-06-29:
> the error here " fatal: unable to access 'https://github.com/espressif/esp-idf.git/': Failed to connect to github.com port 443 after 21109 ms: Couldn't connect to server" indicates that you cannot [...]

Setting IDF-PATH now reports another error
```
error: failed to run custom build command for `esp-idf-sys v0.34.1`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_RELEASE_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `D:\p\esp\target\release\build\esp-idf-sys-733c207e6cbec726\build-script-build` (exit code: 101)
  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "C:\\Users\\梦魇\\.espressif\\esp-idf\\v5.1",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using custom user-supplied esp-idf repository at 'C:\Users\梦魇\.espressif\esp-idf\v5.1' (detected from env variable `IDF_PATH`)
  Using esp-idf v5.1.0 at 'C:\Users\梦魇\.espressif\esp-idf\v5.1'
  C:\Users\梦魇\.espressif\esp-idf\v5.1\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
    import pkg_resources
  Traceback (most recent call last):
    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "D:\p\esp\.embuild\espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\kconfgen\__main__.py", line 16, in <module>
      main()
    File "D:\p\esp\.embuild\espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\kconfgen\core.py", line 319, in main
      config = kconfiglib.Kconfig(args.kconfig)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\p\esp\.embuild\espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\kconfiglib.py", line 947, in __init__
      self._init(filename, warn, warn_to_stderr, encoding)
    File "D:\p\esp\.embuild\espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\kconfiglib.py", line 1085, in _init
      self._parse_block(None, self.top_node, self.top_node).next = None
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\p\esp\.embuild\espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\kconfiglib.py", line 2989, in _parse_block
      prev = self._parse_block(None, parent, prev)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "D:\p\esp\.embuild\espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\kconfiglib.py", line 2977, in _parse_block
      raise KconfigError(
  kconfiglib.KconfigError: D:/p/esp/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-c1d40fc4da51e48a/out/build/kconfigs_projbuild.in:1: 'C:/Users/姊﹂瓏/.espressif/esp-idf/v5.1/components/bootloader/Kconfig.projbuild' not found (in 'source "C:/Users/姊﹂瓏/.espressif/esp-idf/v5.1/components/bootloader/Kconfig.projbuild"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
  CMake Error at C:/Users/梦魇/.espressif/esp-idf/v5.1/tools/cmake/kconfig.cmake:176 (message):
    Failed to run kconfgen
    (D:/p/esp/.embuild/espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe;-m;kconfgen;--list-separator=semicolon;--kconfig;C:/Users/梦魇/.espressif/esp-idf/v5.1/Kconfig;--sdkconfig-rename;C:/Users/梦魇/.espressif/esp-idf/v5.1/sdkconfig.rename;--config;D:/p/esp/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-c1d40fc4da51e48a/out/sdkconfig;--defaults;D:/p/esp/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-c1d40fc4da51e48a/out/gen-sdkconfig.defaults;--defaults;D:/p/esp/sdkconfig.defaults;--env-file;D:/p/esp/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-c1d40fc4da51e48a/out/build/config.env).
    Error 1
  Call Stack (most recent call first):
    C:/Users/梦魇/.espressif/esp-idf/v5.1/tools/cmake/build.cmake:611 (__kconfig_generate_config)
    C:/Users/梦魇/.espressif/esp-idf/v5.1/tools/cmake/project.cmake:464 (idf_build_process)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at C:\Users\梦魇\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\cmake-0.1.50\src\lib.rs:1098:5:

  command did not execute successfully, got: exit code: 1

  build script failed, must exit now
  stack backtrace:
     0: std::panicking::begin_panic_handler
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\std\src\panicking.rs:661
     1: core::panicking::panic_fmt
               at /rustc/e9e6e2e444c30c23a9c878a88fbc3978c2acad95/library\core\src\panicking.rs:74
     2: cmake::find_exe::{{closure}}
     3: <cmake::Version as core::default::Default>::default
     4: cmake::Config::build
     5: build_script_build::native::cargo_driver::build
     6: build_script_build::native::build
     7: <build_script_build::BindgenCallbacks as bindgen::callbacks::ParseCallbacks>::int_macro::{{closure}}
     8: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

Vollbrecht, 2024-06-29:
It seams that you did not actually got the full git repository. You can see by "C:/Users/姊﹂瓏/.espressif/esp-idf/v5.1/components/bootloader/Kconfig.projbuild" that it try to look into that file. If that file does not exist than it cannot proceed. 

Notice that just a git clone on a esp-idf repository is not enough, as esp-idf contains lots of submodules and they are only get fetched if you specify so. That's why its important that if you provide your own IDF_PATH, that you make sure its a fully working esp-idf repository with all submodules included. That does not seem the case here.

To clone with submodules you need to use the "--recursive" flag as we do in our command in the buildscript. However if you do not do that from github you might run into other problems as in general the submodules are relative to github path. So it depends on your git mirror what it provides and what not. You can read about it [here](https://github.com/espressif/esp-idf?tab=readme-ov-file#non-github-forks) what i mean.

In general i would strongly recommend to get a connection that allows directly cloning from github.com, as that would drastically simplify everything as you than would not need to do all this manual intervention with providing your own IDF_PATH etc.

mengyan258, 2024-06-30:
> It seams that you did not actually got the full git repository. You can see by "C:/Users/姊﹂瓏/.espressif/esp-idf/v5.1/components/bootloader/Kconfig.projbuild" that it try to look into that file. If [...]

I am using this command
```
git clone --recursive --depth 1 --shallow-submodules --branch v5.1 https://github.com/espressif/esp-idf.git C:\Users\梦魇\.espressif\esp-idf\v5.1
```

mengyan258, 2024-07-01:
```
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.34.1: Ignoring configuration setting `ESP_IDF_VERSION` (Tag v5.1): custom esp-idf repository detected via $IDF_PATH

error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `D:\p\esp\target\release\build\esp-idf-sys-920865793ce175ef\build-script-build` (exit code: 1)
  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: Some(
          Global,
      ),
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "C:\\Users\\梦魇\\.espressif\\esp-idf\\v5.1",
          ),
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using custom user-supplied esp-idf repository at 'C:\Users\梦魇\.espressif\esp-idf\v5.1' (detected from env variable `IDF_PATH`)
  Error: Could not install esp-idf

  Caused by:
      cannot find binary path

  Stack backtrace:
     0: std::backtrace_rs::backtrace::dbghelp64::trace
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library\std\src\..\..\backtrace\src\backtrace\dbghelp64.rs:91
     1: std::backtrace_rs::backtrace::trace_unsynchronized
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
     2: std::backtrace::Backtrace::create
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library\std\src\backtrace.rs:331
     3: std::backtrace::Backtrace::capture
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library\std\src\backtrace.rs:296
     4: anyhow::error::impl$1::from<which::error::Error>
               at C:\Users\梦魇\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\anyhow-1.0.86\src\backtrace.rs:27
     5: core::result::impl$27::from_residual<embuild::espidf::EspIdf,which::error::Error,anyhow::Error>
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8\library\core\src\result.rs:1989
     6: embuild::espidf::Installer::install
               at C:\Users\梦魇\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\embuild-0.31.4\src\espidf.rs:418
     7: build_script_build::native::cargo_driver::build::closure$3
               at .\build\native\cargo_driver.rs:163
     8: build_script_build::native::cargo_driver::build
               at .\build\native\cargo_driver.rs:200
     9: build_script_build::native::build
               at .\build\native.rs:17
    10: build_script_build::main
               at .\build\build.rs:51
    11: core::ops::function::FnOnce::call_once<enum2$<core::result::Result<tuple$<>,anyhow::Error> > (*)(),tuple$<> >
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8\library\core\src\ops\function.rs:250
    12: std::sys::backtrace::__rust_begin_short_backtrace<enum2$<core::result::Result<tuple$<>,anyhow::Error> > (*)(),enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8\library\std\src\sys\backtrace.rs:155
    13: std::rt::lang_start::closure$0<enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8\library\std\src\rt.rs:159
    14: std::rt::lang_start_internal
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8/library\std\src\rt.rs:141
    15: std::rt::lang_start<enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/ba1d7f4a083e6402679105115ded645512a7aea8\library\std\src\rt.rs:158
    16: main
    17: invoke_main
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    18: __scrt_common_main_seh
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    19: BaseThreadInitThunk
    20: RtlUserThreadStart
```

OhmPuchiss, 2024-09-02:
Got the same problem. Did you solve it ?