esp-rs/esp-idf-sys #324: error: Toolchain esp in XXX\esp_32_webserver\rust-toolchain.toml is custom and not installed

pCoder3000, 2024-07-13:
Hello Firends, 

Im new to rust and esp32 i tried to follow a beginner tutorial: https://www.youtube.com/watchv=o8yNNVFzNnM&list=PL0U7YUX2VnBFbwTi96wUB1nZzPVN3HzgS

I know there an already many open issues and maybe this is a duplicate, but can someone explain to me in simple terms why the rust analyzer is broken after I install the template? 

I get this error when i clone the prject and try to build it with cargo build. I try to compile it under windows 11

`2024-07-13T22:39:01.467639Z ERROR rust_analyzer::main_loop: FetchWorkspaceError:
rust-analyzer failed to load workspace: Failed to load the project at XXX\esp_32_webserver\Cargo.toml: Failed to query rust toolchain version at XXX\getting started\esp_32_webserver, is your toolchain setup correctly?: "XXX\\.cargo\\bin\\cargo.exe" "--version" failed, exit code: 1
stderr:
error: Toolchain esp in XXX\esp_32_webserver\rust-toolchain.toml is custom and not installed`

raulqtech, 2024-09-27:
Is there any news about this topic? I'm having the same issue

ivmarkov, 2024-09-27:
https://github.com/esp-rs/esp-idf-template?tab=readme-ov-file#install-rust--clang-toolchains-for-espressif-socs-with-espup

ivmarkov, 2024-11-29:
^^^ Addresses the problem very well, closing.

closed 2024-11-29 by ivmarkov