esp-rs/esp-idf-sys #326: Error Build: Error data-layout not match

tao-a-man, 2024-07-16:
'esp32s3' is not a recognized processor for this target (ignoring processor)
'+atomics' is not a recognized feature for this target (ignoring feature)
'+mul16' is not a recognized feature for this target (ignoring feature)
'-fp' is not a recognized feature for this target (ignoring feature)
'esp32s3' is not a recognized processor for this target (ignoring processor)
error: data-layout for target `xtensa-esp32s3-none-elf`, `e-m:e-p:32:32-v1:8:8-i64:64-i128:128-n32`, differs from LLVM target's `xtensa-none-elf` default layout, `e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32`
[image]

Vollbrecht, 2024-09-15:
you probably mixed up some esp-hal and esp-idf-hal project? To make any sense of this error you may provide what you exactly did, and how did you setup your project. I strongly suspect that you mixed and matched esp-hal and esp--idf-hal stuff. They both have distinct target definitions ala `xtensa-esp32s3-none-elf` and `xtensa-esp32s3-esp-espidf`

ivmarkov, 2024-09-18:
No activity, closing.

closed 2024-09-18 by ivmarkov

harryhaaren, 2024-11-15:
This also seems to happen on recent `espup` generated projects, which don't have a `rust-toolchain.toml` file, and compiling on `x86_64` based machines. The solution for me was to manually specific `cargo +esp b --release --features ...`.

ivmarkov, 2024-11-16:
espup is not in the business of generating projects. Did you mean esp-idf-template?

joebnb, 2024-12-06:
```
esp@342ba39382aa:~/esp32s3$ cargo run
   Compiling compiler_builtins v0.1.109
   Compiling core v0.0.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.155
   Compiling libc v0.2.167
   Compiling serde v1.0.215
   Compiling crossbeam-utils v0.8.20
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
   Compiling rustix v0.38.41
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
   Compiling rustversion v1.0.18
   Compiling crossbeam-epoch v0.9.18
   Compiling bindgen v0.69.5
   Compiling crossbeam-deque v0.8.5
   Compiling anyhow v1.0.94
   Compiling ignore v0.4.23
   Compiling strum_macros v0.24.3
   Compiling nix v0.29.0
   Compiling which v4.4.2
   Compiling serde_json v1.0.133
   Compiling clang-sys v1.8.1
   Compiling fs_at v0.2.1
   Compiling thiserror v1.0.69
   Compiling cexpr v0.6.0
   Compiling itertools v0.12.1
   Compiling cc v1.1.30
   Compiling prettyplease v0.2.25
   Compiling regex v1.11.1
   Compiling bitflags v1.3.2
   Compiling fastrand v2.2.0
   Compiling rustc-hash v1.1.0
   Compiling normpath v1.3.0
   Compiling lazy_static v1.5.0
   Compiling lazycell v1.3.0
   Compiling remove_dir_all v0.8.4
   Compiling tempfile v3.14.0
   Compiling globwalk v0.8.1
   Compiling cmake v0.1.52
   Compiling strum v0.24.1
   Compiling filetime v0.2.25
   Compiling semver v1.0.23
   Compiling camino v1.1.9
   Compiling cargo-platform v0.1.9
   Compiling envy v0.4.2
   Compiling cargo_metadata v0.18.1
   Compiling autocfg v1.4.0
   Compiling ident_case v1.0.1
   Compiling fnv v1.0.7
   Compiling darling_core v0.20.10
   Compiling num-traits v0.2.19
   Compiling embedded-io-async v0.6.1
   Compiling embuild v0.32.0
   Compiling heapless v0.8.0
   Compiling embassy-time-driver v0.1.0
   Compiling unicode-xid v0.2.6
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
error: data-layout for target `xtensa-esp32s3-espidf`, `e-m:e-p:32:32-v1:8:8-i64:64-i128:128-n32`, differs from LLVM target's `xtensa-none-elf` default layout, `e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32`

'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
   Compiling rustc-std-workspace-core v1.99.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling iana-time-zone v0.1.61
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
'esp32-s3' is not a recognized processor for this target (ignoring processor)
error: could not compile `rustc-std-workspace-core` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: could not compile `core` (lib) due to 1 previous error
```

same error

ivmarkov, 2024-12-06:
> ```
> esp@342ba39382aa:~/esp32s3$ cargo run
>    Compiling compiler_builtins v0.1.109
>    Compiling core v0.0.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
>    Compiling libc v0.2.155
>    Compiling libc v0.2.167
>    Compiling serde v1.0.215
>    Compiling crossbeam-utils v0.8.20
>    Compiling memchr v2.5.0
>    Compiling std v0.0.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
>    Compiling rustix v0.38.41
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
>    Compiling rustversion v1.0.18
>    Compiling crossbeam-epoch v0.9.18
>    Compiling bindgen v0.69.5
>    Compiling crossbeam-deque v0.8.5
>    Compiling anyhow v1.0.94
>    Compiling ignore v0.4.23
>    Compiling strum_macros v0.24.3
>    Compiling nix v0.29.0
>    Compiling which v4.4.2
>    Compiling serde_json v1.0.133
>    Compiling clang-sys v1.8.1
>    Compiling fs_at v0.2.1
>    Compiling thiserror v1.0.69
>    Compiling cexpr v0.6.0
>    Compiling itertools v0.12.1
>    Compiling cc v1.1.30
>    Compiling prettyplease v0.2.25
>    Compiling regex v1.11.1
>    Compiling bitflags v1.3.2
>    Compiling fastrand v2.2.0
>    Compiling rustc-hash v1.1.0
>    Compiling normpath v1.3.0
>    Compiling lazy_static v1.5.0
>    Compiling lazycell v1.3.0
>    Compiling remove_dir_all v0.8.4
>    Compiling tempfile v3.14.0
>    Compiling globwalk v0.8.1
>    Compiling cmake v0.1.52
>    Compiling strum v0.24.1
>    Compiling filetime v0.2.25
>    Compiling semver v1.0.23
>    Compiling camino v1.1.9
>    Compiling cargo-platform v0.1.9
>    Compiling envy v0.4.2
>    Compiling cargo_metadata v0.18.1
>    Compiling autocfg v1.4.0
>    Compiling ident_case v1.0.1
>    Compiling fnv v1.0.7
>    Compiling darling_core v0.20.10
>    Compiling num-traits v0.2.19
>    Compiling embedded-io-async v0.6.1
>    Compiling embuild v0.32.0
>    Compiling heapless v0.8.0
>    Compiling embassy-time-driver v0.1.0
>    Compiling unicode-xid v0.2.6
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> error: data-layout for target `xtensa-esp32s3-espidf`, `e-m:e-p:32:32-v1:8:8-i64:64-i128:128-n32`, differs from LLVM target's `xtensa-none-elf` default layout, `e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32`
> 
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
>    Compiling rustc-std-workspace-core v1.99.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
>    Compiling iana-time-zone v0.1.61
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> error: could not compile `rustc-std-workspace-core` (lib) due to 1 previous error
> warning: build failed, waiting for other jobs to finish...
> error: could not compile `core` (lib) due to 1 previous error
> ```
> 
> same error by runing https://github.com/esp-rs/esp-idf-template generated project

Install `espup`. Run it. Then do `rustup default esp`.  THEN try to compile for xtensa. 

You ate compiling with `/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core` which is the standard Rust nightly, which does not have support for esp32s3 (xtensa). Did you read the readme of `esp-idf-template`?

joebnb, 2024-12-06:
> > ```
> > esp@342ba39382aa:~/esp32s3$ cargo run
> >    Compiling compiler_builtins v0.1.109
> >    Compiling core v0.0.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
> >    Compiling libc v0.2.155
> >    Compiling libc v0.2.167
> >    Compiling serde v1.0.215
> >    Compiling crossbeam-utils v0.8.20
> >    Compiling memchr v2.5.0
> >    Compiling std v0.0.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> >    Compiling rustix v0.38.41
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> >    Compiling rustversion v1.0.18
> >    Compiling crossbeam-epoch v0.9.18
> >    Compiling bindgen v0.69.5
> >    Compiling crossbeam-deque v0.8.5
> >    Compiling anyhow v1.0.94
> >    Compiling ignore v0.4.23
> >    Compiling strum_macros v0.24.3
> >    Compiling nix v0.29.0
> >    Compiling which v4.4.2
> >    Compiling serde_json v1.0.133
> >    Compiling clang-sys v1.8.1
> >    Compiling fs_at v0.2.1
> >    Compiling thiserror v1.0.69
> >    Compiling cexpr v0.6.0
> >    Compiling itertools v0.12.1
> >    Compiling cc v1.1.30
> >    Compiling prettyplease v0.2.25
> >    Compiling regex v1.11.1
> >    Compiling bitflags v1.3.2
> >    Compiling fastrand v2.2.0
> >    Compiling rustc-hash v1.1.0
> >    Compiling normpath v1.3.0
> >    Compiling lazy_static v1.5.0
> >    Compiling lazycell v1.3.0
> >    Compiling remove_dir_all v0.8.4
> >    Compiling tempfile v3.14.0
> >    Compiling globwalk v0.8.1
> >    Compiling cmake v0.1.52
> >    Compiling strum v0.24.1
> >    Compiling filetime v0.2.25
> >    Compiling semver v1.0.23
> >    Compiling camino v1.1.9
> >    Compiling cargo-platform v0.1.9
> >    Compiling envy v0.4.2
> >    Compiling cargo_metadata v0.18.1
> >    Compiling autocfg v1.4.0
> >    Compiling ident_case v1.0.1
> >    Compiling fnv v1.0.7
> >    Compiling darling_core v0.20.10
> >    Compiling num-traits v0.2.19
> >    Compiling embedded-io-async v0.6.1
> >    Compiling embuild v0.32.0
> >    Compiling heapless v0.8.0
> >    Compiling embassy-time-driver v0.1.0
> >    Compiling unicode-xid v0.2.6
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > error: data-layout for target `xtensa-esp32s3-espidf`, `e-m:e-p:32:32-v1:8:8-i64:64-i128:128-n32`, differs from LLVM target's `xtensa-none-elf` default layout, `e-m:e-p:32:32-i8:8:32-i16:16:32-i64:64-n32`
> > 
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> >    Compiling rustc-std-workspace-core v1.99.0 (/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
> >    Compiling iana-time-zone v0.1.61
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > 'esp32-s3' is not a recognized processor for this target (ignoring processor)
> > error: could not compile `rustc-std-workspace-core` (lib) due to 1 previous error
> > warning: build failed, waiting for other jobs to finish...
> > error: could not compile `core` (lib) due to 1 previous error
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > same error by runing https://github.com/esp-rs/esp-idf-template generated project
> 
> Install `espup`. Run it. Then do `rustup default esp`. THEN try to compile for xtensa.
> 
> You are compiling with `/home/esp/.rustup/toolchains/nightly-2024-06-30-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core` which is the standard Rust nightly, which does not have support for esp32s3 (xtensa). Did you read the readme of `esp-idf-template`?

 i'm follow this https://github.com/esp-rs/std-training/blob/main/intro/hardware-check/rust-toolchain.toml ,and i generate a new project,found that should be esp,now im trying to build a hello world

0x53A, 2025-06-27:
The risc-v processors (like the esp32-c6) use the standard rust toolchain for risc-v, so the ``rust-toolchain.toml`` references the default nightly rust compiler.

For extensa mcus like the esp32-s3, you'll need to install the ``esp`` toolchain using espup, and then in ``rust-toolchain.toml`` use ``channel    = "esp"``.

I'm replying here in this old issue because you can still find it using google.