esp-rs/esp-idf-sys #327: Build failed due to load libclang.dll when generating bindings

Sydonian, 2024-07-21:
building logs:
```
   Compiling byteorder v1.5.0
   Compiling nb v1.1.0
   Compiling embedded-io v0.6.1
   Compiling stable_deref_trait v1.2.0
   Compiling embedded-hal v1.0.0
   Compiling futures-core v0.3.30
   Compiling futures-task v0.3.30
   Compiling pin-utils v0.1.0
   Compiling esp-idf-sys v0.35.0
   Compiling pin-project-lite v0.2.14
   Compiling enumset v1.1.3
   Compiling libc v0.2.155
   Compiling futures-util v0.3.30
   Compiling nb v0.1.3
   Compiling const_format v0.2.32
   Compiling embedded-io-async v0.6.1
   Compiling hash32 v0.3.1
   Compiling void v1.0.2
   Compiling critical-section v1.1.2
   Compiling cfg-if v1.0.0
   Compiling num_enum v0.7.2
   Compiling heapless v0.8.0
   Compiling embedded-hal v0.2.7
   Compiling embedded-hal-nb v1.0.0
   Compiling embedded-hal-async v1.0.0
   Compiling embedded-can v0.4.1
   Compiling serde v1.0.204
   Compiling atomic-waker v1.1.2
   Compiling log v0.4.22
   Compiling uncased v0.9.10
   Compiling embassy-time-driver v0.1.0
   Compiling embassy-futures v0.1.1
   Compiling embassy-sync v0.6.0
   Compiling embedded-svc v0.28.0
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: (esp-idf-sys) Too long output directory: `\\?\E:\ESP\projects\testlcd\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-3ae04a86ffd25549\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `c++` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `E:\ESP\projects\testlcd\target\debug\build\esp-idf-sys-d7ae078c1bb86ffe\build-script-build` (exit code: 101)
  --- stdout
  cargo:warning=(esp-idf-sys) Too long output directory: `\\?\E:\ESP\projects\testlcd\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-3ae04a86ffd25549\out`. Shorten your project path down to no more than 10 characters (or use WSL2 and its native Linux filesystem). Note that tricks like Windows `subst` do NOT work!
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.12.4
  pip 24.1.2 from E:\ESP\projects\testlcd\.embuild\espressif\python_env\idf5.2_py3.12_env\Lib\site-packages\pip (python 3.12)
  Requirement already satisfied: pip in e:\esp\projects\testlcd\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (24.1.2)
  Requirement already satisfied: setuptools in e:\esp\projects\testlcd\.embuild\espressif\python_env\idf5.2_py3.12_env\lib\site-packages (71.0.0)
  Collecting setuptools
    Using cached setuptools-71.0.4-py3-none-any.whl.metadata (6.5 kB)
  Using cached setuptools-71.0.4-py3-none-any.whl (2.3 MB)
  Installing collected packages: setuptools
    Attempting uninstall: setuptools
[... 152 lines cut ...]
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2.2") }
  Using esp-idf v5.2.2 at 'E:\ESP\projects\testlcd\.embuild\espressif\esp-idf\v5.2.2'
  Built components: xtensa, esp_ringbuf, efuse, esp_mm, driver, esp_pm, mbedtls, esp_bootloader_format, esp_app_format, bootloader_support, bootloader, esptool_py, partition_table, esp_partition, app_update, spi_flash, pthread, esp_system, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, cxx, esp_common, esp_timer, app_trace, esp_event, nvs_flash, esp_phy, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, sdmmc, fatfs, idf_test, ieee802154, json, mqtt, nvs_sec_provider, openthread, perfmon, spiffs, ulp, usb, wifi_provisioning, main
  thread 'main' panicked at C:\Users\79nos\.cargo\registry\src\mirrors.tuna.tsinghua.edu.cn-df7c3c540f42cdbd\bindgen-0.69.4\lib.rs:622:31:
  Unable to find libclang: "the `libclang` shared library at C:\\Users\\79nos\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

and I have set the LIBCLANG_PATH env:
[image]

thielemans, 2024-07-23:
Can confirm that I have the same issue after updating rustup, & espup update.

The dll is on the specified path but could not be opened for some reason.

The issue persists after (re)installing the ESP IDF cargo subcommands and doing espup remove & espup install.

Vollbrecht, 2024-07-23:
what version of the esp rust toolchain did you currently have installed? E.g what is the result of `rustc +esp --version`. There were some windows specific problems, that lead the team to not be able to create a windows version of the compiler for some of the latest releases. 

You can checkout the respective [repos](https://github.com/esp-rs/rust-build/releases) here and [here](https://github.com/esp-rs/rust).

Depending on your version installed try to downgrade one compiler version earlier and check if that helps.

Sydonian, 2024-07-23:
> what version of the esp rust toolchain did you currently have installed? E.g what is the result of `rustc +esp --version`. There were some windows specific problems, that lead the team to not be able to create a windows version of the compiler for some of the latest releases.
> 
> You can checkout the respective [repos](https://github.com/esp-rs/rust-build/releases) here and [here](https://github.com/esp-rs/rust).
> 
> Depending on your version installed try to downgrade one compiler version earlier and check if that helps. You can see on the linked release page from above, how you can install a specific compiler version. Also than make sure to either delete the "newer" so the other one is actually used.

Here is the toolchain version I using currently:
[image]
and the version of espup:
[image]

I just ran `esp uninstall` and `esp install --toolchain-version 1.77.0` to install an older toolchain, then I got complied successfully. Maybe there is something wrong with the newer veriosn?

thielemans, 2024-07-23:
I tried with 3 espup versions, each time doing an uninstall first and installing the default toolchain (1.79.0.0 nightly) again. Tested on a clean build of my current project.

```
espup 0.12.2
rustc 1.79.0-nightly (e60d66433 2024-06-10) (1.79.0.0)
```
Fails

```
espup 0.12.0
rustc 1.79.0-nightly (e60d66433 2024-06-10) (1.79.0.0)
```
Also fails

```
rustc 1.79.0-nightly (e60d66433 2024-06-10) (1.79.0.0)
espup 0.11.0
```
Works in my case. Would it be possible that the issue is related to the way how espup sets up the environment?

Vollbrecht, 2024-07-24:
@SergioGasquez @MabezDev are you aware of windows specific issues in espup v12 ? Should i transfer the issue?