esp-rs/esp-idf-sys #328: TcpStream::try_clone() -> Function not implemented (os error 88)

victorbnl, 2024-07-23:
Attempting to run `try_clone()` on a TcpStream results in the following error.

```
Os { code: 88, kind: Unsupported, message: "Function not implemented" }
```

In my use case, this makes it impossible to use Apache Thrift on my ESP32.

Here is a minimal reproducible example: https://github.com/victorbnl/esp-rs-tcpstream-clone-88

ivmarkov, 2024-09-15:
Sorry for reacting so late.
`try_clone` is unlikely to be implemented anytime soon, as there's a missing functionality in ESP-IDfFitself: it does not support the Posix [`dup`](https://pubs.opengroup.org/onlinepubs/9699919799/functions/dup.html) syscall. In other words, it cannot "duplicate" a file/socket handle. This also means we cannot support `try_clone`, as after `try_clone` you should end up with two separate TCP streams which have their own lifetime, but for that, we absolutely need `dup`. :)

ivmarkov, 2024-09-15:
If you feel like contributing a `dup` support to ESP-IDF, implementing TCP stream `try_clone` would be a piece of cake afterwards.

closed 2024-09-18 by ivmarkov