esp-rs/esp-idf-sys #330: help with, TCP connection

nglint, 2024-08-02:
hello,I need to take pic with esp32 dev cam and send it using tcp socket to a winsock server , first i tried it using arduino IED the transfer speed was too slow and unacceptable because it use stop and wait arq.... after that i read that using esp32-IDE i can change the tcp options, now after trying it i facing more problems  which the connection is lagging and it send  one packet every second and some time it take more time, but also some time the speed is increasing , i just use the  code for wifi connection and for the tcp connection and for starting the cam, from the espidf examples then i tried many changes..., any advice.

Vollbrecht, 2024-08-02:
The issue mechanism are for reporting bugs, making suggestions etc and are not a support line.

 Also try to reverse the situation. Someone who reads your post could not help you in any meaningful way, as you don't get concrete with any specific problems. Saying " i tried many changes" says exactly nothing, so nothing can be replied to that.

For rust specific esp questions you can join the linked matrix channel mention in the README.

nglint, 2024-08-02:
ok , sorry this first time i use github to ask for help ...  feel free to delete the issue, because i dont find way to delete

nglint, 2024-08-02:
.

closed 2024-08-02 by nglint