esp-rs/esp-idf-sys #333: No IEEE802154 on ESP32C6

turulix, 2024-08-10:
I'm trying to use this to get thread running on my ESP32C6, but can't find the bindings for it anywhere

ivmarkov, 2024-09-15:
Yes - we need someone who can contribute the bindings.
I would gladly use those as well, but I don't have the time to do the contribution myself...

ivmarkov, 2024-09-17:
@turulix 
In the meantime, in `master`:
- `esp-idf-sys` has (most of the) raw bindings to the ESP IDF wrappers of OpenThread (but I have not yet exposed the native OpenThread APIs; they are huge and I'm not sure exposing all of these is necessarily the way to go)
- `esp-idf-hal` does have the `ThreadModem` for c6 (should've been named IEEE802154, but who can pronounce that?!)
- `esp-idf-svc` [has this still-in-progress PR](https://github.com/esp-rs/esp-idf-svc/pull/484)

But with that said, may I ask how do you actually plan to use Thread? If you want matter-over-thread, you also need matter.
If you have other things in mind, I'm genuinely interested how you plan to use it. And in particular, how do you plan to commission your devices in Thread?

In fact, I'm yet to find a Thread Commissioning app for my phone. "Thread" being such a common word definitely does not help...

ivmarkov, 2024-09-17:
I'm closing this as we have the Thread modem on c6 now, and also we have the initial raw Thread bindings exposed.
But would still be very eager to understand where you are going with Thread.

closed 2024-09-17 by ivmarkov

turulix, 2024-09-18:
The issue I'm currently facing is that to use Thread properly, you do require a Thread border router, which essentially forwards requests into and out of the Thread network. There are some off-the-shelf devices that support this role, but I want to build one out of two ESP32C6 modules (which should be possible, as far as I'm aware) because it's cheaper than buying one.

After successfully setting up a Thread border router, the next goal would be to use Matter over Thread to commission devices and actually use them.

Why Thread and not Zigbee? Well, Thread is more robust to device failures than Zigbee. If the Zigbee bridge dies, the entire network goes down, which isn't ideal for smart home devices. With Thread, if the router dies, the network still exists, and things like switches can still be used—just your phone won't be able to reach the devices anymore.

ivmarkov, 2024-09-18:
To use Matter, you also need a Matter Controller. The majority (if not all) of the Matter Controllers (Google Home, Alexa, AppleTV, recent Samsung TVs) are _already_ Thread border routers.

I think the only case where you would need a dedicated TBR is if you plan to use Home Assistant as your Matter controller, as (obviously) the RPI does not support Thread out of the box.

Yes, this can be done with two (or even one) esp32c6 - with Wifi + Thread co-exist mode (if it works reliably) or by just using an $15 SPI Ethernet adaptor which would give you c6 Ethernet connectivity.

I think the shortest path to success might be just altering the [existing ESP IDF C TBR example](https://docs.espressif.com/projects/esp-thread-br/en/latest/) to suit your needs, but if you would like to re-do it all in Rust (need to know Rust well though) I don't mind, as we'll have an immediate use case for my Thread implementation. :)