esp-rs/esp-idf-sys #335: Build fails with version 5.3 of esp_idf

vortex314, 2024-08-15:
After downloading latest versions of esp_idf, espup, rustup update and :+1: 

```shell
cargo generate esp-rs/esp-idf-template cargo
```

 Looks like a version mismatch somewhere 

```
  Using activated esp-idf v5.3.0 environment at '/home/lieven/esp/esp-idf'
  CMake Error at /home/lieven/esp/esp-idf/tools/cmake/tool_version_check.cmake:36 (message):
    

    Tool doesn't match supported version from list ['esp-13.2.0_20240530']:
    /home/lieven/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
```
complete log.
```
Compiling esp-idf-sys v0.35.0
error: failed to run custom build command for `esp-idf-sys v0.35.0`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_RELEASE_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `/home/lieven/workspace/zenoh-projects/esp32-idf/target/release/build/esp-idf-sys-f4610d522c3e3f3b/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  cargo:rerun-if-changed=/home/lieven/workspace/zenoh-projects/esp32-idf/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/lieven/esp/esp-idf" IDF_TARGET="esp32" PATH="/home/lieven/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/lieven/esp/esp-idf/components/espcoredump:/home/lieven/esp/esp-idf/components/partition_table:/home/lieven/esp/esp-idf/components/app_update:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:/home/lieven/.local/bin:/home/lieven/.espressif/tools/xtensa-esp32-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-unknown-linux-gnu/bin/:/home/lieven/.espressif/tools/xtensa-esp32s2-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-unknown-linux-gnu/bin/:/home/lieven/.espressif/tools/xtensa-esp32s3-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-unknown-linux-gnu/bin/:/home/lieven/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/lieven/.cargo/bin:/home/lieven/esp/esp-idf/components/espcoredump:/home/lieven/esp/esp-idf/components/partition_table:/home/lieven/esp/esp-idf/components/app_update:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:/home/lieven/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/home/lieven/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:/home/lieven/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/lieven/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:/home/lieven/.espressif/python_env/idf5.3_py3.10_env/bin:/home/lieven/esp/esp-idf/tools:/home/lieven/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/local/go/bin:/home/lieven/.local/bin:/home/lieven/go/bin:/home/lieven/arm/bin:/usr/local/go/bin:/home/lieven/.local/bin:/home/lieven/go/bin:/home/lieven/arm/bin" PROJECT_DIR="/home/lieven/workspace/zenoh-projects/esp32-idf" SDKCONFIG_DEFAULTS="/home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out/gen-sdkconfig.defaults;/home/lieven/workspace/zenoh-projects/esp32-idf/sdkconfig.defaults" "cmake" "/home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/lieven/esp/esp-idf/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/lieven/.espressif/python_env/idf5.3_py3.10_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy" "-DCMAKE_ASM_FLAGS= -mlongcalls"
  -- Building ESP-IDF components for target esp32
  -- Checking Python dependencies...
  Python requirements are satisfied.
  Constraint file: /home/lieven/.espressif/espidf.constraints.v5.3.txt
  Requirement files:
   - /home/lieven/esp/esp-idf/tools/requirements/requirements.core.txt
  Python being checked: /home/lieven/.espressif/python_env/idf5.3_py3.10_env/bin/python
  -- Project sdkconfig file /home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out/sdkconfig
  Loading defaults file /home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out/gen-sdkconfig.defaults...
  /home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out/gen-sdkconfig.defaults:2 CONFIG_COMPILER_OPTIMIZATION_DEFAULT was replaced with CONFIG_COMPILER_OPTIMIZATION_DEBUG 
  Loading defaults file /home/lieven/workspace/zenoh-projects/esp32-idf/sdkconfig.defaults...
  -- Compiler supported targets: xtensa-esp-elf
  -- Configuring incomplete, errors occurred!
  See also "/home/lieven/workspace/zenoh-projects/esp32-idf/target/xtensa-esp32-espidf/release/build/esp-idf-sys-9c5a21508b2dd7d0/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32",
      ),
[... 28 lines cut ...]
    /home/lieven/esp/esp-idf/tools/cmake/build.cmake:400 (include)
    /home/lieven/esp/esp-idf/tools/cmake/build.cmake:632 (__build_process_project_includes)
    /home/lieven/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/lieven/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  stack backtrace:
     0: rust_begin_unwind
     1: core::panicking::panic_fmt
     2: cmake::fail
     3: cmake::run
     4: cmake::Config::build
     5: build_script_build::native::cargo_driver::build
     6: build_script_build::native::build
     7: build_script_build::main
     8: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

 *  The terminal process "/usr/bin/bash '-c', '~/.cargo/bin/cargo run --release'" terminated with exit code: 101. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

vortex314, 2024-08-15:
Downgrading to esp_id 5.2 solved the build issue.

Vollbrecht, 2024-08-15:
That seams like an env issue on your site. If you are opted to use a activated ESP-IDF environment, we don't manage or install the necessary tools for that ESP-IDF version you are providing. So you need to make sure that you can compile a bare ESP-IDF project with that provided environment. 

Some pointers. Usually the ESP-IDF env installs the xtensa-gcc toolchain in its own tools folder. You are still providing a xtensa-gcc variant installed via espup in the rustfolder. We usually don't use that in esp-idf-hal and are providing always the correct version automatically for each ESP-iDF version.

If in your case the espup xtensa-gcc version is first in your PATH it will always use that specific fixed version. That is what this problem here is about -  not using the correct xtensa-gcc toolchain. 

On a site note ESP-IDF v5.3 was just released recently and we still need to incorporate some changes to make it fully usable with the esp-idf-hal. I am currently a bit busy but i hope that i finish the integration soon with the p4 support.

vortex314, 2024-08-15:
Downgrading esp ifdf to 5.2 solved the issue. I didn't investigate further.

closed 2024-08-15 by vortex314